MarlinFirmware/Marlin #14054: [BUG] Invalid PlatformIO board

GMagician, 2019-05-18:
It seems that `malyanM200v2.json` is not correct and it causes a exception in PlatformIO when using sam boards. Anyone as seen it?

RudyBenner, 2019-05-18:
Not at all user-friendly in VSCode

GMagician, 2019-05-18:
If I run:
platformio boards
with such json I get an error

xC0000005, 2019-05-18:
I haven't updated the malyanM200 json file in forever. For all lerdge x, k, and s development, as well as malyanM200  V1, V2, V3 and M300, I'm using the arduino IDE to build. I'll look and see if I can fix it later this weekend, at least to the point where you can build for the V1 and V2 using the HAL_STM32 (ST's arduino core).

GMagician, 2019-05-18:
I don't need to compile malyanM200V2, I'm trying to build for Adafruit Grand Central but json prevent me to use it

GMagician, 2019-05-18:
This is error I get:

> AssertionError: Missed J-Link Device ID for malyanM200v2

xC0000005, 2019-05-18:
I’m not near a machine I can build on, but looking at the other board (the black) in the same dir, it has a few more attributes under its debug node in the json:
    "jlink_device": "STM32F407VE",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F40x.svd",

I’d copy these in, save the file, and see if it lets you build. They’re not right, but since you aren’t even working on that board, it hardly matters so long as the change unblocks you. PIO should get proper json definitions for the M200/M300 series soon, since ST’s core now officially supports them.

GMagician, 2019-05-18:
I temporary solved deleting the file. Don't know why none is seeing such error

ivankravets, 2019-05-20:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/14070

closed 2019-05-20 by GMagician