MarlinFirmware/Marlin #14055: TMC2208 StealthChop2 autotune procedure not respected

Arti4ever, 2019-05-18:
### Description

It seems that the StealthChop2 autotune procedure is not respected for TMC2208 (and similar) stepper driver.

According to the TMC2208 datasheet, StealthChop2 need the following procedure in order to work properly and with the maximum efficiency :
 
1. enable stepper
2. stand still at least for 130ms
3. move at medium speed for about 100 fullsteps

During homing, for example, these 3 steps are not respected because steppers start moving right after they are enabled and without waiting for 130ms.

Link to the datasheet (page 38) : 
[https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC222x_datasheet_Rev1.07.pdf](url)

Maybe it would be a good thing to add a delay of 130ms (or more) after each stepper enable (for TMC2208).

teemuatlut, 2019-05-18:
Sounds like good idea.

joseitor, 2019-05-21:
According datasheet; it must be in standstill for more than 130ms (a delay of 150 or 200ms would be great as you said); and between 64 and 400 fullsteps with a velocity between 60-300 RPM. We should calculate the linear velocity that correspond to those RPM and recommend them for homing velocity.
StelthChop2 autotune is a very important issue.

gloomyandy, 2019-05-21:
Hmm the thing is though can you actually guarantee that a homing operation will be able to fulfill those needs. Say for instance the head is already right up against the endstops, it may no be able to move enough to perform the tuning. Maybe we need to perform a set of tuning moves after homing to be sure? 

Also I'm a little confused about the standstill criterion what exactly does that mean, does that mean disabled or is just not moving enough? In which case does that mean that if you print a long straight line (so say no movement on X, but movement on Y), then the X axis will retune next time it moves? That would mean the steppers are constantly being autotuned, is that a good idea? If not it may be better to have a tune routine and then set the values explicitly and turn off autotune? 

Maybe I should go read the datasheet again!

joseitor, 2019-05-24:
I guess it will continue autotuning during the next movements if homing operation it's not long enough. Also I read in the datasheet that "It automatically learns the
best settings during the first motion after power up and **further optimizes the settings in subsequent
motions**" so not a big issue. I think that add a first delay to the 130ms standstill would be a first step to ensure that an instantaneous movement after power the driver up doesn't disturb the autotune process.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle