MarlinFirmware/Marlin #14058: [BUG] JUNCTION_DEVIATION wrecks Z moves (disobeys acceleration / 'jerk' limits)

gruvin, 2019-05-19:
## Steps to reproduce

*Hopefully...*

1. Disable JUNCTION_DEVIATION as sanity check, then run the following test code ...

```
M201 X3600 Y3600 Z150 E8000 ; sets maximum accelerations, mm/sec^2
M203 X150 Y150 Z18 E50 ; sets maximum feedrates, mm/sec
M204 P3000 R1500 T3000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X16.00 Y16.00 Z0.30 E0.20 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
G90 ; use absolute coordinates
G21 ; set units to millimetres
M83 ; use relative distances for extrusion

G28
G92 E0 ; Reset E coordinate. For old Marlin bug. Can't hurt.
G1 X0 Y0 Z20 F8000
G1 X20 F9000
G1 X30 Z10 F9000 ; <=== where things go wrong
```

Observe that the last line, `G1 X30 Z10 F9000` works as expected, especially Z movement

2. Enable JUNCTION_DEVIATION. Then run the same test code. 

### Observe that Z motion from last line is now broken.

Stepper pulses are emitted with seemingly no acceleration curve, at some random and seemingly constant and errant feed rate. X axis does not appear to be affected (in this case.)

3. I found a **work-around**, for this specific gcode test fragment only ...
```
...
...
G1 X20 F9000
M400 ; <== Work around for obscure JUNCTION_DEVIATION planner bug!
G1 X30 Z10 F9000
```

Obviously, I cannot predict where to put additional workaround  `M400`s throughout the rest of the print. So far, I have had only one print suffer Z axis funnies outside print preparation code. So, it's relatively rare. I use 0.3mm Z-hop.

**Expected behavior:**

Z to, well, "screw up" but without losing steps. (See what I did there? :yum:)

**Actual behavior:**
Destroyed prints, with occasional high pressure steam emissions from certain cranial orifices.

#### Additional Information

- This bug only appears here with the *EXACT* code shown above. Almost any minor change to anything before the last line -- especially after or including the `G28` -- will remove the buggy operation. (Yes, it took AGES to narrow this down!)
-  The feed rate on that last line doesn't appear to matter. It can be anything. 
-  My Z max F/R is 18mm/min, A is 500mm/s and has been stress tested extensively under extreme conditions, at higher 20mm/min and 600mm/s^2, before settling on lower values for extra confidence.
-  I *think* I have had X/Y layer shifting issues on the odd print. I had one that I printed three times, with the exact same shift, at the exact same height. In that case, I accidentally forgot to rotate the part as I normally did and it printed fine, the forth time. I was thinking hot end collision and only just now realized it could actually be this. Hmmm. 
- ` #define JUNCTION_DEVIATION_MM 0.02` -- not overridden by EEPROM config.
- [ZIP file containing my `Configuration.h` and `Configuration_adv.h` files: Configation[_adv].h.zip](https://github.com/MarlinFirmware/Marlin/files/3194612/Configation._adv.h.zip)
- **Firmware VERSION** (branch) `bugfix-1.1.x`

#### Video showing correct operation
[coming soon]

#### Video showing errant operation
[coming soon]

#### Video showing correct operation again, with the `M400` work-around, as detailed above.
[coming soon]

#### Video showing the problem gone away after *merely* changing the test code as follows (without the `M400`) ...

```
G28
G92 E0 ; reset E coordinate (For old Marlin bug. Can't hurt.)
G1 X0 Y0 Z20 F8000
G1 X19 F9000 ; <=== very minor change. Problem goes away.
G1 X30 Z10 F9000
```
[coming soon]

### Related bug? May offer some clue?

LINEAR_ADVANCE goes incredibly slow without JUNCTION_DEVIATION enabled. (K factor 0.4.) Cause unknown. No workaround yet discovered. Changes to X/Y/Z/E jerk certainly don't help. Main limitation seems to be extruder speed. I can limit that to slow it down even more but not speed things up - even using silly extreme values. 🤷‍♂

gruvin, 2019-05-19:
Right now, I'm just hoping someone can duplicate this bug on their printer. We'll take it from there, shall we? 😉

gruvin, 2019-05-19:
For sanity (again) and since I maintain my own git clone repo, with historical merging and so on ... I thought it prudent to download a **fresh Marlin (tag) 1.1.9** and then **manually edit** Configuration[_adv].h files*, then run the test again.

**The same fault was observed.**

FWIW, I have attached the fresh new and ostensibly identical config files, herein. 

[Marlin_fresh_1.1.9_Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3194730/Marlin_fresh_1.1.9_Config_files.zip)

gloomyandy, 2019-05-19:
If you want this to get attention it would probably help if you reproduce it with bugfix2.0.x most developer attention is now focused on that version.

gruvin, 2019-05-19:
Err ... OK. But v2.x is still Alpha, right? Cautions to not use and such. Thus, I have no intention to use it until it's at least beta. That just makes no sense.

Nevertheless, I'll do as you suggest by downloading v2, going through the learning curve to set it up (it's looks way different) just to reproduce this bug. Perhaps it will help.

I would expect to encounter many other bugs in v2 if I dare use the thing in production. Seems every time Marlin gets a new feature, something else breaks. So I won't be doing that. :yum:

reloxx13, 2019-05-19:
2.0.x exists since 2017.
the fw is in continuously development. use it and update it in a regular interval.
use git and a fork to update ur local 2.0.x marlin for easy merging.

gruvin, 2019-05-20:
> 2.0.x exists since 2017.

That's interesting. I certainly am using it now. Seems really good. So, thanks. ;-)

It's no wonder not many people are using it though (proportionally, from what I can tell) because ...

- The only v2 branch is `bugfix-2.0.x`
- There is no release version at all. The latest release is still `1.1.9`.
- There's this big warning not to use v2 bugfix (the only existing branch) ...

[image]

No doubt I'm missing something. :yum:

Thanks again.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle