MarlinFirmware/Marlin #14059: [BUG] 2.0.x LPC1768 TMC2130 steppers no longer move

dp250f, 2019-05-19:
### Description

Was using repository from 5/11/2019, stepper motors all worked.
Cloned new repository form 5/18/2019, steppers no longer move.

### Steps to Reproduce

1. Clone repository from today, configure and compile for LPC1768 (BOARD_BIGTREE_SKR_V1_3) and TMC2130 drivers in SPI mode

**Expected behavior:** Steppers move

**Actual behavior:** Steppers don't move

#### Additional Information

This seemed to work fine from when I had compiled it earlier this week, using the build from 5/11/2019.  Not sure if there was a marlin update or a library update which broke it, but something definitely did.
[bug report.zip](https://github.com/MarlinFirmware/Marlin/files/3194698/bug.report.zip)

dp250f, 2019-05-19:
..and I can confirm it's the TMCStepper library update from 0.3.2 to 0.3.3 which broke it.  If I specify TMCStepper@0.3.2 in my platformio.ini, it works like it did before.

thisiskeithb, 2019-05-19:
I ran into the same issue: https://github.com/MarlinFirmware/Marlin/issues/14051

doggyfan, 2019-05-19:
Not related, but in your configuration.h file @dp250f, I don't think you need to specify Z2 AND E1, it's one or the other really.

```cpp
#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2130
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130
#define E1_DRIVER_TYPE TMC2130
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
```

doggyfan, 2019-05-19:
Also, have you tried using HW SPI being as the TMC chips have their very own hardware SPI bus ?

in configuration_adv.h, try commenting out this line ..

```cpp
#define TMC_USE_SW_SPI
```

doggyfan, 2019-05-19:
Another also, but maybe useful tip is that if you specify the upload drive in your platformio.ini file in the LPC section, Atom/platformio will automatically copy the firmware.bin file onto the little SD card that is plugged into the SKR 1.3 board for you when you compile it - if you so wished.

In my case, this is ..

```
[env:LPC1768]
upload_port = D:\
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip
framework         = arduino
```

dp250f, 2019-05-19:
> 
> 
> Not related, but in your configuration.h file @dp250f, I don't think you need to specify Z2 AND E1, it's one or the other really.
> 
> ```c++
> #define X_DRIVER_TYPE  TMC2130
> #define Y_DRIVER_TYPE  TMC2130
> #define Z_DRIVER_TYPE  TMC2130
> //#define X2_DRIVER_TYPE A4988
> //#define Y2_DRIVER_TYPE A4988
> #define Z2_DRIVER_TYPE TMC2130
> //#define Z3_DRIVER_TYPE A4988
> #define E0_DRIVER_TYPE TMC2130
> #define E1_DRIVER_TYPE TMC2130
> //#define E2_DRIVER_TYPE A4988
> //#define E3_DRIVER_TYPE A4988
> //#define E4_DRIVER_TYPE A4988
> //#define E5_DRIVER_TYPE A4988
> ```

Yeah, I was getting desperate.  I've removed it since I got it to go again.
I have upload_port set to E: since that's the drive letter it gets.
I was wondering about hardware SPI.  Instructions for my board say to use sw spi (pins file seems to require it)

doggyfan, 2019-05-19:
oh ok about the hardware SPI. That's a shame :(

gloomyandy, 2019-05-19:
@doggyfan On the SKR V1.3 the pins connected to the SPI pins on the TMC drivers are...
  MOSI      P4_28
  MISO      P0_05
  SCK       P0_04
That pin combination does not have a hardware SPI controller connected to it. On the SKR board hardware SPI is on:
  SCK          P0_07
  MISO         P0_08
  MOSI         P0_09
and
  SCK          P0_15
  MISO         P0_17
  MOSI         P0_18
So you would need to change the way that the SPI bus is connected to the drivers.

Also I think that the TMC library currently uses that HAL to provide hardware SPI support and that only supports a single hardware SPI device (which more often than not will be in use for the SD card and/or LCD).

doggyfan, 2019-05-19:
oops, yes of cause @gloomyandy, I looked at the schematic and got it wrong, the TMC's are not connected to a hardware SPI port, shame.

Am surprised Marlin can only do a single hardware SPI, that's a shame too. That could be solved though if need be.

hoe hum lol

jjansen85, 2019-05-19:
Strange, i just updated to the latest (May 19) Marlin + TMC 0.3.3. Also on a SKR V1.3. No problems at all.. 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3194895/Config.zip)

dp250f, 2019-05-19:
> 
> 
> Strange, i just updated to the latest (May 19) Marlin + TMC 0.3.3. Also on a SKR V1.3. No problems at all..
> 
> [Config.zip](https://github.com/MarlinFirmware/Marlin/files/3194895/Config.zip)

It looks like you're using an older configuration.h and configuration_adv.h
Which E0 driver are you using?

jjansen85, 2019-05-20:
E0 is a A4988

> > Strange, i just updated to the latest (May 19) Marlin + TMC 0.3.3. Also on a SKR V1.3. No problems at all..
> > [Config.zip](https://github.com/MarlinFirmware/Marlin/files/3194895/Config.zip)
> 
> It looks like you're using an older configuration.h and configuration_adv.h
> Which E0 driver are you using?

E0 is a A4988

gloomyandy, 2019-05-20:
There has been an update to the stepper library to 0.3.4, so it is probably worth making sure you have that version and trying things again.

dp250f, 2019-05-22:
Thanks!  Version 0.3.4 is working.

closed 2019-05-22 by dp250f