MarlinFirmware/Marlin #14062: [FR] G28 Start Script

sensei73, 2019-05-19:
Hello there, thanks to all for your Hard work!

I have a TouchMi sensor --> see here!

It is activated by a magnet placed on the LEFT side of the bed. So I need to go to Max_Bed_X to activated the sensor. It will be nice and user-friendly, if we could have a G28 start script possibility to activate this sensor as we have a G29 end script (I can disable it with it).

Actually I can achieve it by a dirty trick, a modification of G28.cpp but it's nasty.

Thanks in advance.

FanDjango, 2019-05-20:
> a modification of G28.cpp but it's nasty.

Let's see it

sensei73, 2019-05-29:
never mind, I needed this for a TouchMI probe but I did a real implantation of the probe there.
https://github.com/MarlinFirmware/Marlin/pull/14101

closed 2019-05-29 by sensei73