MarlinFirmware/Marlin #14067: [BUG] Statusscreen completely "broken" after updating firmware

th33xitus, 2019-05-19:
### Description
I updated from commit `a43e892` to the most recent commit `3cd19a8`.
After flashing the firmware my `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` shows the following

[image]

I can control the printer via Octoprint without any problems, all functions work, all the monitoring via Octoprint works as well. There are no functional issues with the printer itself misbehaving. It is just the display which is messed up for an unknown reason.

The hardware i am using is a modified RAMPS1.6+ combined with an Arduino DUE, TMC2130 Stepper Driver on all axes and a 12864 Full Graphic Display.

#### Additional Information
The following files are attached:
```
- 	Configuration.h
- 	Configuration_adv.h
- 	pins_RAMPS.h
- 	pins_RAMPS4DUE.h
```
[Data.zip](https://github.com/MarlinFirmware/Marlin/files/3195365/Data.zip)

trajesus, 2019-05-19:
Having the same problem, updated the firmware on my printer yesterday and that happened.

gloomyandy, 2019-05-20:
@th33xitus for some reason I can't read your configuration files, but if you are using SPI for your TMC drivers you might want to make sure you have the latest (changed yesterday) TMC library (0.3.4) and try again.

@trajesus it may help if you upload your configuration files. Oh and the same comment above also applies to you.

If things are still broken then you may have to work your way back to identify what change has caused the problem.

th33xitus, 2019-05-20:
@gloomyandy Why aren't you able to read my files? You have to download the `Data.zip` and extract them :D The archive contains the 4 files i listed at the end of my post.

I don't understand why a problem with the display should be caused by an outdated TMC library. Yes i compiled the FW with the 0.3.3, i just updated, recompile and will upload the FW again in the next minutes to validate. But i can't imagine it will change anything because everything else works. If there would be a problem regarding the SPI, i would get some driver errors in the console, axis not moving would be the result of that.

Ans yes, if no one has a clue or an idea i guess i need to flash several commits until the error occurs to be able to give a more precisely problem report.

robustini, 2019-05-20:
Same problem here, with RuRamps V1.3 and Arduino 2, TMC2130 in SPI.
Two printers, same boards, drivers and LCD, same issue...

[image]

gloomyandy, 2019-05-20:
@th33xitus Yes I know how to read a zip file :-) For some reason when I unzip your file on a Chromebook the files are unreadable. I don't normally have a problem with other zip files.

Anyway the reason why I suggested trying the updated TMC libs is because the 0.3.3 version had changes to the way that the TMC lib uses the SPI interface and and some systems the SPI bus is shared between the TMC drivers and the display. The changes to the TMC libs resulted in that bus being operated in a different SPI mode. If your system is sharing the SPI bus then there was a chance this change was messing up your display (some displays will continue to see SPI signals even though their chip select is not active, that is why you get noise on the displays if some systems when things like the SD card is active). Som trying with the updated TMC lib (that reverted the SPI changes) was a quick way to check if this is a problem or not.

robustini, 2019-05-20:
Solution (here:)
In the file "u8g_com_sw_spi_shared.cpp" search for "DELAY_NS(125)", replace 125 with 905 and the display will work again, here it works.. ;)

th33xitus, 2019-05-20:
I can confirm, this change will fix the issue! Thanks @robustini !
I will edit the heading that a solution was found.

`u8g_com_HAL_DUE_sw_spi_shared.cpp` in line 90
```
-   DELAY_NS(125);
+   DELAY_NS(905);
```

@thinkyhead can this change please be merged into a new commit soon? :)

trajesus, 2019-05-20:
I also confirm it fixes the display.
However, now it is my TMC2130 that does not work, I must have another problem.

trajesus, 2019-05-20:
here are my configurations and the pins file for the RADDS if anyone could help
[data.zip](https://github.com/MarlinFirmware/Marlin/files/3197090/data.zip)

th33xitus, 2019-05-20:
@trajesus Did you update the TMC library? If yes, maybe revert back to an older version of it and try again.
What is the exact problem you have now?

trajesus, 2019-05-20:
I am using plataformaIO and it is choosing version 0.3.3, I will try to force it to version 0.3.2.
The problem now is that the steppers don’t move, any on them.

th33xitus, 2019-05-20:
Try to force it to use version 0.3.4 first. Thats the most recent version.
Maybe that will fix it.

robustini, 2019-05-20:
@trajesus I'm printing now, so I think your problem is elsewhere.

trajesus, 2019-05-20:
I can only find 0.3.3 as the latest version, will try the 0.3.2 and see if it works.

trajesus, 2019-05-20:
@robustini I think so too, will try to change the tmc library version and see if it works first, otherwise I will revert to an older marlin 2.0 version I have

trajesus, 2019-05-20:
Or open another issue since this one is solved

robustini, 2019-05-20:
The latest library version is 0.3.4.

robustini, 2019-05-20:
@trajesus Try to compile the firmware with Arduino.

trajesus, 2019-05-20:
I think I have to use platformIO since I'm using an Arduino DUE.
Can't change the library version, it automatically uses the 0.3.3

robustini, 2019-05-20:
Why? I use Arduino DUE and i compile with Arduino.

trajesus, 2019-05-20:
@robustini really? will try then, thanks :)

robustini, 2019-05-20:
@trajesus Please uncomment "#define TMC_DEBUG" in the "Configuration_adv.h", then post here the "M122" output.

trajesus, 2019-05-20:
@robustini used the arduino software as you said with the 0.3.4 version of the TMC library, it worked.
Problem solved, thank you so much :)

boelle, 2019-07-21:
closing since solved

closed 2019-07-21 by boelle