MarlinFirmware/Marlin #14073: [QUESTION] What's HAS_TIMER's purpose

GMagician, 2019-05-20:
In fastio there is a HAS_TIMER define that I can't see used in Marlin, then why is it defined?

ManuelMcLure, 2019-05-20:
It is used:
```
HAS_TIMER          85 src/HAL/HAL_STM32/fastio_STM32.h #define PWM_PIN(P)              HAS_TIMER(P)
HAS_TIMER          55 src/HAL/HAL_STM32F1/fastio_STM32F1.h #define PWM_PIN(P)              HAS_TIMER(P)
HAS_TIMER          55 src/HAL/HAL_STM32F4/fastio_STM32F4.h #define PWM_PIN(P)              HAS_TIMER(P)
HAS_TIMER          54 src/HAL/HAL_STM32F7/fastio_STM32F7.h #define PWM_PIN(P)              HAS_TIMER(P)
```

xC0000005, 2019-05-20:
in HAL_STM32, it's correctly defined as digitalPinHasPWM.

tpruvot, 2019-05-20:
Marlin/src/HAL/HAL_STM32F1/HAL.h:132:  #define digitalPinHasPWM(P) (PIN_MAP[P].timer_device != nullptr)
Marlin/src/HAL/HAL_STM32F1/fastio_STM32F1.h:53:#define HAS_TIMER(IO)         (PIN_MAP[IO].timer_device != nullptr)

GMagician, 2019-05-20:
It's defined yes it's used no
I searched in the whole project and I only found the define...

tpruvot, 2019-05-20:
you didnt search well
#define PWM_PIN(P)              HAS_TIMER(P)

GMagician, 2019-05-20:
@tpruvot that's not completely correct, this is a define of a define, so only PWM_PIN is used and could be declared directly with HAS_TIMER content (both are inside fastio) but while PWM_PIN has a sense (it's used for sanity checks) the other?

tpruvot, 2019-05-20:
its much more complicated, its not because a pin has a timer associated by default than PWM is enabled on it... and i dont even talk about the remap ones...

GMagician, 2019-05-20:
There is also `USEABLE_HARDWARE_PWM` that is defined and never used...

GMagician, 2019-05-20:
@tpruvot I know about it but since PWM_PIN is used in code and HAS_TIMER isn't I'm wondering why this has to be defined

GMagician, 2019-05-21:
@Bob-the-Kuhn I think you added `USEABLE_HARDWARE_PWM` to HAL. Since it's never used in whole Marlin code, why has it to be present?

Bob-the-Kuhn, 2019-05-21:
It was used at one time. Sounds like the code has changed and it's no longer used.

Bob-the-Kuhn, 2019-05-21:
USEABLE_HARDWARE_PWM was part of the original "PWM for any pin" code back in the early days of the LPC1768 HAL.  That HAL has been re-written multiple times since then.  I'm not surprised that it's no longer used.

closed 2019-05-22 by GMagician