MarlinFirmware/Marlin #14075: [2.0.x] Marlin video - LINUX, ESP32 & Teensy HALs - need some advice

Bob-the-Kuhn, 2019-05-21:
I'm slowly putting together a video on how to install and upload Marlin 2.0.x. I have 90% of the PlatformIO with AVR, LPC1768/9 & DUE/SAM3X8 portion.  I'm talking to the STM32 experts in another issue.

I should mention the LINUX, ESP32 & Teensy HALs in this video.  

Since there's no RAMPS style I/O boards for the ESP32 & Teensy HALs I currently am just planning a single slide mentioning they're available and are mostly used as learning tools and for creating small projects..  

I haven't the faintest idea what the LINUX HAL is useful for.  Some enlightenment would be helpful.

p3p, 2019-05-21:
The Linux HAL is just for debugging Marlin, being able to run Marlin as an executable on a computer makes things a lot simpler, the pin states are logged to csv while inputs (thermistors / endstops) are simulated, It's limited to Linux because it has realtime timers that do interrupts(ish) so it's a direct port. In theory it could run on a raspberry pi using its gpio directly and I may have a branch that does just that but although it mostly works in a very limited fashion I'm not willing to support it for public use.

Bob-the-Kuhn, 2019-05-21:
Thanks - will not mention the LINUX HAL.

doggyfan, 2019-05-21:
I think you should mention @Bob-the-Kuhn,. I had no idea what the linux build was for either, and now I do, and can see how very useful it can be, maybe not for the average user, but certainly for some interested gals n guys.

Including it as a simple "it's for running it in Linux for testing" is very useful information, and information is key to all sorts of things.

I'm looking forward to seeing your video(s), can't wait really. Video's say so much more than 100's of pages or text.

xebbmw, 2019-06-17:
@Bob-the-Kuhn: it seems that vivian-ng created a board for ESP32, it is more a DIY board. More details here: [ESP32 MRR_ESPA](https://github.com/vivian-ng/Hardware/tree/master/MRR_ESPA)

simon-jouet, 2019-06-18:
For the ESP32 it's fully usable as a controller now, there are few board based on my design but we are still refining the pinout and a couple of more things. Most of the discussion is going on here https://github.com/simon-jouet/ESP32Controller/issues, Vivian-ng's board is based on the R1 schematic and Ringel just made a PR for the R2 schematic (which I still have to review).

A while back I made a  RAMPs compatible ESP32 board but the huge difference in IO pins available between an Atmega and an ESP32 made is really not practical. @Bob-the-Kuhn if you want more info on the ESP32 work I should be able to help if needed. I'm still planning to get a new version of the board at some point (not sure when), I might be able to send one your way then. @thinkyhead and @p3p have a R1 board already.

boelle, 2019-07-21:
can we close this one?

boelle, 2019-10-24:
@Bob-the-Kuhn still working on this one?

vivian-ng, 2019-11-08:
Just to mention, the ESP32-based board MRR ESPA is at this [repo](https://github.com/maplerainresearch/MRR_ESPA) now.
If anyone wants to try it, the repo has a link on how to order one.

I2S version (MRR ESPE) is at this [repo](https://github.com/maplerainresearch/MRR_ESPE). I have soldered a few together and will make it available to those who want to try it.

boelle, 2019-11-24:
i think this one has served its purpose for now so closing it, we can always reopen if that is not the case

closed 2019-11-24 by boelle