MarlinFirmware/Marlin #14077: BLTouch Trigger, Nozzle into Bed

NelsonBlvd, 2019-05-21:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

FanDjango, 2019-05-21:
I can see the probe deploying, it triggers, the trigger is not seen by the printer controller.

Try this first:

`#define Z_MIN_PROBE_ENDSTOP_INVERTING true` first.

It is a V3.0, the Marlin version is pre bugfix-2.0.x.

I don't know if it is an "early" V3.0 (pre Apr 5). These were slightly problematic.

You need bugfix-2.0.x to set these probes into 5V mode.

If they are not set explicitely to 5V mode, they will operate in open drain mode. That means you might need a resistor to pull up the signal line.

NelsonBlvd, 2019-05-21:
Honerstly, ive tried both, setting it true/false has no effect on the printer

NelsonBlvd, 2019-05-22:
changed to bugfix2.0 so i could also enable the 5v on BL Touch and nothing changed, still the EXACT same behavior as before with a couple of LCD screen bugs....
[Marlin2.0.zip](https://github.com/MarlinFirmware/Marlin/files/3205740/Marlin2.0.zip) < that is my "bugfix2.0" version of marlin.

d3dorsett, 2019-06-19:
I have a similar issue with an SKR 1.3 board. 

Switching #define Z_MIN_PROBE_ENDSTOP_INVERTING from true to false did not make a difference, however, flipping the 2-pin (whtie & black) wires did fix the issue.

Maybe it is just my understanding of how Z_MIN_PROBE_ENDSTOP_INVERTING works. Does it invert the 2-pin connector (white & black) or the 3-pin (Red, Yellow, Brown)?

ManuelMcLure, 2019-06-19:
``Z_PROBE_ENDSTOP_INVERTING`` does not affect the direction you need to connect the wires - they have to be connected in the right polarity regardless of the setting of ``Z_PROBE_ENDSTOP_INVERTING``. There is no way to change the polarity of the connection in software.

What that setting does is determine whether the endstop/probe is normally open or normally closed.

Adanufgail, 2019-08-18:
I can confirm the same issue on a EZBOARD Lite with both a Genuine BLTouch and a 3DTouch, and I can confirm swapping the black and white wires fixed the crashing issue. Setting "#define Z_MIN_PROBE_ENDSTOP_INVERTING" to either true or false did nothing.

InsanityAutomation, 2019-08-18:
That is an electrical item. These are not dry contact switches and there is a diode forcing polarity. Signal and ground here do matter. See the comment in Configuration_adv around all the bltouch options

   * Note: If the trigger signal of your probe is not being recognized, it has been very often
   *       because the BLACK and WHITE wires needed to be swapped. They are not "interchangeable"
   *       like they would be with a real switch. So please check the wiring first.

boelle, 2019-08-19:
closed  as its a hardware issue and not marlin related

closed 2019-08-19 by boelle