MarlinFirmware/Marlin #14079: [BUG] BLACK_STM32F407VE  uses 12864 to cause scrambling

FLYmaker, 2019-05-21:
### Description

### Steps to Reproduce

1. #ifndef MOTHERBOARD
  #define MOTHERBOARD 1810
#endif

2. #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** Screen messy code 。Partial normal
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3202950/Configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-05-21:
Support for this board was added recently: #6 by @jmz52 - maybe they could help out?

jmz52, 2019-05-21:
I've tested this smart controller before PR submission and it was working.
LCD display on my controller was extremely bleak and I was not able to adjust contrast to any reasonable level. Problem's root cause was USB power - it was not strong enough for 128x64 LCD. Contrast problem disappeared after I've connected 12V to my 3D shield.

I'll test this smart controller again with the latest code.

Bob-the-Kuhn, 2019-05-21:
I'm new to the STM world.

Curiosity question.  Are you using this to control a printer?  If yes, how are you connecting to the steppers, heaters, fans, … ?

FLYmaker, 2019-05-22:
> 我是STM世界的新手。
> 
> 好奇心问题。你用这个来控制打印机吗？如果是的话，你是如何连接踏步机，加热器，风扇......？

I will make PCB drawings and hand them to the factory for production

FLYmaker, 2019-05-22:
> 我在PR提交之前测试了这个智能控制器并且它正在工作。
> 我的控制器上的LCD显示非常暗淡，我无法将对比度调整到任何合理的水平。问题的根本原因是USB电源 - 它对于128x64 LCD来说还不够强大。我将12V连接到3D防护罩后，对比度问题消失了。
> 
> 我将使用最新代码再次测试这个智能控制器。

Don't use 3.3V to power the screen. Use 5V instead.

jmz52, 2019-05-22:
@FLYmaker 
I am using 5V to power the smart controller. Screen itself is powered via built-in DC-DC converter.

FLYmaker, 2019-05-22:
> @FLYmaker
> 我使用5V为智能控制器供电。屏幕本身通过内置DC-DC转换器供电。

Can you compile a bootloader for this motherboard and upload firmware via SD card

Bob-the-Kuhn, 2019-05-22:
To really make use of the "upload firmware via SD card" feature the code should also have the USB MMC capability added in and a method added to Marlin of invoking the firmware upload feature (usually it sending it through reset).

doggyfan, 2019-05-22:
It's wonderful to see the STM32F4's finding their way into Marlin :)

boelle, 2019-07-21:
@FLYmaker is this one "fixed"?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle