MarlinFirmware/Marlin #14081: [BUG] Mesh is invalid.

guestisp, 2019-05-21:
I'm getting this error every time I call "M420 V", but i'm still able to tilt the mesh, before printing, with "G29 J"

Any idea ?

```
Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 30,224)                                                              (224,224)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 |   .     -0.241  -0.114  -0.054  +0.044  +0.110  +0.148  +0.183  +0.181  +0.148
Recv:    |
Recv:  8 |   .     -0.318  -0.195  -0.149  -0.056  +0.013  +0.014  +0.033  +0.078  +0.013
Recv:    |
Recv:  7 |   .     -0.360  -0.244  -0.175  -0.069  -0.008  +0.009  +0.037  +0.049  +0.021
Recv:    |
Recv:  6 |   .     -0.406  -0.286  -0.196  -0.068  +0.018  +0.066  +0.111  +0.131  +0.133
Recv:    |
Recv:  5 |   .     -0.469  -0.332  -0.231  -0.088  +0.012  +0.056  +0.101  +0.154  +0.154
Recv:    |
Recv:  4 |   .     -0.525  -0.362  -0.262  -0.112  -0.004  +0.056  +0.120  +0.171  +0.194
Recv:    |
Recv:  3 |   .     -0.577  -0.405  -0.283  -0.126  -0.010  +0.068  +0.145  +0.203  +0.245
Recv:    |
Recv:  2 |   .    -0.608  -0.441  -0.318  -0.158  -0.030  +0.053  +0.141  +0.212  +0.214
Recv:    |
Recv:  1 |   .     -0.7  -0.387  -0.296  -0.139  -0.017  +0.067  +0.131  +0.222  +0.232
Recv:    |
Recv:  0 |   .     -0.553  -0.74  -0.267  -0.120  -0.006  +0.080  +0.166  +0.220  +0.242
Recv:         0       1       2       3       4       5       6       7         9
Recv:     ( 30, 30)                                                              (224, 30)
Recv: 
**Recv: Mesh is invalid**
Recv: Storage slot: 0
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 3.00
```

guestisp, 2019-05-21:
Well, is it just me, or i'm missing the first row of points ?

Roxy-3D, 2019-05-21:
You never filled in the left column of the mesh.    Your nozzle can reach those locations, but your Z-Probe is mounted on the right side of the nozzle and can not reach (or probe) those locations.

If you do a G29 P3 it will fill in 'reasonable' values for those mesh points.   If they are too far off...  you can edit them later.

guestisp, 2019-05-21:
Technically, I can reach that locations even with the probe. It was a wrong configuration in the firmware, i've detected it after opening the issue, so we can close.

One side question: is possible, with UBL enabled, to just probe one or multiple points and see the probe result ?

Roxy-3D, 2019-05-21:
Yes...   Sort of...   You can move the nozzle where ever you are interested, and do a G29 P1 T and it will start probing and the closest, unprobed location.  When you have seen enough points in the area, you can stop the probe by pressing and holding the encoder wheel.

guestisp, 2019-05-21:
I don't need to probe the mesh. Just doing a single probe and see the results. In other words: i move the nozzle in a specifc corner, then read the probe result. Then I move on a different corner, read the probe result and I adjust until all 4 corners are more or less equal. It's a fast and simple way to level the bed  without paper sheet and similiar. If not properly levelled, the next UBL mesh will fix.

Roxy-3D, 2019-05-21:
If you are trying to physically level the bed...   You can try G29 P1 U T       That will probe points on the outside of the bed, pretty far from each other.    After 8 or 10 points, you can stop it, and use the information to raise or lower a side or corner the 'right' amount.

boelle, 2019-07-21:
@guestisp tried what @Roxy-3D suggested?

guestisp, 2019-07-21:
No, because i've moved from a mesh levelling to 3points

boelle, 2019-07-21:
then how do we test if the bug is still present?

guestisp, 2019-07-21:
I don't know, actually i have a different issue with 3point. Keep this open

boelle, 2019-10-13:
@guestisp i guess we still need to have this one open?

boelle, 2019-10-13:
closing this one

it seems mostly to be user error. if the problem is different (and it looks like that) please open a new issues with updated configs etc

closed 2019-10-13 by boelle