MarlinFirmware/Marlin #14083: [BUG ?] TMC5160 not working since update to TMCStepper 0.3.4

Patag, 2019-05-21:
Hi all,
Thanks in advance for your great (and probably hard) work.
Here is my issue
### Configuration
Printer still under assembling
SKR v1.3
TMC5160 using SPI on X/Y/E (sensorless homing activated on X/Y)
TMC2130 using SPI on Z1/Z2 (dual Z)
Marlin bugfix 2.0 from May, 19
### Description
Since TMCStepper lib upgrade to 0.3.4, X axis moves became weird (especially homing), impossible sometimes. LCD values related to TMC stepper not valid regarding default settings in conf*.h (Y and E nor wired, so not tested yet).
M502/M500 doesn't change anything.
Revert to TMCStepper lib 0.3.2 and all run fine.

#### Additional Information
Same configuration.h and configuration_adv.h, only TMCStepper lib changed between M122

**_M122 with 0.3.4_**
		X	Y	Z	Z2	E
Enabled		true	false	false	false	false
Set current	800	800	800	800	800
RMS current	792	792	795	795	792
MAX current	1117	1117	1121	1121	1117
Run current	15/31	15/31	25/31	25/31	15/31
Hold current	7/31	7/31	12/31	12/31	7/31
Global scaler	133/256	133/256			133/256
CS actual		0/31	7/31	12/31	12/31	29/31
PWM scale	0	0	0	0	0
vsense				1=.18	1=.18
stealthChop	false	true	true	true	false
msteps		0	256	256	256	256
tstep		2097408	max	max	max	2097408
pwm
threshold		0	39	658	658	0
[mm/s]		-	101.36	3.00	3.00	-
OT prewarn	true	false	false	false	true
OT prewarn has
been triggered	true	false	false	false	true
off time		0	4	4	4	0
blank time	24	24	24	24	54
hysteresis
-end		12	2	2	2	12
-start		5	1	1	1	1
Stallguard thrs	1	1	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
stallguard		X		X	X
sg_result		1022	0	0	0	896
fsactive						X
stst			X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:3F:F7:FF
		Y	0x80:07:40:00
		Z	0x81:0C:00:00
		Z2	0x81:0C:00:00
		E	0x01:FF:7F:FC

**_M122 with 0.3.2_**
		X	Y	Z	Z2	E
Enabled		true	false	true	true	false
Set current	800	800	800	800	800
RMS current	782	782	795	795	782
MAX current	1103	1103	1121	1121	1103
Run current	31/31	31/31	25/31	25/31	31/31
Hold current	15/31	15/31	12/31	12/31	15/31
Global scaler	66/256	66/256			66/256
CS actual		15/31	15/31	12/31	12/31	15/31
PWM scale	0	0	19	20	0
vsense				1=.18	1=.18
stealthChop	false	true	true	true	false
msteps		256	256	256	256	256
tstep		max	max	max	max	max
pwm
threshold		39	39	658	658	4150
[mm/s]		101.36	101.36	3.00	3.00	1.00
OT prewarn	false	false	false	false	false
OT prewarn has
been triggered	false	false	false	false	false
off time		4	4	4	4	4
blank time	24	24	24	24	24
hysteresis
-end		2	2	2	2	2
-start		1	1	1	1	1
Stallguard thrs	1	1	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
stallguard		X		X	X
sg_result		34	0	0	0	0
fsactive
stst		X	X	X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x81:0F:00:26
		Y	0x80:0F:40:00
		Z	0x81:0C:00:00
		Z2	0x81:0C:00:00
		E	0x80:0F:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

teemuatlut, 2019-05-22:
Can you get me a `M122 V` on v0.3.4.

Is your problem only on XYE or is Z (with TMC2130) affected as well?

Patag, 2019-05-22:
Hi @teemuatlut,
Thanks for your help.
Z1/Z2 (TMC2130) are working fine.
Problem is on X at least, Y and E being not wired at the moment, can't be tested.

M122 V Lib 0.3.4

SENDING:M122 V **_right after init_**
		X		Y		Z		Z2		E
GCONF		0x00:00:00:00	0x00:00:00:04	0x00:00:00:04	0x00:00:00:04	0x00:00:00:00
IHOLD_IRUN	0x00:0A:0F:07	0x00:0A:0F:07	0x00:0A:19:0C	0x00:0A:19:0C	0x00:0A:0F:07
GSTAT		0x00:00:00:00	0x00:00:00:05	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
IOIN		0x00:40:00:80	0x30:00:00:D2	0x11:00:00:7A	0x11:00:00:7A	0x08:04:00:00
TPOWERDOWN	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80
TSTEP		0x10:01:00:00	0x00:0F:FF:FF	0x00:0F:FF:FF	0x00:0F:FF:FF	0x20:00:80:00
TPWMTHRS	0x00:00:00:00	0x00:00:00:27	0x00:00:02:92	0x00:00:02:92	0x00:00:00:00
TCOOLTHRS	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
THIGH		0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
CHOPCONF	0x00:3F:F6:00	0x00:00:82:84	0x00:02:82:84	0x00:02:82:84	0x03:FF:BF:F8
COOLCONF	0x00:01:00:00	0x00:01:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
PWMCONF	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4
PWM_SCALE	0x01:F8:00:00	0x00:00:15:23	0x00:00:00:00	0x00:00:00:00	0x00:FF:E0:00
DRV_STATUS	0xFF:BF:FE:00	0x80:07:40:00	0x80:0C:00:00	0x80:0C:00:00	0x3F:F7:FE:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

SENDING:M122 V **_after 10mm move on X only_**
		X		Y		Z		Z2		E
GCONF		0x00:00:00:00	0x00:00:00:04	0x00:00:00:04	0x00:00:00:04	0x00:00:00:00
IHOLD_IRUN	0x00:0A:0F:07	0x00:0A:0F:07	0x00:0A:19:0C	0x00:0A:19:0C	0x00:0A:0F:07
GSTAT		0x00:00:00:00	0x00:00:00:05	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
IOIN		0x02:00:04:00	0x30:00:00:D0	0x11:00:00:7A	0x11:00:00:7A	0x01:00:40:00
TPOWERDOWN	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80
TSTEP		0xF2:00:00:1F	0x00:0F:FF:FF	0x00:0F:FF:FF	0x00:0F:FF:FF	0x40:02:00:00
TPWMTHRS	0x00:00:00:00	0x00:00:00:27	0x00:00:02:92	0x00:00:02:92	0x00:00:00:00
TCOOLTHRS	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
THIGH		0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
CHOPCONF	0x00:10:00:20	0x00:00:82:84	0x00:02:82:84	0x00:02:82:84	0x03:FF:BF:F8
COOLCONF	0x00:01:00:00	0x00:01:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
PWMCONF	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4
PWM_SCALE	0x01:BF:FC:00	0x00:00:15:23	0x00:00:00:00	0x00:00:00:00	0x00:1F:FC:00
DRV_STATUS	0x00:3F:EF:F8	0x80:07:40:00	0x80:0C:00:00	0x80:0C:00:00	0x3F:FB:FF:80
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

teemuatlut, 2019-05-22:
I pushed a quick patch for the rms calculation if you want to try it out.

Black6spdZ, 2019-05-22:
Not sure if it's exactly related but I too am having issues with skipped steps with my almost exact same setup "SKR v1.3, X,Y,Z 5160, E0, E1 2130 @ 12v" using spreadcycle mode. It's an older build from 4-28. With spreadcycle disabled it works almost flawlessly albeit louder :/

teemuatlut, 2019-05-22:
Try the master branch with the patched rms calculation.

Black6spdZ, 2019-05-22:
Will do!

Patag, 2019-05-22:
Sorry for this noob question, but how can be sure I compile your patch ?
I'm using Pio/Vscode.
In platformio.ini, I replaced the line
  `TMCStepper@<1.0.0`
by this one 
  `https://github.com/teemuatlut/TMCStepper/archive/v0.3.4.zip` in the lib_deps section of the LPC1768 environment.
If that's ok, it doesn't change anything in the X axis behavior.
By the way, I just noticed wrong LCD values in the TMC driver section for hybrid thresholds (0 rather 100 and SteathChop Off rather On on X and E only. Y, Z1 and Z2 seem ok. Maybe a shift in registers settings ?
I'm doubting the problem is in RMS calculation (unless a side effect).
Thanks again

teemuatlut, 2019-05-22:
You're pulling the same version as PIO library manager already knows. Basically the very same code as before you changed anything.
The link to the master branch is `https://github.com/teemuatlut/TMCStepper/archive/master.zip`.

The LCD variables will hopefully be fixed by #14074. My testing at least was promising.

Patag, 2019-05-22:
OK
Replaced 
`https://github.com/teemuatlut/TMCStepper/archive/v0.3.4.zip`
by
`https://github.com/teemuatlut/TMCStepper/archive/master.zip`
But the issue is still alive here...
Maybe I made mistake(s) in conf files, so let me attach them.
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/3208945/conf.zip)

teemuatlut, 2019-05-22:
Try reverting your microstepping and steps per mm back to normal levels and see if that changes the weird behavior.

The are some anomalies in your ``M122` output but it's likely just a bad read.

Black6spdZ, 2019-05-24:
i downloaded 5-22 build and set all microsteps to 32. on home command x an y homes ok but as soon as z moves im getting "driver error, PRINTER HALTED, please reset"

Black6spdZ, 2019-05-24:
update: so i reset and initiated home again and the damn X and Y moved in the wrong direction. somethings not quite right

teemuatlut, 2019-05-24:
What version of the library are you using now?
The error is raised by a TMC2130 but did you ground the CLK pin on the TMC5160 boards?

Patag, 2019-05-24:
> Try reverting your microstepping and steps per mm back to normal levels and see if that changes the weird behavior.
> 
> The are some anomalies in your ``M122` output but it's likely just a bad read.

Hi @teemuatlut , changes made as requested but still bad results. Most of LCD values for TMC5160 are not initialized with default values. Only TMC2130 ones are correct.
M122
		X	Y	Z	Z2	E
Enabled		false	false	false	false	false
Set current	800	800	700	700	800
RMS current	792	792	673	673	792
MAX current	1117	1117	949	949	1117
Run current	15/31	15/31	21/31	21/31	15/31
Hold current	7/31	7/31	10/31	10/31	7/31
Global scaler	133/256	133/256			133/256
CS actual		31/31	31/31	10/31	10/31	31/31
PWM scale	0	0	0	0	128
vsense				1=.18	1=.18
stealthChop	false	false	true	true	false
msteps		256	256	16	16	16
tstep		4261412865	134234112	max	max	2228224
pwm
threshold		19	0	658	658	0
[mm/s]		0.00	-	3.00	3.00	-
OT prewarn	true	false	false	false	false
OT prewarn has
been triggered	true	true	false	false	true
off time		8	0	4	4	0
blank time	54	54	24	24	54
hysteresis
-end		12	12	2	2	4
-start		1	8	1	1	1
Stallguard thrs	1	1	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
stallguard		X	X	X	X
sg_result		1016	1016	0	0	960
fsactive		X	X			X
stst				X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:7F:EF:FC
		Y	0x03:FF:BF:FE
		Z	0x81:0A:00:00
		Z2	0x81:0A:00:00
		E	0x01:FF:7F:FC
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

M122 V
		X		Y		Z		Z2		E
GCONF		0x00:00:00:00	0x00:00:00:00	0x00:00:00:04	0x00:00:00:04	0x00:00:00:00
IHOLD_IRUN	0x00:0A:0F:07	0x00:0A:0F:07	0x00:0A:15:0A	0x00:0A:15:0A	0x00:0A:0F:07
GSTAT		0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
IOIN		0x20:07:FC:00	0xE4:00:FF:C0	0x11:00:00:78	0x11:00:00:78	0x20:00:00:00
TPOWERDOWN	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80	0x00:00:00:80
TSTEP		0x00:01:00:20	0x01:00:40:00	0x00:0F:FF:FF	0x00:0F:FF:FF	0x08:00:20:00
TPWMTHRS	0x00:00:00:13	0x00:00:00:00	0x00:00:02:92	0x00:00:02:92	0x00:00:00:00
TCOOLTHRS	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
THIGH		0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
CHOPCONF	0x00:00:03:FF	0x00:FF:BF:F8	0x04:02:82:84	0x04:02:82:84	0x00:00:7F:DF
COOLCONF	0x00:01:00:00	0x00:01:00:00	0x00:00:00:00	0x00:00:00:00	0x00:00:00:00
PWMCONF	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4	0x00:05:05:B4
PWM_SCALE	0x00:00:00:00	0x00:7F:E0:00	0x00:00:00:14	0x00:00:00:13	0x00:00:00:00
DRV_STATUS	0x00:1F:FB:FF	0x3F:F7:FF:00	0xA1:0A:00:00	0xA1:0A:00:00	0x01:FF:BE:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

Thanks

teemuatlut, 2019-05-24:
As said the LCD values need a patch that hasn't been merged in yet.
I also don't yet know if you grounded the CLK pin on TMC5160 drivers.

Patag, 2019-05-24:
Ooops, sorry, I was thinking you were talking to @Black6spdZ.
Ok for the patch. And I didn't ground CLK pin on TMC5160, but it works as expected with the 0.3.2 lib.
I'll try anyway
Thanks

teemuatlut, 2019-05-24:
Looks like I was but it would apply to all using the drivers.

I'll see about creating a test branch that reverts some of the SPI changes on LPC and see if that fixes things for you. Most of the changes between v0.3.2 and 0.3.4 have been with UART.

Black6spdZ, 2019-05-24:
I rolled back to mt older 4-28 release presumably with v0.3.2 and it works but still with the spreadcycle step skipping problem

Patag, 2019-05-24:
Bravo @teemuatlut !
It works great. This time, LCD and M122 values are correct and IAW config*.h. Thanks for you patience.
For my culture, was it the SPI operations optimisation ? (As far I remember, you talked about doubling SPI exchanges speed, no ?).
Thanks again

Black6spdZ, 2019-05-25:
I'm closer but still having problems with stealthchop. I found I had the wrong rsense and also grounded clk pins and that immediately fixed the driver overheat and coil short errors. but still getting pauses and stalls with stealthchop enabled

thisiskeithb, 2019-05-25:
I had to disable StealthChop or I skip steps with a set of a genuine Watterott 5160s.

Black6spdZ, 2019-05-25:
Ok, at least i know its not just me at this point. Im running 17hm19-2004s 0.9 motors with 32 microsteps at 12v

teemuatlut, 2019-05-25:
The LPC SW SPI SCK frequency was updated from around 70kHz to 2.4MHz. An increase of about 34x.
It's not that neither part would be capable of the speeds but the current implementation doesn't properly follow the MODE3 spec and I think the lower speed masks this issue. But it'll do for now and I'll revisit the speeds a bit later.

@thisiskeithb @Black6spdZ Update to `TMCStepper@v0.3.5`. It has the updated rms calculation.
As a more general notion as I know you've done some of these but for `TMC5160`
* Remember to update R sense values
* Current PCB design requires you to ground the **CLK** pin. Do not confuse with the SPI SCK pin.
* Update to latest TMCStepper version. v0.3.5 at the time.

Also the main advantage of TMC5160 is the much greater current capabilities. You're no longer limited by the thermals of the driver, so don't be afraid to bump up the current. Just don't hurt your motors or their mounts.

thisiskeithb, 2019-05-25:
@teemuatlut Already jumped CLK and GND and updated my rsense value. I tried using the 0.3.5 libs earlier, but they kept erroring out when compiling. I set the dependency back to 0.3.4 and I was able to compile again.
Still, [I’m getting beautiful prints](https://i.imgur.com/1iH97QB.jpg).

teemuatlut, 2019-05-25:
What's the error...?

thisiskeithb, 2019-05-25:
@teemuatlut It *was* erroring out, but now it's not. I'll have to re-flash after this print is done and do some more testing.

Patag, 2019-05-27:
Hi @teemuatlut,
Can I close this one ?

teemuatlut, 2019-05-27:
Yes if your comms are working again with the updated version.
I'm looking to re do the speed upgrade soon but hopefully I'll find a test subject or two to test it out before making a release.

Patag, 2019-05-28:
All is working again with 0.3.5 lib. If I can help for test purpose, just let me know.

closed 2019-05-28 by Patag

teemuatlut, 2019-05-28:
This is the branch with re-upgraded LPC SW SPI and hopefully properly written MODE3
https://github.com/teemuatlut/TMCStepper/tree/upgrade_LPC_SPI

It works if nothing changes and nothing breaks.

Patag, 2019-05-28:
I'll test it soon.

Patag, 2019-05-28:
Sorry if I don't use the proper way (issue closed) to talk to @teemuatlut , but it's the only one I know.
I compiled Marlin with your new test lib and unfortunately, it doesn't work : the system goes in Driver Error/Printer halted about 2 secs after main screen appears (with all displayed values consistent).
Test conf : Marlin from today (May, 28), Fysetc LCD and the rest as described above.

teemuatlut, 2019-05-28:
Interesting. Nothing about that change should cause that to happen and even if a read fails, it still shouldn't crash the system.

Could you try a few additional things as this is not what I saw in my tests
Does the board boot if you disable all TMC drivers?
Does the board boot if you disable SW SPI?
What about with just one driver enabled with SW SPI also enabled?

If you're not exhausted at that point, what happens with a clean build of Marlin with just one driver and SW SPI?

Patag, 2019-05-29:
> Does the board boot if you disable all TMC drivers?

Yes, it boots normally.

Further tests I intend to perform :
- Only one TMC5160 with and without SW SPI
- Only one TMC2130 with and without SW SPI
Hope I will be able to do that for this evening (UTC+2 here)

Patag, 2019-05-29:
Here are my test results :

- TMC5160 only (X) / SW_SPI : Driver Error / Printer Halted 4-5 secs after information screen displayed

- TMC5160 only (X) / no SW_SPI : Normal boot but TMC connection error on LCD status line

- TMC2130 only (Z) / no SW_SPI : Normal boot but TMC connection error on LCD status line

- TMC2130 only (Z) / SW_SPI : Normal boot, no error, stepper operative (THE GOOD news !)

- TMC2130 * 2 (Z/Z2) / SW_SPI : Normal boot, no error, steppers operative (THE GOOD news again !)

All test performed with fresh Marlin installation from May,28, .piolibdeps folder deleted before the first clean/compile commands.
In platformio.ini, replaced TMCStepper@<1.0.0 with https://github.com/teemuatlut/TMCStepper/archive/upgrade_LPC_SPI.zip

Issue seems coming from TMC5160 communications.

Thanks and "Bon courage"

kAdonis, 2019-05-29:
I just tested the upgrade_LPC_SPI-branch on my printer - Normal boot, no error, stepper operative
CoreXY with TMC5160 for X/Y, SW_SPI on RAMPS/Re-Arm

teemuatlut, 2019-05-29:
@Patag Your board uses SW SPI lines so when configuring HW SPI you wouldn't expect it work.
The TMC5160 and TMC2130 SPI code is the very same code as one is inherited from the other.
Did you ground the CLK pin yet?

@kAdonis Good to hear. Just to make sure, did you power cycle to clear the drivers after uploading the firmware?

kAdonis, 2019-05-29:
Yes, everything is working after power cycle

Patag, 2019-05-29:
@kAdonis Are you sure you deleted  .piolibdeps folder before compilation ? I've been trapped by the lib manager.
@teemuatlut No, CLK still not grounded. I'll try to do it this evening. Another point, E (TMC5160) driver is installed but related stepper in not plugged. Any potential problem ?

kAdonis, 2019-05-29:
@patag I deleted .piolibdeps and .pioenvs folders, I am absolutly sure I am using the right TMC library.
I checked the code, compiled and uploaded twice.
You definitely should connect CLK to Gnd

Patag, 2019-05-29:
Shame on me, I was so stubborn.
CLK is now connected to ground on X and it works like as expected.
Soldering iron is still hot for the 2 others.
Sorry @teemuatlut and @kAdonis for making you waste your time...
Thanks a lot anyway

eightheads, 2019-06-04:
> Shame on me, I was so stubborn.
> CLK is now connected to ground on X and it works like as expected.
> Soldering iron is still hot for the 2 others.
> Sorry @teemuatlut and @kAdonis for making you waste your time...
> Thanks a lot anyway

Did you get your TMC5160s running reliably? I'm having trouble with a set of Bigtreetech ones running the latest Marlin 2.0 code and latest TMCStepper v0.3.5. They are requiring a lot more current than previous TMC drivers(2208 & 2130) on the same printer to get any sort of reliability out of them while running Stealthchop and are still skipping steps occasionally. Fast precise start and stop movements like auto-level probing are harsh and running them without Stealthchop is loud and defeats their purpose. I have tried running them stock, removing the CLK pin completely, and wiring CLK to GND all with the same issues. This is all running on a BIQU SKR v1.3 control board over SPI with TMC_USE_SW_SPI enabled.

teemuatlut, 2019-06-04:
Did you remember to update the RSENSE values?

eightheads, 2019-06-04:
> Did you remember to update the RSENSE values?

Yes, I have RSense set to 0.075

Patag, 2019-06-04:
@eightheads : no, didn't print anything yet ( E stepper still missing to finish assembling). I only tested  moves/homing via LCD menus. So we can't talk about reliability tests.

eightheads, 2019-06-04:
> @eightheads : no, didn't print anything yet ( E stepper still missing to finish printer assembling). I only tested moves/homing via LCD menus. So we can't talk about reliability tests.

I have mine to where they will move via LCD menu and Octoprint ok, but anything faster than around 60mm/s and they will start losing steps, and this is at a lot higher current than with other TMC drivers. 

Current settings with TMC 2130/2208:
X: 800
Y & Z(dual steppers parallel on both): 900

Current for 5160 with any kind of reliability:
X: 1100
Y: 1750
Z: 1250

At these settings the stepper motors do not seem to be running hot yet, but I'm still losing steps. I'm concerned to increase much more than that...

teemuatlut, 2019-06-04:
Post your `M122 VX` and I'll check the calculation.

eightheads, 2019-06-04:
@teemuatlut here is where I left off. I just pulled this:

Send: M122
Recv: 		X	Y	Z
Recv: Enabled		true	true	true
Recv: Set current	1000	1750	1250
Recv: RMS current	990	1744	1244
Recv: MAX current	1396	2459	1754
Recv: Run current	19/31	31/31	25/31
Recv: Hold current	9/31	15/31	12/31
Recv: Global scaler	133/256	146/256	128/256
Recv: CS actual		9/31	15/31	12/31
Recv: PWM scale	41	37	27
Recv: stealthChop	true	true	true
Recv: msteps		16	16	16
Recv: tstep		max	max	max
Recv: pwm
Recv: threshold		98	98	658
Recv: [mm/s]		100	100	3
Recv: OT prewarn	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false
Recv: off time		3	3	3
Recv: blank time	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1
Recv: -start		1	1	1
Recv: Stallguard thrs	0	0	0
Recv: DRVSTATUS	X	Y	Z
Recv: stallguard				X
Recv: sg_result		0	0	0
Recv: fsactive
Recv: stst		X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:09:40:00
Recv: 		Y	0x80:0F:40:00
Recv: 		Z	0x81:0C:40:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK

eightheads, 2019-06-04:
@teemuatlut 
Send: M122 VX
Recv: 		X
Recv: GCONF		0x00:00:00:04
Recv: IHOLD_IRUN	0x00:0A:13:09
Recv: GSTAT		0x00:00:00:05
Recv: IOIN		0x30:00:00:52
Recv: TPOWERDOWN	0x00:00:00:80
Recv: TSTEP		0x00:0F:FF:FF
Recv: TPWMTHRS	0x00:00:00:62
Recv: TCOOLTHRS	0x00:00:00:00
Recv: THIGH		0x00:00:00:00
Recv: CHOPCONF	0x14:00:81:03
Recv: COOLCONF	0x00:00:00:00
Recv: PWMCONF	0x00:05:05:B4
Recv: PWM_SCALE	0x00:00:0F:08
Recv: DRV_STATUS	0x80:09:40:00
Recv: Testing X connection... OK

eightheads, 2019-06-04:
@teemuatlut here are my 5160 setting in config_adv. Note: I upped the current via eeprom to match my M122 above. I'm only running them on XYZ at the moment.

#if HAS_TRINAMIC

  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT       800  // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_MICROSTEPS     16  // 0..256
    #define X_RSENSE      0.075
  #endif

  #if AXIS_IS_TMC(X2)
    #define X2_CURRENT    800
    #define X2_MICROSTEPS  16
    #define X2_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT      1000
    #define Y_MICROSTEPS     16
    #define Y_RSENSE      0.075
  #endif

  #if AXIS_IS_TMC(Y2)
    #define Y2_CURRENT    800
    #define Y2_MICROSTEPS  16
    #define Y2_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT      1000
    #define Z_MICROSTEPS     16
    #define Z_RSENSE      0.075
  #endif

  #if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT    800
    #define Z2_MICROSTEPS  16
    #define Z2_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(Z3)
    #define Z3_CURRENT    800
    #define Z3_MICROSTEPS  16
    #define Z3_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT    800
    #define E0_MICROSTEPS  16
    #define E0_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E1)
    #define E1_CURRENT    800
    #define E1_MICROSTEPS  16
    #define E1_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E2)
    #define E2_CURRENT    800
    #define E2_MICROSTEPS  16
    #define E2_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E3)
    #define E3_CURRENT    800
    #define E3_MICROSTEPS  16
    #define E3_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E4)
    #define E4_CURRENT    800
    #define E4_MICROSTEPS  16
    #define E4_RSENSE    0.11
  #endif

  #if AXIS_IS_TMC(E5)
    #define E5_CURRENT    800
    #define E5_MICROSTEPS  16
    #define E5_RSENSE    0.11
  #endif

teemuatlut, 2019-06-04:
Unfortunately I have to admit that I don't have further ideas right at this time.
The current calculation seems to be spot on and the configuration seems alright.

eightheads, 2019-06-04:
@teemuatlut This is what it is doing in practice unless current settings are a lot higher than other drivers. Note: the same printer setup works great with other TMC drivers and was running TMC2208 via UART  on the same control board before changing out for the TMC5160s.
https://www.youtube.com/watch?v=mbs_DO4Mh4w

Black6spdZ, 2019-06-05:
I don't know if its just my specific combination of hardware but stealthchop absolutely will not work correctly for my machine

eightheads, 2019-06-05:
> I don't know if its just my specific combination of hardware but stealthchop absolutely will not work correctly for my machine. just tried latest build out with same results as before. it is really bad with my two Z steppers in parallel

Is it doing like in my video above when trying to run Stealthchop?

Black6spdZ, 2019-06-05:
yep almost exactly and I just figured it out.. stealthchop is a no go @12v especially with paralleled Z motors. I connected my adjustable lab supply up and ran it up and down while increasing the voltage. at 24v it sill had a little drone but at 26v was practically silent, like a tmc driver should be. the single X and Y motors were silent at about 18v though. well damn, guess I'll be switching out hot-end and fans and pray my bed can be rewired for 24v :(

eightheads, 2019-06-05:
> @eightheads: yep almost exactly and I just figured it out.. stealthchop is a no go @12v especially with paralleled Z motors. I connected my adjustable lab supply up and ran the bed up and down while increasing the voltage. at 24v it sill had a little drone but at 26v was practically silent, like a tmc driver should be. the single X and Y motors were silent at about 18v though. well damn, guess I'll be switching out hot-end and fans and pray my bed can be rewired for 24v :(

I am testing these on a 12v printer currently... However, all other TMC drivers run great at 12v in stealthchop

Black6spdZ, 2019-06-05:
I thought the same because i have some 2130s on another printer running 12v fine, smaller 1.8d motors though

Black6spdZ, 2019-06-07:
got in my new nice fanless 24v 15a psu, bumped it up a little to 26v and it runs smooth and silent now

perikanttila, 2019-07-30:
Even this issue is closed, I had the same kind of issue. I used TMC5160 V1.0 from BIQU. That version does have pin 6 floating, whereas it should be wired to the ground of the stepstick. Also the pin towards the SKR 1.3 board shall be cut. as pins 5 and 6 are shorted in mainboard. When I fixed that - i had no issue with marlin 2.0.x (I used experimetal SQUARE wave steppin optin, and the minimum step was default) now I have X and Y running TMC5160 and Z and E0 with TMC2130.