MarlinFirmware/Marlin #14086: [BUG] (Marlin Bed-Leveling Vs Skew-Correction)

abghrnjd, 2019-05-22:
I've added a G29 command in my starting script to do bed leveling process before every print. It works fine and compensates my bed during prints. But when I print larger models, they're all skewed in both XZ and YZ plane. So I decided to use Marlin's Skew-Correction option.

I printed [Calibration Squares](https://www.thingiverse.com/thing:2563185) and entered measurements for XZ and YZ plane. I left XY corrections as default (Because I have no problem in this plane). Unfortunately it sounds like Bed-Leveling and Skew-Correction don't get along as expected! Many different problems happen such as:

- I get "Probing failed" error on the LCD.
- Z-axis doesn't move (moves for compensating but not for a new layer)
- Bed-Leveling process fails sometimes.
- bed center (X=0,Y=0) gets shifted...
 Is there a bug?

Ludy87, 2019-05-22:
I have no problems with G29 and skew

abghrnjd, 2019-05-24:
I un-commented #define SKEW_CORRECTION . I works fine just for Z skew-Correction.

boelle, 2019-07-21:
so this one if "fixed" ?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle