MarlinFirmware/Marlin #14089: [BUG] MKS SBASE v1.3 E3D Pt100 watchdog reset

aharshac, 2019-05-22:
### Description

I'm trying to use Marlin *bugfix-2.0.x* on MKS SBASE with E3D Pt100 amplifier.

Powering the P100 using 3V3 from the ICSP header.
I've connected P1.31 on EXP1 to the  analog output on the Pt100 amplifier PCB.

### Steps to Reproduce

1. pin_MKS_SBASE.h
    ```
    #define TEMP_0_PIN          5 // P1_31 as ADC input
    ```
2. Configuration.h
    ```
    #define TEMP_SENSOR_0 20
    ```

**Expected behavior:**
* MKS SBASE should boot and load Marlin firmware.
* LED1 (D1) - blink and LED7 (D7) - constant ON.
* On-board SD card should be accessible on PC.

**Actual behavior:**
* MKS SBASE boots up but marlin firmware is not loaded
* LED1 (D1) - **constant ON** and LED7 (D7) - constant ON.
* On-board SD card is not accessible on PC.

### Additional information
I've diagnosed this problem as being caused by `#define TEMP_SENSOR_0 20` as changing it to 3 or 5 does not cause  this issue.

aharshac, 2019-05-22:
Update: Same issue with TEMP_SENSOR_0 as 147 (Pt100 with 4k7 pullup)

InsanityAutomation, 2019-05-24:
Ive got this hardware and can poke a bit this weekend. For a temporary workaround, can try adding #define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE and see if you get correct readings.

boelle, 2019-07-21:
would his config help us?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle