MarlinFirmware/Marlin #14091: [Question] DUE HAL

GMagician, 2019-05-22:
1) In DUE HAL there is a Tone function that seems never called, Am I wrong?
2) DUE define PWM pins from 2 to 13 but looking at schematics it seems more PWM are available. Why these are not listed? Theoretically availability depends on shield mounted not by board itself. Any help to clarify?

Bob-the-Kuhn, 2019-05-22:
It looks like the DUE tone() functions have been replaced by the buzzer code.  I expect they can be deleted.

What documents are you referring to on the PWM pins?

Due's hardware PWM module has 8 blocks that can control pins directly.  Each block has four options as to what hardware pin it controls: Pin_a_active_high,  Pin_b_active_high,  Pin_c_active_low and Pin_d_active_low.  Marlin dedicates most of the 8 blocks to FETs that (usually) drive heaters and fans.  I think the others are don't care and the pins associated with then are used for other purposes.

Yes, the shield ultimately dictates what PWM pin is tied to what function.  That's why each unique shield ahs it's own pins_xxxx.h file.

GMagician, 2019-05-22:
This is from SAM datasheet:

> ̶ 9-channel 32-bit Timer Counter (TC) for capture, compare and PWM mode, Quadrature Decoder Logic and 2-bit Gray Up/Down Counter for Stepper Motor
̶ Up to 8-channel 16-bit PWM (PWMC) with Complementary Output, Fault Input, 12-bit Dead Time Generator
Counter for Motor Control

2 timers are used for stepper and temp, I think...

But looking also to some pinout images on web you can see PWM pins in other header (some in the one used by 12864 full discount LCD)

Bob-the-Kuhn, 2019-05-22:
I'm sorry. I was wrong about DUE's TONE code. It is definitely used.

tone(pin,freq, length) is called from line 71 in buzzer.cpp.

I've closed PR  #14094.

closed 2019-05-22 by Bob-the-Kuhn

GMagician, 2019-05-23:
@Bob-the-Kuhn that's ok. I don't know why searching through the code I didn't found ::tone call