MarlinFirmware/Marlin #14098: [BUG] 1.1.9 & 2.0.x MBL Y-axis moves cause uncoupled stepper to skip

orcinus, 2019-05-23:
### Description

I used to run a fairly old build (1.1.4) because i had various issues with IDEX on 1.1.9 and recently decided to try out 1.1.9 again, as well as 2.0.x.

I had MESH_BED_LEVELING working fine on 1.1.4, but on 1.1.9 and 2.0.x, the Y moves it does between the points are so violent the stepper skips nearly all the steps and just buzzes out. My Y axis mechanics are not the problem, as this happens when i uncouple the stepper from it completely too (freely running stepper). It also happens if i pump the driver current as far as it will go.

I'm currently on 2.0.x (behaviour is the same in 1.9.x) and i'm running MESH_BED_LEVELING (manual, with LCD), on a straightforward IDEX cartesian machine (200x300x300), with TMC2100's, on an Arduino Mega (RUMBA variant).

Amax: 1000, 1000, 100, 4000
A-travel: 1000
A-retract: 3000
Vmax: 150, 150, 6, 30

I've tried with jerk (8, 8, 0.4, 6) and junction deviation (tried 0.2 and 0.5), with and without S-curve. Same result. Could it be a driver timing issue?

### Steps to Reproduce

1. Start MBL
2. Let it home
3. Wait for the first Y move between calibration points

orcinus, 2019-05-23:
A few thoughts and things i haven't ruled out yet:
- It looks like the MBL probing uses XY_PROBE_SPEED for the move speed - it defaults to 8000, perhaps i had it set lower before? Although, 8000 isn't terribly fast, and i've definitely done moves that are as fast or faster... Odd.
- Maybe the stepper is busted, will try a different stepper.
- Maybe the driver is busted, will try swapping X and Y.

orcinus, 2019-05-24:
Scratch that first one - it was always 8000.

orcinus, 2019-05-24:
Replaced stepper (same make and model) - same issue.
Replaced driver - same issue.

I'm out of ideas...

orcinus, 2019-05-26:
After further checking... it looks like since 1.1.9/2.0.x i cannot go higher than about 6500 on the feedrate or the 0.9 deg steppers simply stall / jerk back/forth. I could swear i used to be able to before, with no problem.

Am i going crazy?

Did the max frequency for AVR drop in 1.1.9/2.0.x?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

orcinus, 2019-08-04:
Forgot to follow up here…

I haven’t solved it, and i’m still limited to 6000 or the 0.9 degree stepper skips steps. It looks like the max pulsing frequency dropped considerably in 1.1.9 / 2.0.x and it’s unable to perform moves as fast as before on 8-bit hardware, with 0.9 degree steppers at 32x microstepping.