MarlinFirmware/Marlin #14103: [BUG] Mixed TMC TMC2130 and TMC2208

Ludy87, 2019-05-23:
Could there be a problem if the method is not called?

@teemuatlut or @thinkyhead 

```c++
Marlin\src\feature\tmc_util.cpp:143:23: warning: 'get_pwm_scale' defined but not used [-Wunused-function]
static uint32_t get_pwm_scale(TMC2208Stepper &st) { return st.pwm_scale_sum(); }
^
Marlin\src\feature\tmc_util.cpp:146:28: warning: 'get_driver_data' defined but not used [-Wunused-function]
static TMC_driver_data get_driver_data(TMC2208Stepper &st) {
^
```

teemuatlut, 2019-05-23:
No the compiler will just optimize the whole function away.
I've already addressed this in some commit in some branch and I can no longer find the commit nor the branch. But it's there, somewhere.

Ludy87, 2019-05-23:
Okay, I'm unsure. I was asked in a Whatsapp group if mixing is possible when the warung is issued.

teemuatlut, 2019-05-28:
@Ludy87 I'm trying to follow up on my recent items. Has this been fixed for you now?

Ludy87, 2019-05-28:
I have to ask tomorrow or compile myself.

Ludy87, 2019-05-29:
thx @teemuatlut 
no more warning message, thanks for your work

closed 2019-05-29 by Ludy87