MarlinFirmware/Marlin #14105: [BUG] Marlin 2.0 BF Commit-5/20-SDCard issues when compiling

Stady234, 2019-05-23:
### Description

Compile error while enabled SDCard options.

### Steps to Reproduce

SDCARD enabled.. 

```
C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp: In static member function 'static void GcodeSuite::M20()':

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:53:22: error: 'port' was not declared in this scope

   SERIAL_ECHOLNPGM_P(port, MSG_BEGIN_FILE_LIST);

                      ^

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:53:47: error: 'SERIAL_ECHOLNPGM_P' was not declared in this scope

   SERIAL_ECHOLNPGM_P(port, MSG_BEGIN_FILE_LIST);

                                               ^

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp: In static member function 'static void GcodeSuite::M27()':

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:154:22: error: 'port' was not declared in this scope

     SERIAL_ECHOPGM_P(port, "Current file: ");

                      ^

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:154:44: error: 'SERIAL_ECHOPGM_P' was not declared in this scope

     SERIAL_ECHOPGM_P(port, "Current file: ");

                                            ^

C:\Users\Brandon\AppData\Local\Temp\arduino_build_233203\sketch\src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp:169:10: error: 'class CardReader' has no member named 'getStatus'

     card.getStatus(

          ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1

```
**Expected behavior:** No error

**Actual behavior:** error

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

https://drive.google.com/file/d/10sepyM0gEJbl-p_CUCnjcZ47l3YgGdiG/view?usp=sharing

Ludy87, 2019-05-24:
```M20-M30_M32-M34_M524_M928.cpp``` That file is not in the current 2.0.x.

Pull the new version from bugfix-2.0.x!

boelle, 2019-07-21:
@Stady234 tried what @Ludy87 suggested?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle