MarlinFirmware/Marlin #14106: [BUG] Helpful website references offline

jellewie, 2019-05-24:
### Description

Linked to websites that don't exist (anymore)

### Steps to Reproduce

Click on the links (listed all below)

**Expected behavior:**

Taken to a website that has content

**Actual behavior:**

404 error or no website

#### Additional Information

**Marlin_main.cpp**
37; [http://linuxcnc.org/handbook/gcode/g-code.html](http://linuxcnc.org/handbook/gcode/g-code.html)
38; [http://objects.reprap.org/wiki/Mendel_User_Manual:_RepRapGCodes](http://objects.reprap.org/wiki/Mendel_User_Manual:_RepRapGCodes)

**SdInfo.h**
43; [http://www.sdcard.org/developers/tech/sdcard/pls/simplified_specs](http://www.sdcard.org/developers/tech/sdcard/pls/simplified_specs)

Even the **Code of conduct linked** in a bug report has a 404?
[https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md](https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md)

**Version**
1.1.x downloaded a few days again, but the same problem on the current GitHub files on the website

Ludy87, 2019-05-24:
http://marlinfw.org/

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

jellewie, 2019-10-21:
Just as a heads up, this still isn'' fixed. This bug is unfixable by just 'waiting'. Anyone should have known if the have read it. 
Although I can understand the not fixing it part. The code of conduct linked in the bug report template should really be fixed