MarlinFirmware/Marlin #14111: [BUG] SKR v1.3 - fan slows down/stops when heater is on

040medien, 2019-05-24:
I just upgraded my printer with an SKR v1.3. I noticed that any time my single hotend is heating up, the hotend fan is turning off. When it almost reached temperature, it's turning off again, but running at a slower speed. Because of this, I'm suffering from heat creep.

I'm **not** using ADAPTIVE_FAN_SLOWING
I am using E0_AUTO_FAN_PIN

While I can't put my finger on it, I think that the part fan is also running at a slower speed.

Please see configs for additional context:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3218589/Configs.zip)

jjansen85, 2019-05-24:
I guess you connected the fan to the HE1 terminals?
I use the same setup, but did not experience this problem.

040medien, 2019-05-25:
That's right. Would you mind sharing your configs?

gloomyandy, 2019-05-25:
I have a similar setup which operates with no problems. My config is here:
https://github.com/gloomyandy/Marlin/tree/myskr1.3/Marlin

But what you describe sounds like a wiring problem or possibly an issue with your power supply. How do you have things wired? What voltage fans are you using? Does running the heated bed impact the fan speeds at all?

040medien, 2019-05-25:
Thank you Andy, that was the right hint! I am using two breakout boards between the board and the hotend connected with a ribbon cable.  Turns out they are using a common ground. It seems the terminal labels for the heater were flipped. Once I swapped the leads for the heater, everything worked fine!

closed 2019-05-25 by 040medien