MarlinFirmware/Marlin #14112: TMC Connection Error - TMC2130, MKS GEN V1.4

Garyr14, 2019-05-24:
I'm running MKS GEN V1.4 with 2130's (V1.0), Bugfix 2.0x. Been fine since June, 18. Last update was in November. Tried updating to the latest version a few days ago, but all I'm getting is TMC Connection Error.

Tried comparing configuring.h and configuration_adv.h to working versions, but I can't figure out what's wrong. I assume this has something to do with pin assignments and/or SPI? What changed in the past few months that may be causing this? Thanks for your help.

teemuatlut, 2019-05-25:
Plenty has changed back and forth in the past few weeks. Make sure to use the latest version of the library. Version 0.3.5 was released yesterday and should hit the library managers within a day or two.

Ludy87, 2019-05-25:
June, 18.? 
---

no problems with MKS Gen v1.4 and Watterott TMC2130

Garyr14, 2019-05-25:
Sorry - June 22, 2018

Garyr14, 2019-05-25:
For what it's worth, I just flashed the latest version of bugfix 2.0.x, using the 0.3.5 version of the library.  I did not set the drivers to TMC2130.  I was able to get the board to boot but of course, M122 returned "unknown command".  BLTouch did seem to work.

I then activated the TMC2130's on the X, Y, Z, E0 and E1 motors and uploaded, but get the TMC Connection Error again.  

All pins were left as the defaults.  Should i be making any changes?

Ludy87, 2019-05-25:
send your configuring.h and configuration_adv.h

Garyr14, 2019-05-25:
Attached - Thanks!

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3219623/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3219627/Configuration.h.txt)

Ludy87, 2019-05-26:
You are using an older version. I suggest you an update before and the improvement of the configuration.h

Then try again.

----
change:

```c++
//#define X_DRIVER_TYPE  TMC2130
//#define Y_DRIVER_TYPE  TMC2130
//#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE TMC2130
//#define E1_DRIVER_TYPE TMC2130
```

to

```c++
#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130
#define E1_DRIVER_TYPE TMC2130
```

Garyr14, 2019-05-26:
Thanks, but I only commented those lines out for testing.  When I activate those lines, I get the connection error.

One thing I just noticed is that I get about 26 error messages during compile (when I activate the TMC2103's as you indicate above): 

Error while detecting   libraries included by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\feature\babystep.cpp
--
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\feature\tmc_util.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\bedlevel\abl\G29.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\calibrate\G28.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\control\M17_M18_M84.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\control\M226.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\control\M400.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\control\M80_M81.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\feature\advance\M900.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\feature\trinamic\M569.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\feature\trinamic\M906.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\feature\trinamic\M911-M914.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\geometry\G92.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\host\M114.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\lcd\M0_M1.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\gcode\motion\G4.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\lcd\menu\menu_main.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\lcd\menu\menu_tmc.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\Marlin.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\configuration_store.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\endstops.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\motion.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\planner.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\stepper.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\stepper_indirection.cpp
Error while detecting libraries included   by   C:\Users\Garyr\AppData\Local\Temp\arduino_build_829557\sketch\src\module\temperature.cpp

The compile still works, but when I flash the board, I get the TMC Connection Error.

teemuatlut, 2019-05-26:
Remove the `.pioenvs` and `.piolibdeps` folders to clean you environment and try again.

Garyr14, 2019-05-26:
Done.  I even re-installed the Arduino IDE.  Now I'm getting the same errors plus:

C:\Users\Garyr\AppData\Local\Temp\arduino_build_491481\sketch\src\module\planner.cpp: In function 'set_machine_position_mm':

C:\Users\Garyr\AppData\Local\Temp\arduino_build_491481\sketch\src\module\planner.cpp:2765:1: internal compiler error: Segmentation fault

Which stops the compilation.

Commenting out the "#define X_DRIVER_TYPE  TMC2130" lines results in a clean compilation.

teemuatlut, 2019-05-26:
The folders are PIO specific and aren't used by A IDE, so sure enough that wouldn't have any effect.

You can try building with PlatformIO  or cleaning the Arduino IDE build folder.
The errors you're seeing aren't coding issues in either Marlin or the associated libraries. Sometimes it also helps to hammer the compile button a few times more...

Garyr14, 2019-05-26:
Thanks again.  I just realized i installed the windows app instead of PlatformIO.  I removed the app and re-installed platform IO and re-compiled.  Fine w/o the TMC2130's, same errors ("Errors while detecting libraries....") with the TMC20130's.  I'll keep trying...

Garyr14, 2019-05-26:
Below is the output after uploading.  Does anything not look right?

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\Garyr\AppData\Local\Temp\arduino_build_279903/Marlin.ino.hex"
avrdude: writing flash (162518 bytes):

Writing | ################################################## | 100% 40.61s

avrdude: 162518 bytes of flash written
avrdude: verifying flash memory against C:\Users\Garyr\AppData\Local\Temp\arduino_build_279903/Marlin.ino.hex:
avrdude: load data flash data from input file C:\Users\Garyr\AppData\Local\Temp\arduino_build_279903/Marlin.ino.hex:
avrdude: input file C:\Users\Garyr\AppData\Local\Temp\arduino_build_279903/Marlin.ino.hex contains 162518 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 35.52s

avrdude: verifying ...
avrdude: 162518 bytes of flash verified

avrdude done.  Thank you.

teemuatlut, 2019-05-28:
@Garyr14 You can close the issue if this problem has been resolved by updating to the latest bugfix and the latest library version.

Garyr14, 2019-05-28:
Unfortunately, it has not been resolved for me yet, as i still get the "TMC Connection Error".  I noticed several discussions about soldering pins, rewiring, etc.  I have done none of this.  My TMC2130's did work fine with the Bugfix 2.0.x build from several months ago.  I should also note that they are Fysetc, not Waterott.  I'm reluctant to switch as they have been fine and I've never had problems with them running too hot (usually 28 C).

mersadk, 2019-06-01:
I had a same issue with MKS GEN V1.1. LCD said "TMC Connection Error", while in console I was getting "Testing Y connection... Error: All LOW". 
When I disabled LCD it started working. 
I had a look at board schematics and it looks like pin D49 (Y_CS_PIN) is used by LCD.
Which pin should be safe to remap Y_CS_PIN to?

mersadk, 2019-06-01:
Small update, it still doesn't work as it should, but when LCD is disabled it doesn't report any error.
One stepper motor moves slower than the other one.

Garyr14, 2019-06-01:
As a troubleshooting test, if you comment out the lines which invoke the TMC2130’s in configuration.h, does it give you the error?

mersadk, 2019-06-02:
@Garyr14 there is not much to comment out in configuration.h, I can only switch from TMC2130 to some other drivers.

I was able to make it work yesterday, but I'm not sure what exactly helped.
I've freshly cloned Marlin repository (bugfix-2 branch) and configured it as I would normally (corexy, steps....). Only special thing I did is in configuration_adv.h file, under TMC2130 section, I've overwritten X_CS_PIN and Y_CS_PIN to pins 42 and 44, and it started working (make sure to fix wiring as well).
Also I've noticed that if I connect printer to PC with USB cable without turning on power supply, I will get "TMC Connection Error" message on display.

```
  /**
   * Override default SPI pins for TMC2130, TMC2160, TMC2660, TMC5130 and TMC5160 drivers here.
   * The default pins can be found in your board's pins file.
   */
  #define X_CS_PIN          42
  #define Y_CS_PIN          44
```

Garyr14, 2019-06-10:
Hello,
I wanted to provide an update, in case others have a similar issue.  After much research, I realized (duh) that I never properly configured my  Fysetc drivers.  All i did originally was insert them and set marlin to use them.  The older builds (pre December, 2018) evidently did not do a full check, which is why (I assume) they did not give me the TMC Connection Error prior to me trying to flash with the latest build. 

Instead of trying to solder pins on the Fysetc v1.0, I replaced them with the BIGTREETECH V2.0 and properly wired them to the appropriate pins (after some troubleshooting!).  I was then able to successfully flash Marlin and not get the TMC Connection Error.  Furthermore, I'm now hearing what stealthchop sounds like.  Evidently, i was never in stealthchop mode before, even though m122 reported so.  WOW - What a difference!  Everything is working great.  

My only potential problem is that the stepper motors, especially the Y axis motor, seem to be running hot (north of 70C).  I have active cooling on the drivers, so they're o.k.  I'm only concerned about the actual motors.  I assume lowering the vRef will help or do i lower the current, which is currently set at 900.  BTW, microsteps are set to 16, rsense at .11 for all axis' and extruders.

Thanks again for everyone's help.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle