MarlinFirmware/Marlin #14114: [BUG] TMC2130 at 256 microsteps incorrect step/mm during homing

damienmg, 2019-05-25:
### Description

When upping the number of microsteps to 256, the homing speed result in doing a safe homing at around 1/8th of the actual position (i.e. X is at around 8mm instead of 156). The board believe it is at the actual position.

Moving the AXIS directly with G1 commands result in correct movement.

### Steps to Reproduce

1. Configure Marlin 2.0.x-bugfix with a Bl-Touch or other probe and Z_SAFE_HOMING.
2. Switch from 16 to 256 microsteps for TMC2130: https://github.com/damienmg/Marlin/commit/71a1f2ed3c5bef8ae7eda2560875e769d819ab36. Interpolation does not matter apparently.

3. Send `G28`

**Expected behavior:** Home X and Y then move back to the center of the bed to home Z

**Actual behavior:** Home X and Y then move back to 8mm in X and Y the bed to home Z. The board believe it is in the middle of the bed.

Also doing a G29 do X and Y movement very slow too.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
See this branch: https://github.com/damienmg/Marlin/tree/dmg-ender3-skr13-256-microsteps

* Hardware: Ender 3. SKR 1.3. TMC2130

* Provide pictures or links to videos that clearly demonstrate the issue.
https://photos.app.goo.gl/jkgusK3HaYTjQooM8

teemuatlut, 2019-05-25:
Do not use 256 microsteps even on 32bit boards. That would be pushing 1.3 million steps per second at 100mm/s.
Also make sure to use the latest version of the library. Version 0.3.5 was released yesterday.

doggyfan, 2019-05-25:
You **can** use 256 micro-stepping (not that it'll be accurate without providing a calibrated micro-stepping curve table for **each** stepper) **if** you go more slowly (for whatever reason, ie CNC machining).

damienmg, 2019-05-25:
Ok I was just testing to understand the difference. 1.3 million steps per second does not seems that much to me but I am more used to develop on backend machine :)

The problem also happen at 64 microsteps. I'll try 32.

I had v0.3.4 installed this morning:
```
$ platformio lib update TMCStepper
Library Storage: /home/dmg/Documents/3DPRINTING/Firmwares/Marlin/.piolibdeps
Updating TMCStepper                      @ 0.3.4          [Up-to-date]
```

I'll try to figure out how to update to 0.3.4. To be fair this is just an experiment for me to see what it change and be able to explain it.

teemuatlut, 2019-05-25:
1.3MHz is not much if you don't have to do anything else. Also these are not PLCs or FPGAs =)
1/64th is something I've been thinking myself about using but haven't done the change yet.

You can pull `v0.3.5` before PIO updates the lib manager if you use a direct link in `platformio.ini`:
https://github.com/teemuatlut/TMCStepper/archive/v0.3.5.zip

gloomyandy, 2019-05-25:
Looks like you have EEPROM enabled did you remember to reset your settings back to the default after this change? So M502 then M500.

damienmg, 2019-05-25:
@gloomyandy: yes I did.

@teemuatlut: I just tried with 64 microsteps and 0.3.5 for TMCStepper, still same issue (in addition to having the homing sensitivity becoming impossible to correctly tune.

teemuatlut, 2019-05-25:
If you could post your `M122 X` and `M122 XV` and `M503`. I'd like to see if the bits are set correctly.

damienmg, 2019-05-25:
I just started a print so it will have to wait some hours. I'll do that next!

damienmg, 2019-05-26:
Strangely today the 64 microsteps seems to work fine. Maybe overheating? Anyway, here the report for both 256 and 64 microsteps.

[reports.64msteps.txt](https://github.com/MarlinFirmware/Marlin/files/3220063/reports.64msteps.txt)
[reports.256msteps.txt](https://github.com/MarlinFirmware/Marlin/files/3220064/reports.256msteps.txt)

I'll try a print with 64 microsteps and report here.

damienmg, 2019-05-26:
So with 64 microsteps it seems to go allright now, so maybe it is only happening at faster speed. The card fan is also not fixed at the moment so it could have moved and made the driver overheat.

damienmg, 2019-05-31:
Confirmed after several print that 64 musteps is fine as well as going to slow feedrate in 256 musteps. I'll just close this bug as being an issue with putting too much on the board / steppers.

closed 2019-05-31 by damienmg