MarlinFirmware/Marlin #14119: Re-Arm+RAMP1.6Plus NOT WORKİNG MARLİN 2.0

orcunim, 2019-05-25:
Hello,

İ use TFT28. My printer pause, start, pause, start for 1-2 layer. İ upload new version. İ check it, all settings true.
İ open
#define RepRap Discount Smart Controller
But platformio give yellow error on upload
[image]
[image]

Ludy87, 2019-05-25:
Unuse Warning no error

orcunim, 2019-05-25:
But TFT28 dont working. what should I do?

Ludy87, 2019-05-25:
TFT28 has its own controller maybe bad command is given.

orcunim, 2019-05-25:
How can I fix ?  not currently active.  temperatures do not appear.  moving home is not working going home.  not connected briefly.

orcunim, 2019-05-25:
Problem : Static electric //

closed 2019-05-25 by orcunim