MarlinFirmware/Marlin #14120: Ender3-pro usb input broken, How can I communicate with mainboard?

sgsalman, 2019-05-25:
Hello there, 

My micro usb input is broken on mainboard. I need to connect my machine to computer to set the bed level and firmware update, so I can't now. Can I communicate by ISP to usb as I did before with micro usb?  I bought the bltouch v3 kit, so I have Isb to Usb part. Can I use this "ISP to USB" to communicate btw. comp and printer?

best regards.

reloxx13, 2019-05-26:
search for "flash without bootloader"

stop asking questions here.

sgsalman, 2019-05-26:
@reloxx13 you guys act like everyone know everythings. I don't know something and I'm asking.

I aldready booted my printer, I just can't communicate with printer. thanks for info...

scippo97, 2019-05-26:
it literally say when you submit a bug report

"if you have any questions or need help use discord. this is ONLY for bugs"

reloxx13, 2019-05-26:
and u guys act like this is a help forum.

sgsalman, 2019-05-26:
@scippo97 okey u right, could u give me discord link, I couldnt find it.

closed 2019-05-26 by sgsalman