MarlinFirmware/Marlin #14121: [BUG] cannot compile TMC2130 for LPC1768

Arthedian, 2019-05-25:
### Description

In the past I successfully compiled Marlin 2 for bigtreetech skr 1.1. Today i downloaded New Marlin 2 from the webpage, but sadly it gives me compilation error.

### Steps to Reproduce

1. Download Marlin 2 from official webpage
2. copy Ender 3 setting
3. set the bigtreetech skr 1.1
4. I have TMC 2208 in standalone on X,Y,Z and LV8729 on E0

**Expected behavior:**

successfull compilation

**Actual behavior:** 

> Compiling .pioenvs/LPC1768/libb6b/TMCStepper_ID5513/source/TMC2130Stepper.cpp.o
> .piolibdeps/TMCStepper_ID5513/src/source/SW_SPI.cpp: In member function 'void SW_SPIClass::init()':
> .piolibdeps/TMCStepper_ID5513/src/source/SW_SPI.cpp:14:57: error: cannot convert 'volatile LPC_GPIO_TypeDef' to 'fastio_reg {aka volatile LPC_GPIO_TypeDef*}' in assignment
> mosi_register = gpio_port(LPC1768_PIN_PORT(mosi_pin));
> ^
> .piolibdeps/TMCStepper_ID5513/src/source/SW_SPI.cpp:15:57: error: cannot convert 'volatile LPC_GPIO_TypeDef' to 'fastio_reg {aka volatile LPC_GPIO_TypeDef*}' in assignment
> miso_register = gpio_port(LPC1768_PIN_PORT(miso_pin));
> ^
> .piolibdeps/TMCStepper_ID5513/src/source/SW_SPI.cpp:16:55: error: cannot convert 'volatile LPC_GPIO_TypeDef' to 'fastio_reg {aka volatile LPC_GPIO_TypeDef*}' in assignment
> sck_register = gpio_port(LPC1768_PIN_PORT(sck_pin));
> ^
> *** [.pioenvs/LPC1768/libb6b/TMCStepper_ID5513/source/SW_SPI.cpp.o] Error 1
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3219583/Configuration.zip)

gloomyandy, 2019-05-25:
Make sure that your TMCStepper library is up to date (latest version is 0.3.5) see this post for how to force an update if needed: https://github.com/MarlinFirmware/Marlin/issues/14114#issuecomment-495902787

All of those errors are in the TMCStepper library so if you are still having problems @teemuatlut may be able to help further.

Oh and make sure you are using the official version of Marlin not the one from Bigtreetech.

teemuatlut, 2019-05-25:
Update your LPC framework.

Arthedian, 2019-05-25:
I tryied downloading current version from github. Now I have different problem: 

> Dependency Graph
> |-- <Servo> 1.0.0

> Archiving .pioenvs/LPC1768/lib368/libTMCStepper.a
> .piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
> .piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:153:3: error: 'noInterrupts' was not declared in this scope
> noInterrupts(); // Need 100% focus on instruction timing
> ^~~~~~~~~~~~
> .piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:153:3: note: suggested alternative: 'detachInterrupt'
> noInterrupts(); // Need 100% focus on instruction timing
> ^~~~~~~~~~~~
> detachInterrupt
> .piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:2037:3: error: 'interrupts' was not declared in this scope
> interrupts();
> ^~~~~~~~~~
> .piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:2037:3: note: suggested alternative: 'intptr_t'
> interrupts();
> ^~~~~~~~~~
> intptr_t
> Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_emac.c.o
> Indexing .pioenvs/LPC1768/lib368/libTMCStepper.a
> Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_exti.c.o
> *** [.pioenvs/LPC1768/lib59f/Adafruit NeoPixel/Adafruit_NeoPixel.cpp.o] Error 1
> > |-- <LiquidCrystal> 1.0.0
> > |-- <U8glib-HAL> 0.4
> > |-- <TMCStepper> 0.3.5
> > |-- <Adafruit NeoPixel> 1.1.8

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle