MarlinFirmware/Marlin #14124: Error: Checksum Mismatch, Last Line: xx

orcunim, 2019-05-25:
Hello,

İ use re-arm, ramps 1.6+ and TFT28. İ upload last version but i started printer. the printer stops responding, and then resumes. İ try so much method in forum. İ print on pc. All great. After that copy sd card take tft. Touch print always pause start and error in screen.

[image]

orcunim, 2019-05-26:
Help please :(

SharePARTS, 2019-06-20:
I am suffering from the same problem.
I use marlin 2.0 firmware and  mks sgen board and mks tft 3.5.

boelle, 2019-09-24:
@orcunim or @SharePARTS 

Configurations, please
Please ZIP up your Configuration.h and Configuration_adv.h files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

dodoguigui, 2019-10-09:
[Checksum Mistmatch issue Marlin  2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3708035/Checksum.Mistmatch.issue.Marlin.2.0.x.zip)
Hello,

I have the same problem. It's for Ender 3 with MKS Gen-L, MKS TFT 28, BlTouch and TMC 2208 in UART mode.

Please find Configuration.h and Configuration_adv.h file

Regards

AnHardt, 2019-10-09:
Would not surprise me if the software serials for the TMC 2208 would block interrupts long enough to miss the interrupt requests from the hardware serial.

boelle, 2019-10-12:
@orcunim still having issues?

boelle, 2019-11-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-03 by boelle

gokulvenkat, 2019-12-10:
I too have this issue did anyone found the solution

Krzysiekzg, 2019-12-27:
Hi to all
for me in jgaurora a5 on skr 1.3 and marlin 2.0 also displayed error checksum mismatch last line, additionally the printer stopped randomly for 1-3 seconds spoiling the printout. this only occurred when printing from a USB connected with a cable to the mks tft 2.8 display. printing from a computer was ok

I think it's all because of the noise on the cables. I led the USB cable differently to which I connect the pendrive and the problem disappeared.

You can also create a protective screen of aluminum foil and insulation. it's worth checking it helped me out.