MarlinFirmware/Marlin #14127: [BUG] (TMC Driver Parameter mismatch M913 vs. LCD )

Silverpumax, 2019-05-25:
Hi all,
Thanks in advance for your great (and probably hard) work.
Here is my issue

MKS GEN L  with  MKS Mini LCD 12864
TMC5160 using SPI on X
Marlin bugfix 2.0 from May, 19
TMCStepper version=0.3.5

### Description
**Issue 1.)**
The LCD shows other/old values as set by M913.
After setting a value with the LCD --> M913 shows the new value correct.

**Issue 2.)**
A value of 0 with M913 is ignored by marlin.

**Issue 3.)**
A value of 0 set via LCD --> M913 shows the new value correct.
(A value of 50 set via LCD --> M913 shows the new value correct.)

### Steps to Reproduce

**Issue 1.)**
M913 X50
M913 --> X stealthChop max speed: 50 
M122 --> threshold 202 / 49.71  [mm/s]		
LCD  Hybrid threshold --> X: 101       **>>> Bug**

**Issue 2.)**
M913 X0
M913 --> X stealthChop max speed: 50         **>>> Bug**
M122 --> threshold 202 / 49.71  [mm/s]	**>>> Bug**
LCD  Hybrid threshold --> X: 101                   **>>> Bug**

**Issue 3.)**
LCD  Hybrid threshold set to  X:  0                 
M913 --> X stealthChop max speed: 0           **>>> Ok**
M122 --> threshold 65535 / 0.15  [mm/s]	**>>> Ok ?**

**Expected behavior:** [What you expect to happen]

For only using stealth shop i want set Hybrid threshold to zero with M913
(shutoff  Hybrid threshold)

**Actual behavior:** [What actually happens]

Can not  set threshold to zero( as descript in TMC data sheet) for stealth shop only mode.

#### Additional Information

teemuatlut, 2019-05-26:
A patch to the TMC LCD variables was merged in right around the time you posted this.
Try updating to the very latest commit and see if it solves the issues.

thisiskeithb, 2019-05-26:
I just ran into a similar issue with the latest TMC Stepper libs (manually specified 1.3.5) and the latest Marlin bugfix-2.0.x commit (df77334).

Some terminal output:

```
Send: M913
X stealthChop max speed: 201
Y stealthChop max speed: 201
Z stealthChop max speed: 50
E stealthChop max speed: 238
ok
[...]
Send: M913 X200 Y200 E200
ok
[...]
Send: M500
echo:Settings Stored (666 bytes; crc 34105)
ok
[...]
Send: M913
X stealthChop max speed: 201
Y stealthChop max speed: 201
Z stealthChop max speed: 50
E stealthChop max speed: 238
ok
```

We're keeping you busy @teemuatlut :)

teemuatlut, 2019-05-26:
That is likely just be because there are conversions from one unit to another and back again.

Silverpumax, 2019-05-26:
I am tryed updating to the very latest commit.
I see LCD issue is resolved.

**But (sub) issue 2 is still present !**
**I can not set Hybrid threshold to zero with M913**
Try
M913 X0
evaluate
**M913 nor M122 nor LCD shows the correct value.** (It shows the old/last value set)
With the LCD however i can set the Hybrid threshold value to zero (eg.65535)
Try
M913 or M122 after setting the value zero with the LCD.
 :)
**PS:**
In the TMC Datasheet ,for example TMC2130 on mid page 44,  i found the sentence:
"Set TPWMTHRS zero if you want to work with stealthChop only. "
Why set You in this case TPWMTHRS to 65535 (0xFFFF) ?

teemuatlut, 2019-05-26:
Good to hear the LCD is working properly now.
I'll take a look at the M913 issue when I can. There are a few things that take priority for now.

teemuatlut, 2019-05-27:
I looked into the `M913 X0` issue and it's there but I'm not sure if I'll address it in the first place.

First I need to point out that `TPWMTHRS` and `M913` work in different units.
With `M913` you command in millimeters per second while
`TPWMTHRS` is in units of time (clocks) between two interpolated step pulses.
Basically these two are the inverse of each other (also depends on microsteps and steps per millimeter).
So asking for a very low number like 0 would indeed result in a really big step interval.

The correct configuration for using just stealthChop is to enable stealthChop options and disable Hybrid Threshold.
On the other hand if you want to just use spreadCycle, you should disable all stealthChop options. Then the Hybrid Threshold doesn't have any effect either so it doesn't matter if it's enabled or disabled.

At this time having have to enable Hybrid Threshold to avoid a compile error is a special circumstance, but in normal operation there are more appropriate ways of configuring just one mode.

Silverpumax, 2019-05-27:
Thanks for this answer and  thanks for spending a lot of (Your) time for this issue.

**I am fully satisfied if i can set `M913 X0` like i can setting the Hybrid threshold to zero with the LCD.**
**Both should be give the same result --> threshold 65535  / [mm/s] -**   >> seen with M122

With this way i have three Modes for **external control**.
```
Mode A:   M569 S1 X  with M913 X0    -->  Stealthchop only
Mode B:   M569 S1 X  with M913 X20   -->  Hybrid
Mode C:   M569 S0 X                  -->  Spreadcycle
```
The tests with this three modes give good results for me !
However i am miss only to set  `M913 X0`

I know ,it is not easy to fully understand the TMC datasheet.
I am only at the begining to understand something !
_So i give my very great respect for Your work !_

teemuatlut, 2019-05-28:
Speeds below the `M913` speed parameter will be executed in stealthChop if stealthChop is enabled.
Speed above the parameter will be executed in spreadCycle.

So your mode A if you wanted to use just stealthChop by modifying Hybrid Threshold values, should be to set a **high** value.
```
Mode A:   M569 S1 X  with M913 X1000    -->  Stealthchop only
```

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle