MarlinFirmware/Marlin #14128: Tool change not updating the software endstops

mikeshub, 2019-05-26:
### Description

On tool change software endstops are not updated. This allows the carriage to be driven past the physical extents of the machine.

### Steps to Reproduce

1. Home the axis with the offset
2. Switch to the other tool head
3. Use G0 to go to the software endstop position. This will result in the carriage driving into the machine.

**Expected behavior:**

The machine shouldn't ever crash the carriage

**Actual behavior:**

Angry stepper motor noises.

#### Additional Information

This post is FYI. I have fixed this issue and included it with an already open PR that tweaks some double head nozzle settings for the Creator Pro. 

https://github.com/mikeshub/Marlin/commit/7fa9b7c3595467a5a863ed448ef800e3887eea8c

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

mikeshub, 2019-07-21:
Sorry. Closing was the right action. I meant to when this got fixed a little while back.