MarlinFirmware/Marlin #14129: UBL on coreXY z axis continuing to lower (increase z)

mpieroway, 2019-05-26:
**Description**

When executing the mesh test or running UBL on my corexy at each point the table lowers (increase in z) and continues at each measurement point until hitting max z. Using current 1.1.9 Marlin on a MKS Gen2Z V1.2 with an inductive probe. I saw a thread here indicating there was a missing parantheis in the planner.cpp, though when I checked the indicated line, the code was correct.

 **Steps to Reproduce**

run UBL through LCD or through g codes

**Expected behavior:** 

Expect the UBL to run and allow me to level the build table

**Actual behavior:**

build table continues to lower (increase z) without the z adjusting to the sensor.

[configuration h.txt](https://github.com/MarlinFirmware/Marlin/files/3219963/configuration.h.txt)
[configuration adv h.txt](https://github.com/MarlinFirmware/Marlin/files/3219964/configuration.adv.h.txt)

eikeime, 2019-05-26:
U might need to invert probe endstop.
In your case change
`#define Z_MIN_ENDSTOP_INVERTING true`
to
`#define Z_MIN_ENDSTOP_INVERTING false`

mpieroway, 2019-05-26:
I will give it a shot and let you know

mpieroway, 2019-05-26:
@eikeime I made the change and now at homing the z increases (table lowers) 10mm and doesn't complete the home move.

mpieroway, 2019-05-26:
I was reading issue #12725 and noticed that on some boards its a memory issue that causes the mesh to fail. When looking at my compiler I see that my sketch uses 183990 bytes of 253952 (72%). I commented out SDSUPPORT and it still fails on the first mesh point on build cold mesh. Any thoughts?

mpieroway, 2019-05-27:
Not a memory issue, I tried the steps in #12725 and ended up returning to previous settings. in both instances when running G29 P1 the printer homes and moves to the first point on the mesh and then gets caught in a probe loop. The LCD reads probe failed, and in order to stop the probing loop I have to reset the printer or kill power.

Roxy-3D, 2019-05-27:
> and noticed that on some boards its a memory issue that causes the mesh to fail. When looking at my compiler I see that my sketch uses 183990 bytes of 253952 (72%). I commented out SDSUPPORT and it still fails on the first mesh point on build cold mesh. Any thoughts?

It isn't program memory that is important.   It is the static RAM memory where you can run into problems.   You should have at least 1800 bytes of variable memory reported by Arduino.

Also...   are you running current code?   Two or three weeks ago a bug was fixed that sounds similar to what you are seeing.

mpieroway, 2019-05-27:
@Roxy-3D I downloaded the current one from github on 19 May to rebuild my firmware.

I do have sufficient memory

Roxy-3D, 2019-05-27:
You have sufficient program space.   I have not seen any numbers about variable space.     Please confirm you have at least 1800 bytes of variable memory.    

Also...  one more question:   When you do a G29 T    do you have the entire mesh filled in values?  Or do you have some dots in the mesh?

mpieroway, 2019-05-27:
From the compiler - Global variables use 3613 bytes (44%) of dynamic memory, leaving 4579 bytes for local variables. Maximum is 8192 bytes.

G29 T

    ( 15,185)                      (185,185)
        0       1       2       3       4      
 4 |  0.000   0.000   0.000   0.000   0.000 
   |
 3 |  0.000   0.000   0.000   0.000   0.000 
   |
 2 |  0.000   0.000   0.000   0.000   0.000 
   |
 1 |  0.000   0.000   0.000   0.000   0.000 
   |
 0 |[ 0.000]  0.000   0.000   0.000   0.000 
        0       1       2       3       4      
    ( 15, 15)                      (185, 15)

Roxy-3D, 2019-05-27:
OK!   Thank You!     

Obviously....   You shouldn't be seeing the problems you are seeing.    Let's try to narrow down what could be going wrong.   I'm not suggesting this as a permanent solution...   But you have:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 2  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -20// Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
```
Let's set your MESH_INSET to 25 so we know both the probe and nozzle can reach every place on the bed.     Does a G29 P1 still fail?

mpieroway, 2019-05-27:
@Roxy-3D  Before I make that change..... the head comes forward from the safe home position about 15-20mm for the first probe point.

mpieroway, 2019-05-27:
@Roxy-3D I changed the MESH INSET and results are the same with the probe loop. table does not stop at sensor flag, but sensor flag stops the table on z home

Roxy-3D, 2019-05-27:
> before I make that change..... the head comes forward from the safe home position about 15-20mm for the first probe point.  Is that normal from what you see?

That is normal.   Where ever you home, the odds are that is not located at a mesh point.    When it starts probing the bed to generate the mesh, it will go to the closest mesh point.   It will continue doing that, expanding outwards.    But it is very unlikely your first mesh point is where you homed.

> I do apologize, I am learning coding through 3d printing rather than having back ground knowledge to start with. lol

Not a problem...    But Marlin firmware isn't going to be the kindest, most easy way to learn about software.

> I changed the MESH INSET and results are the same with the probe loop. table does not stop at sensor flag, but sensor flag stops the table on z home

@gloomyandy   Any ideas what we should try next?

mpieroway, 2019-05-27:
@Roxy-3D I appreciate the warning, I have some good friends locally that code frequently who have been excellent tutors. Though I have to have taken on the challenge of the corexy with my printer. lol.

Thank you for your time and assistance

mpieroway, 2019-05-30:
@Roxy-3D I kept searching and using different key words in my search parameters and found a solution that seems to be working. At the very least UBL cold mesh has run without failing. 

I found the solution to the probe failing in #7784. 

> You need to disable
//#define Z_MIN_PROBE_ENDSTOP
and enable
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

I made the changes in marlin re uploaded and all seems well. 

Thank you again for your assistance!

closed 2019-05-30 by mpieroway