MarlinFirmware/Marlin #14133: [BUG] Z2 axis compile error

doggyfan, 2019-05-26:
Compile error for dual Z axis ..

```
In file included from Marlin\src\module/stepper.h:226,
from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper_indirection.h:286:24: error: 'Z2_AXIS' was not declared in this scope
extern TMC_CLASS(Z2) stepperZ2;
^~
Marlin\src\module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
Marlin\src\module/stepper_indirection.h:286:24: note: suggested alternative: 'E2_AXIS'
extern TMC_CLASS(Z2) stepperZ2;
^~
Marlin\src\module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
Marlin\src\module/stepper_indirection.h:65:79: error: template argument 4 is invalid
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\module/stepper_indirection.h:66:40: note: in expansion of macro '____TMC_CLASS'
#define ___TMC_CLASS(MODEL, A, I, E) ____TMC_CLASS(MODEL, A, I, E)
^~~~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
In file included from Marlin\src\feature\../module/stepper.h:226,
from Marlin\src\feature\babystep.cpp:30:
Marlin\src\feature\../module/stepper_indirection.h:286:24: error: 'Z2_AXIS' was not declared in this scope

extern TMC_CLASS(Z2) stepperZ2;
^~
Marlin\src\feature\../module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\feature\../module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:286:24: note: suggested alternative: 'E2_AXIS'
extern TMC_CLASS(Z2) stepperZ2;
^~
Marlin\src\feature\../module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\feature\../module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:65:79: error: template argument 4 is invalid
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\feature\../module/stepper_indirection.h:66:40: note: in expansion of macro '____TMC_CLASS'
#define ___TMC_CLASS(MODEL, A, I, E) ____TMC_CLASS(MODEL, A, I, E)
^~~~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^~~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^~~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^~~~~~~~~~
Marlin\src\feature\../module/stepper_indirection.h:286:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(Z2) stepperZ2;
^~~~~~~~~
*** [.pioenvs\LPC1768\src\src\feature\babystep.cpp.o] Error 1
*** [.pioenvs\LPC1768\src\src\Marlin.cpp.o] Error 1
```

teemuatlut, 2019-05-26:
Follow up PR coming soon.

DkOuvE, 2019-05-26:
Just pulled the latest and now having the same issue.

boucz, 2019-05-27:
@teemuatlut do you know when your PR will be merged? (your corrections works for me)
Thanks!

teemuatlut, 2019-05-27:
Whenever @thinkyhead or @Roxy-3D finds the time to push the button =)

boucz, 2019-05-27:
ok, thanks! ;)

lokus77, 2019-05-27:
Dual Z, LPC1768. Same error. Thanks a lot @teemuatlut

closed 2019-05-28 by doggyfan