MarlinFirmware/Marlin #14136: SD card NOT read / Mounted SKR1.3

jarjartee, 2019-05-26:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-05-26:
Try enabling `#define LPC_SD_ONBOARD`.

jarjartee, 2019-05-26:
like this :
// LPC-based boards have on-board SD Card options. Override here or defaults apply.
  #ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    //#define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      #define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif

#endif // SDSUPPORT

jarjartee, 2019-05-26:
going to try to do with PSU connected - suspect to low USB power from my mac

thisiskeithb, 2019-05-26:
Don't forget the changes to the serial ports in `Configuration.h`.

hudja, 2019-05-28:
same here. I cannot connect with Mac, nor I can see my SD card. I can connect and see SD card with original SKR1.3 firmware. I am using the same settings as suggested above. 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

#define USB_SD_ONBOARD

jarjartee, 2019-05-28:
Tried various combos USB cables / SD cards but suspect some incompatibility between SKR board / firmware / MacOS. No offence to BigTreeTech (theyre doing great work) but communication / customer support beyond hardware failures is difficult. Having said that I suppose SKR 1.3 is fairly ‘bleeding edge’ until bugs are quashed - hoping BigTree will come up with a fix soon

hudja, 2019-05-29:
I flashed latest stock Marlin 2.0 Configuration.h and Configuration_adv.h yesterday (not the ones inside config/examples/Creality/Ender-3 folder), and communication and SD card worked fine with on Mac. I then started to introduce step-by-step changes from Ender-3 pre-configured Configuration.h file. I lose the configuration when changing any of the following lines of config (from/to):

#define HEATER_0_MAXTEMP 275 to #define HEATER_0_MAXTEMP 255
#define BED_MAXTEMP      150 to #define BED_MAXTEMP      110
//#define X_DRIVER_TYPE  TMC2208 to #define X_DRIVER_TYPE  TMC2208
Changing other drivers also kills the communication. However, when I specify #define X_DRIVER_TYPE  TMC2208_STANDALONE communication and SD card works fine.

I am flashing the bare SKR 1.3 board without any external components connected. The board is connected to Mac by USB. I tried plugging in the UART TMC2208 driver into X, when specifying #define X_DRIVER_TYPE  TMC2208 - I still cannot talk to the board with driver installed.

None of the changes in Configuration_adv.h Ender-3 file affected the communication.

I will try to connect external power to the board today and some components, to see if it changes anything.

hudja, 2019-05-29:
So, it turns out that it has smth to do with **#define TEMP_SENSOR_BED** 1 setting. If I use it in TMC2208 UART mode, then it throws me an MAX BED temp error and does not boot (that is why not comm and SD card). I can boot and comm using dummy temperature reading with **#define TEMP_SENSOR_BED 998**. If I use TMC2208 _STANDALONE, then it works fine with **#define TEMP_SENSOR_BED 1**  also.

hudja, 2019-05-29:
Update. If I increase #define BED_MAXTEMP to 275, then I am able to boot and comm with #define TEMP_SENSOR_BED 1 and #define X_DRIVER_TYPE TMC2208. It seems to me that during the boot it reads the temperature too high for a second, and kicks in the protection. After the boot, termistor readings are correct (at least they show room temp).

hudja, 2019-05-29:
Turned out it got fixed yesterday! Just get the latest marlin. Now I can comm with #define TEMP_SENSOR_BED 1 and #define X_DRIVER_TYPE TMC2208

https://github.com/MarlinFirmware/Marlin/issues/14139

Genosystem, 2019-06-01:
I still have the problem today.

aaltman, 2019-07-09:
I'm stuck here right now.  I think I also have invalid serial settings.  How can I work around that to flash the board?  I have an Arduino if that could work.

thinkyhead, 2019-07-11:
Some boards may benefit from the new setting…
```cpp
#define THERMAL_PROTECTION_GRACE_PERIOD 1000
```

boelle, 2019-09-24:
@jarjartee is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle