MarlinFirmware/Marlin #14139: [BUG] Marlin halts immediately after turning on with Err: MAXTEMP Bed

scippo97, 2019-05-26:
###Printer

Ender 3 with SKR V1.3, TMC2208 on XYZE in UART mode, stock CR10 Display

### Description

When you turn on the printer with a firmware compiled after commit 48d1bd29c causes the following issue:

Printer starts, Marlin logo comes in. Buzzer starts buzzing intermittently.

Buzzer now starts buzzing constantly, the following message appears on the LCD:

Err: MAXTEMP BED

PRINTER HALTED

Please reset

### Steps to Reproduce

1) Build a marlin firmware on any commit after 48d1bd29c
2) Flash firmware and turn on the printer

**Expected behavior:** Printer turns on

**Actual behavior:** Printer halts

#### Additional Information

This happens regardless of configuration.h and / or configuration_adv.h you're using. The stock ender 3 configuration.h and configuration_adv.h were used, the only mods to it being defining the TMC2208 drivers.

Reverting Temperature.cpp fixes the problem altogether.

Strahler85, 2019-05-26:
I have the same issue after pull latest changes

thisiskeithb, 2019-05-26:
I just ran into the same issue with this board on an Ender-3 after killing a print with the power switch and turning it back on.

Using files from the df77334  commit.

rst00, 2019-05-26:
I have the same issue. Reverted back to pre 48d1bd2

doggyfan, 2019-05-26:
The thermistor code is currently going through a bit of a change to fix the thermal protection when using user thermistors.

Normal service will be resumed (I hope) today sometime.

UnderXirox, 2019-05-28:
hello with the mks sbase also it bug
Sur machine cr10s 

avec 
#define TEMP_SENSOR_0 5     // E3DV6
#define TEMP_SENSOR_BED 5   // Sensors Stock Creality3D

and DRV8825

mais avec 
// Bipass bug termistance
//#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE

la machine indique les bonne temps aussi bien buse que bed

Pilson66, 2019-05-28:
Have the same issue with SKR 1.3

doggyfan, 2019-05-28:
For now, to bypass that error, you can add/uncomment this in your pins file ..

```cpp
// Ignore temp readings during develepment.
#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE
```

govido, 2019-05-28:
This 'solution' seems to be working for now.
The temperature is at 52°C for the fraction of a second, and then drops to 26°C.
Is it possible that the board measures way higher even before the lcd starts?
[image]
[image]

doggyfan, 2019-05-28:
Trying commenting out that line again and try increasing the grace period from 1000ms to say 4000ms ..

So with ..

```cpp
// Ignore temp readings during develepment.
//#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE
```

grace period setting is in **Marlin\src\HAL\HAL_LPC1768\HAL.h** ..
```cpp
// A grace period to allow ADC readings to stabilize, preventing false alarms
#define THERMAL_PROTECTION_GRACE_PERIOD 4000
```

Patag, 2019-05-28:
Works as it should now.  Thanks @doggyfan !

doggyfan, 2019-05-28:
So the grace period was too short then ( 1 second) ?

You're now using a longer grace period (4 seconds or so) ?

doggyfan, 2019-05-28:
DON'T leave **BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE** uncommented, it's ONLY meant for testing with.

Patag, 2019-05-28:
No worries @doggyfan. The printer is still under assembling and testing. It has never seen any piece of filament yet :)
Anyway, I performed a test with THERMAL_PROTECTION_GRACE_PERIOD set to 4000,  BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE disabled, a bed preheating to 80°C and a short (about 10 secs) cooldown via LCD then a manual RST. The reboot went strait to MAXTEMP BED/PRINTER HALTED.
It looks like the bed temp threshold triggering the halt is wrong a initialization (Only a feeling, I'm a noob in 3D printing and even more in Marlin programming :))

govido, 2019-05-28:
So... I tested 2000, 3000, 4000 and even 8000ms the error is still present.
Any other ideas?
Or is there something in platformio I have to do to get the change compiled correctly?
I already deleted the .pioenvs folder...

doggyfan, 2019-05-28:
In that case, I think the problem _might_ that their is no grace period applied to the min/max testing, but not sure.

Try this temperature.cpp. It has the grace period applied to the min/max testing. Make sure that you have **BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE** disabled.

[Temperature_test.zip](https://github.com/MarlinFirmware/Marlin/files/3229133/Temperature_test.zip)

govido, 2019-05-28:
This works 👍 
I left the HAL.h at default 1000ms and changed the temperature.cpp.
What is different in your version?

doggyfan, 2019-05-28:
I simply added a grace period to the min/max temperature testing, which just means the min/max is ignored for the grace period at boot-up.

doggyfan, 2019-05-28:
OK, I've put in a pull request to add the change into Marlin.

doggyfan, 2019-05-28:
OK, the fix has been done.

Try the current Marlin release. If it's fixed it (should have), then you can close this @scippo97.

scippo97, 2019-05-29:
I'll try today and let you know

UnderXirox, 2019-05-29:
Ok thx 
Merci pour le correctif probleme regle pour moi aussi
 thx

scippo97, 2019-05-29:
Ok, can confirm new changes properly fix this bug on SKR V1.3, Can close this issue.

Genosystem, 2019-06-01:
I've the bug today in commit e4260a4
Using BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE works

doggyfan, 2019-06-02:
@Genosystem Can you post your config files ?

I'm actually very surprised that the LPC1768 boards are the only boards to need the grace period, because their ADC system is no different to any other board. I don't get why the 8-bit boards don't need the grace period too, it doesn't make sense, unless their is a bug in the code on those boards ??

Genosystem, 2019-06-02:
I've attached the configs.
I can see that when I turn on the printer, temperatures drop from 37-45 degrees to room temperature. 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3244647/Config.zip)

doggyfan, 2019-06-02:
@Genosystem try increasing the grace period time length to say 4000 (4 seconds).

So in the **Marlin\src\HAL\HAL_LPC1768\HAL.h** file, set the grace period ..
```cpp
#define THERMAL_PROTECTION_GRACE_PERIOD 4000
```

See if that helps.

Genosystem, 2019-06-02:
Yeah, I edited the message above to put that.

doggyfan, 2019-06-02:
So a longer grace period worked OK then ?

doggyfan, 2019-06-02:
That's OK then.

Try some shorter than 4000 grace period values (1500, 2000 etc) if you can, see how low you can go without Marlin throwing a temperature error at boot-up.

Genosystem, 2019-06-02:
Hello
3000: OK
2000 OK
1500: OK
1000: FAIL

You cannot reopen this issue? More users have the problem in others forums

doggyfan, 2019-06-02:
I can't reopen the issue, but looks like it's fixed. Use double the lowest value that worked for you.

So 3000 or so looks like a nice safe value to use for probably everyone.

scippo97, 2019-06-02:
Reopened by request, however I can't see why a grace period of 4000 is such a bother, my printer works flawlessly after the patch

doggyfan, 2019-06-02:
I think you can re-close the issue @scippo97, it's very much fixed I think, just needed a slightly longer grace period for some people is all.

Genosystem, 2019-06-02:
When has fixed? I'dont see any commits of this

doggyfan, 2019-06-02:
@Genosystem you're right of coarse,, though I do have a pull request in (3000 grace period for all hardware types) for it to be committed to Marlin, which hopefully will go through.

closed 2019-06-02 by scippo97