MarlinFirmware/Marlin #14140: [BUG] Values Cutoff with USE_BIG_EDIT_FONT Enabled

thisiskeithb, 2019-05-26:
### Description

Enabling `USE_BIG_EDIT_FONT` causes values to be cut off:
[image]

### Steps to Reproduce

Grab the df77334 bugfix-2.0.x commit, and enable `USE_BIG_EDIT_FONT`.

I recompiled the 24b1818 commit from a few of days ago and the text is back to normal:

[image]

**Expected behavior:** See full values when editing through the "big edit font" screens.

**Actual behavior:** Values are cut off.

**Configs:** [Ender-3 - SKR 1.3 - 5-25-19.zip](https://github.com/MarlinFirmware/Marlin/files/3220415/Ender-3.-.SKR.1.3.-.5-25-19.zip)

thisiskeithb, 2019-05-27:
I tested the commits in between the two I had tested previously and the bug was introduced in 44caf70.

Patag, 2019-06-05:
In Marlin\src\lcd\dogm\ultralcd_DOGM.cpp, I just reverted to :
`        lcd_moveto(((lcd_chr_fit - 1) - (vallen + 1)) * one_chr_width, baseline); // Right-justified, leaving padded by spaces`
from :
`       lcd_moveto((lcd_chr_fit - (vallen + 1)) * one_chr_width, baseline); // Right-justified, leaving padded by spaces`
It is not tested on all possible "big edit font" screens,, but it seems to be running fine on most usual of them.

closed 2019-06-10 by thinkyhead