MarlinFirmware/Marlin #14143: [Feature Request] Add confirmation screen before aborting print

EgeSS, 2019-05-26:
I accidentally pressed Abort Print instead of Pause Print (12864 GLCD)during a 40-hour print.
I literally wasted about 400 grams of filament because of this. I think it would be nice if there was an option to add a confirmation screen before aborting the print such as the system in Ultimaker printers. This can happen to anyone and it's very frustrating.

reloxx13, 2019-05-26:
imho uneccessary waste of flash (cause strings costs alot of bytes).

pay more attention what u do next time.

LoganFraser, 2019-05-26:
An option if your print gets cancelled part way through is to measure the height of the print, lower the print into the bed by that amount in your slicer (object z offset), continue printing, and then glue the two pieces together. I've done this after a nozzle clog and a power outage to reasonably decent results.

forkoz, 2019-05-26:
You must not have updated recently because this feature is already implemented. Also for start of print from SD.

InsanityAutomation, 2019-05-26:
This is already there. I added it several months ago :)

EgeSS, 2019-05-27:
I'm currently running Marlin 1.1.9 release. When was this released?

EgeSS, 2019-05-27:
Sorry for my lack of knowledge since I'm new to 3D printing but what do I need to uncomment in order to enable this feature?

ManuelMcLure, 2019-05-27:
This is only in bugfix-2.0.x. 1.1.x is no longer getting any new features and only getting critical bug fixes, so if you want this you'll have to make the jump to bugfix-2.0.x.

EgeSS, 2019-05-27:
Ok then, thanks for your help!

closed 2019-05-27 by EgeSS

alakond, 2019-07-09:
> This is already there. I added it several months ago :)

How to disable it? I do not need it

thinkyhead, 2019-07-11:
> How to disable it? I do not need it

```diff
-     MENU_ITEM(submenu, MSG_STOP_PRINT, menu_abort_confirm);
+     MENU_ITEM(submenu, MSG_STOP_PRINT, []{ ui.abort_print() ; ui.goto_previous_screen(); });
```

alakond, 2019-07-12:
Magic! its work!  Thnx!