MarlinFirmware/Marlin #14144: [BUG] warnings: Marlin/src/sd/cardreader.cpp: In static member function 'static void CardReader::presort()':

sl1pkn07, 2019-05-26:
### Description

warnings about build cardreader.cpp

### Steps to Reproduce

1. [First Step]
download marlin 2.0 from git
2. [Second Step]
edit the default configuration scripts, you can apply this commit https://github.com/sl1pkn07/Marlin/commit/7441c1500e02ebffb35e3f3a080cf218633ba44e with the all changes 
3. [and so on...]
start build with Atom+PlatformIO (linux), all up to date

**Expected behavior:** [What you expect to happen]

build without warnings (warning about external components is alloweb because is external)

**Actual behavior:** [What actually happens]
~~~
Compiling .pioenvs/LPC1768/lib8c6/TMCStepper/source/SW_MODE.cpp.o
Marlin/src/sd/cardreader.cpp: In static member function 'static void CardReader::presort()':
Marlin/src/sd/cardreader.cpp:765:72: warning: array subscript 27 is above array bounds of 'char [27]' [-Warray-bounds]
                                     sortnames[I][SORTED_LONGNAME_MAXLEN] = '\0'; }while(0)
                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin/src/sd/cardreader.cpp:942:11: note: in expansion of macro 'SET_SORTNAME'
           SET_SORTNAME(0);
           ^~~~~~~~~~~~
Marlin/src/sd/cardreader.cpp:765:72: warning: array subscript 27 is above array bounds of 'char [27]' [-Warray-bounds]
                                     sortnames[I][SORTED_LONGNAME_MAXLEN] = '\0'; }while(0)
                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin/src/sd/cardreader.cpp:852:13: note: in expansion of macro 'SET_SORTNAME'
             SET_SORTNAME(i);
             ^~~~~~~~~~~~
~~~

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
https://github.com/sl1pkn07/Marlin/commit/7441c1500e02ebffb35e3f3a080cf218633ba44e
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

chrisqwertz, 2019-05-26:
Looks like an off by one error to my. I think that correct way would to replace `SORTED_LONGNAME_MAXLEN` with `SORTED_LONGNAME_MAXLEN - 1` in the places where the errors are thrown

robbycandra, 2019-05-27:
Fixed with PR #14148

sl1pkn07, 2019-05-28:
hi

yep, fixed by #14148 

greetigs and closing

closed 2019-05-28 by sl1pkn07