MarlinFirmware/Marlin #14145: [BUG] Problem compiling TMC support with undefined HYBRID_THRESHOLD

oechslein, 2019-05-26:
### Description

When compiling with TMC support 
`#define X_DRIVER_TYPE  TMC2208`
but disabled HYBRID_THRESHOLD
`//#define HYBRID_THRESHOLD`

I'm getting the following (Error message from compiler):
```
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC2208Stepper, 'X', '0', (AxisEnum)0u>]':
Marlin\src\feature\tmc_util.cpp:655:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC2208Stepper, 'X', '0', (AxisEnum)0u>' has no member named 'get_pwm_thrs'
uint32_t tpwmthrs_val = st.get_pwm_thrs();
^
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC2208Stepper, 'Y', '0', (AxisEnum)1u>]':
Marlin\src\feature\tmc_util.cpp:664:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC2208Stepper, 'Y', '0', (AxisEnum)1u>' has no member named 'get_pwm_thrs'
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC2208Stepper, 'Z', '0', (AxisEnum)2u>]':
Marlin\src\feature\tmc_util.cpp:673:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC2208Stepper, 'Z', '0', (AxisEnum)2u>' has no member named 'get_pwm_thrs'
```

### Steps to Reproduce

See above it is straight forward (branch bugfix-2.0.x / df7733425cda02f7e88e7eb9496be72a17d335c1)

#### Additional Information

Problem was introduced with #14074:
_tmc_status calls get_pwm_thrs, but get_pwm_thrs is only defined when HYBRID_THRESHOLD is enabled.

Workaround is to enable HYBRID_THRESHOLD  and set limits to very high values.

teemuatlut, 2019-05-26:
https://github.com/MarlinFirmware/Marlin/pull/14134

oechslein, 2019-05-26:
Already covert in PR

closed 2019-05-26 by oechslein