MarlinFirmware/Marlin #14151: [BUG] Compile error with RADDS & TMC5160

akurz42, 2019-05-27:
### Description

Current 2.0 Bugfix won't compile for RADDS with TMC5160 Stepper Drivers. Snapshot from 22nd May works with the exact same config. These are the errors:
```
Compiling .pioenvs\DUE\src\src\gcode\calibrate\G28.cpp.o
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC5160Stepper, 'X', '0', (AxisEnum)0u>]':
Marlin\src\feature\tmc_util.cpp:655:31:   required from here

Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC5160Stepper, 'X', '0', (AxisEnum)0u>' has no member named 'get_pwm_thrs'
           uint32_t tpwmthrs_val = st.get_pwm_thrs();
                                                   ^
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC5160Stepper, 'Y', '0', (AxisEnum)1u>]':
Marlin\src\feature\tmc_util.cpp:664:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC5160Stepper, 'Y', '0', (AxisEnum)1u>' has no member named 'get_pwm_thrs'
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC5160Stepper, 'Z', '0', (AxisEnum)2u>]':
Marlin\src\feature\tmc_util.cpp:673:31:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC5160Stepper, 'Z', '0', (AxisEnum)2u>' has no member named 'get_pwm_thrs'
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC5160Stepper, 'E', '0', (AxisEnum)3u>]':
Marlin\src\feature\tmc_util.cpp:685:32:   required from here
Marlin\src\feature\tmc_util.cpp:579:51: error: 'class TMCMarlin<TMC5160Stepper, 'E', '0', (AxisEnum)3u>' has no member named 'get_pwm_thrs'
Compiling .pioenvs\DUE\src\src\gcode\calibrate\G33.cpp.o

scons: *** [.pioenvs\DUE\src\src\feature\tmc_util.cpp.o] Error 1
 [ERROR] Took 24.45 seconds
```

### Steps to Reproduce

1. Update configs with values from working version (22/05/19)
2. PlatformIO PIO Build
3. --> Error

**Expected behavior:** Compiles like with older versions.

**Actual behavior:** Compile Error as above.

teemuatlut, 2019-05-27:
See https://github.com/MarlinFirmware/Marlin/pull/14134

akurz42, 2019-05-27:
Thanks! I'd never have seen the connection between the 3 issues...

Cheers, 
Alex

akurz42, 2019-05-27:
I can also confirm that commenting TMC_DEBUG did the trick.

closed 2019-05-27 by akurz42