MarlinFirmware/Marlin #14152: [BUG] Re-ARM failed send commands over USB. conection is OK and printer is ON

sl1pkn07, 2019-05-27:
### Description

i have issue whit the usb connection over MacOS (the pc more near at printer) (and presumably, yoth other OS)

i have configured the unit like the wiki say (http://marlinfw.org/docs/basics/install_rearm.html)

when open the pronterface and i try connect with the unit. the connection is OK. but any command inserted is recognized as Unknown, include the internal commands launched from pronterface iself

the printer works iself over the LCD

components: 
Original Re-ARM
RAMS 1.4 Premium from StaticBoards
TMC2130 over SPI from StaticBoards
Stock Creality ender 3 LCD (+ custom cable following the pins_rearm file)
stock thermisor and endstop from Ender 3 printer
24V power supply
Trianglelabs BLtouch (connected to servo 0 and wired the +5v probe pin from rearm J2 +5v pin)

### Steps to Reproduce

1. [First Step]
Download Marlin code from GIT (df77334. not tested with recently commits)

2. [Second Step]
apply this configuration https://github.com/sl1pkn07/Marlin/commit/7441c1500e02ebffb35e3f3a080cf218633ba44e (or clone the repository/branch (sL1-Changes_ReARM))

3. [and so on...]
build and upload to the rearm. and reboot the unit. using Atom+Platformio (linux)
4. []
connect the unit, with the shield and all components, include thermisor, drivers, endstops, etc to a USB to PC
5. []
Open the pronterface and try to connect

**Expected behavior:** [What you expect to happen]

connect and send/received commands it without problem

**Actual behavior:** [What actually happens]

Connection is OK, but when the protenface send her own commands, or send commands by hand, say " Unknown command". if for example send a "G29" over LCD and finish the leveling, is possible see the output, but not all times

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

https://github.com/sl1pkn07/Marlin/commit/7441c1500e02ebffb35e3f3a080cf218633ba44e

* Provide pictures or links to videos that clearly demonstrate the issue.

![Captura de pantalla 2019-05-25 a las 22 13 11](https://user-images.githubusercontent.com/462213/58374182-54db4e00-7f3a-11e9-8c78-1c8f0a4ecd6e.png

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sl1pkn07, 2019-05-27:
https://github.com/MarlinFirmware/Marlin/issues/10781#issuecomment-496229575

same issue if set `TEMP_SENSOR_BED` to 0

sl1pkn07, 2019-05-28:
@syurev, with "the board hung again after I changed TEMP_SENSOR_BED from 0 to 1." you mean halt by `max_temp_bed` error?

i'm not sure why, but i raise this error today :/

sl1pkn07, 2019-05-28:
the halt problem is fixed with https://github.com/MarlinFirmware/Marlin/pull/14167

the main problem persist

testing a dongle Bluetooth HC-06 connected, pin to pin (tx and rx crossed), in the port J4, configure the baudrate to `115200` bauds, and enable `SERIAL_PORT_2` to 0, i can able connect throught pronterface with the printer over BT

[image]

if try the same with the USB:

[image]

Bob-the-Kuhn, 2019-05-28:
I'm not making any headway on my project so I thought I'd see if I could be of help.

Here's what I did:
1. Downloaded your commit, compiled it & uploaded to my Re-ARM. **RESULT:**  USB port came up for about a second and then hung.  No coms over USB.
2. Downloaded latest 2.0.x, set motherboard to Re-ARM, port to -1, compiled, copied to SD, moved SD to Re-ARM and rebooted.  **RESULT:** same as before (no comms over USB)
3. Set port 2 to 0, ... **RESULT:**  same as before (no comms over USB)
4. Enabled SDSUPPORT and RepRap full graphics LCD, … **RESULT:** Everything works fine.
5. Removed port 2, SDSUPPORT and RepRap full graphics LCD, …  **RESULT:** Connects just fine over USB.
6. Dropped your configuration files in, ..  **RESULT:**  SYSTEM hung (turns out it was because of bed over temperature error due to nothing attached to the thermal inputs).
7. Switched LCD to my RepRap full graphics LCD - now I could see the hang was because of the bed over temperature error.
8. Set bed and extruder to 998.  **RESULT:** SD card is mapped to PC as a drive and have normal USB communications.

Strange stuff.

I'm guessing that, at least on a Windows 10 PC, once the PC sees a port problem it'll continue to fail it until ???.  After that a good USB connection always comes up.

Have you tried your config with the very latest 2.0.x?

Bob-the-Kuhn, 2019-05-28:
One theory: They just fixed a bug that corrected an out of bounds array subscript.  This type of problem has been known to cause all kinds of strange behavior with different symptoms depending on the options enabled.

sl1pkn07, 2019-05-28:
yes, i test with it. my latest comment is with it (not exactly, is 0f04ba016fbd5397428125330b9caba6fe9d02ee + my changes + this PR https://github.com/MarlinFirmware/Marlin/pull/14167 applied)

Bob-the-Kuhn, 2019-05-30:
PR #14167 was merged a day ago.  Are you now fully functional with **today's** 2.0.x?

syurev, 2019-05-30:
For mine it work just fine for now.

sl1pkn07, 2019-05-31:
febase ith latest commit https://github.com/MarlinFirmware/Marlin/commit/9d9030a39c747da92adc2e0a9847a63e7bb8eaa6 same issue in Macos Mojave. but in windows 10 (with laptop) works without problem

i'm not sure if is a problem of the apple driver or what...

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sl1pkn07, 2019-07-21:
The issue still here....

boelle, 2019-09-24:
@sl1pkn07 is the issue still there?

boelle, 2019-10-12:
@sl1pkn07 is your mac up again?

sl1pkn07, 2019-10-12:
nope. and idk when i can power on again

anyway, i close the issue, i have connect the printer over wifi with the ESP01 module, not need connect over usb anymore, so...

closed 2019-10-12 by sl1pkn07