MarlinFirmware/Marlin #14154: SKR v1.3 with TMC 2130 compilation error

biskero, 2019-05-27:
Hello,

I am compiling Marlin bugfix-2.0.x

 for SKR v1.3 with TMC 2130, dual extruder
getting errors during compilation:

`Compiling .pioenvs/LPC1768/src/src/gcode/bedlevel/mbl/M421.cpp.o
Marlin/src/feature/tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_debug_enum) [with TMC = TMCMarlin<TMC2130Stepper, 'X', '0', (AxisEnum)0>]':
Marlin/src/feature/tmc_util.cpp:655:31:   required from here
Marlin/src/feature/tmc_util.cpp:579:38: error: 'class TMCMarlin<TMC2130Stepper, 'X', '0', (AxisEnum)0>' has no member named 'get_pwm_thrs'; did you mean 'en_pwm_mode'?
uint32_t tpwmthrs_val = st.get_pwm_thrs();`

There are other but are the same for Y, Z, E1, E2
Any suggestions?

thx

teemuatlut, 2019-05-27:
See https://github.com/MarlinFirmware/Marlin/pull/14134

biskero, 2019-05-27:
thanks a lot, commenting TMC_DEBUG worked but I got another error:

`Marlin/src/module/stepper_indirection.cpp:161:86: error: conflicting declaration 'TMCMarlin<TMC2130Stepper, 'E', '1', (AxisEnum)3> stepperE1'
#define __TMC_SPI_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_CS_PIN, ST##_RSENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)`

any idea on this ?
thx

boelle, 2019-07-21:
@biskero so this one is "fixed" ?

biskero, 2019-07-21:
@boelle I think so!

boelle, 2019-07-21:
oki, then i will do a close, you can always ask for a reopen should you find that its not fixed

closed 2019-07-21 by boelle