MarlinFirmware/Marlin #14155: [BUG] Marlin 2.0.x: SKR v1.3 with TMC 2130 Dual Extruder compilation error

biskero, 2019-05-27:
Hello,

I am compiling Marlin bugfix-2.0.x for SKR v1.3 with TMC 2130, dual extruder E0 and E1,
getting errors during compilation:

`Marlin/src/module/stepper_indirection.cpp:161:86: error: conflicting declaration 'TMCMarlin<TMC2130Stepper, 'E', '1', (AxisEnum)3> stepperE1' #define __TMC_SPI_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_CS_PIN, ST##_RSENSE, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)`

any idea on this ?
thx

teemuatlut, 2019-05-27:
Please include your config files for testing.

biskero, 2019-05-27:
Here are the config files
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3224798/config.zip)

thx

mikeshub, 2019-05-27:
Config files aren't needed here. A recent update to some macros broke a number of things. This also broke double extruders for TMC2208. Here is the problem:

stepper_indirection.cpp line 175

`
  #if ENABLED(DISTINCT_E_FACTORS) && E_STEPPERS > 1
    #define TMC_SPI_DEFINE_E(AI) TMC_SPI_DEFINE(E##AI, E##AI)
    #define TMC_UART_DEFINE_E(SWHW, AI) TMC_UART_DEFINE(SWHW, E##AI, E##AI)
  #else
    #define TMC_SPI_DEFINE_E(AI) TMC_SPI_DEFINE(E##AI, E)
    #define TMC_UART_DEFINE_E(SWHW, AI) TMC_UART_DEFINE(SWHW, E##AI, E)
  #endif
`

is incompatible with this marco in stepper_indirection.h lin 69

`
  #define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
  #define ___TMC_CLASS(MODEL, A, I, E) ____TMC_CLASS(MODEL, A, I, E)
  #define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
  #define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
  #define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)

`

The macro in the .h expands like this:

_TMC_CLASS(E1_DRIVER_TYPE, TMC_E1_LABEL, E1_AXIS)

It can only give a declaration in the form of 

TMC_UART_DEFINE(SW, E1, E1)

Whereas the macro in the .cpp line 59 is giving a declaration in the form of

TMC_UART_DEFINE(SW, E1, E)

So one of the two defines has to be changed.

biskero, 2019-05-27:
Hello,
from the error seems something like that.
Ok, so I will wait for a bug fix.
Any idea if someone is working on it?
thx, and great support !

mikeshub, 2019-05-27:
I am working on it.

teemuatlut, 2019-05-27:
Fix added to https://github.com/MarlinFirmware/Marlin/pull/14134. Just waiting for Travis checks to complete.

mikeshub, 2019-05-27:
Cool, I'll stop working on it then.

biskero, 2019-05-27:
I can test it when ready, thx again!

mikeshub, 2019-05-27:
That fixed it.

biskero, 2019-05-27:
ok so I just need to download the marlin again?
thx

mikeshub, 2019-05-27:
I was feeling impatient so I copy pasted from here:

https://raw.githubusercontent.com/MarlinFirmware/Marlin/2fcf495881c79cc284694404e93086c44c83e02d/Marlin/src/module/stepper_indirection.h

Otherwise once the commit is pushed all the way you can use the desktop client or command line to get the latest code. I recommend against downloading the zip file. It is much better to use the source control tools.

biskero, 2019-05-27:
I will wait and use the source code tool, thx!

teemuatlut, 2019-05-30:
@biskero You can close the issue if the problem has been fixed in the recent changes to the firmware.

biskero, 2019-05-30:
great, I compiled yesterday and worked!
thx a lot for the support!

closed 2019-05-30 by biskero