MarlinFirmware/Marlin #14157: TMC2130 drivers detected but not moving

FilipKernan, 2019-05-27:
### Description

I recently upgraded to tmc2130s for my three axes on my delta printer.  Whenever I send any move command, like G28 for example, there is no motion from the motors.  In addition, about a second after the move command is sent, the printer halts and displays the following output.

```
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
```

When I run M122 I get the following output.

```
Send: M122
Recv: 		X	Y	Z
Recv: Enabled		true	true	true
Recv: Set current	800	800	800
Recv: RMS current	795	795	795
Recv: MAX current	1121	1121	1121
Recv: Run current	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31
Recv: CS actual		12/31	12/31	12/31
Recv: PWM scale		20	19	20
Recv: vsense		1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true
Recv: msteps		16	16	16
Recv: tstep		1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0
Recv: [mm/s]		-	-	-
Recv: OT prewarn	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false
Recv: off time		5	5	5
Recv: blank time	24	24	24
Recv: hysteresis
Recv: -end		2	2	2
Recv: -start		3	3	3
Recv: Stallguard thrs	0	0	0
Recv: DRVSTATUS	X	Y	Z
Recv: stallguard
Recv: sg_result		0	0	0
Recv: fsactive
Recv: stst		X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0C:00:00
Recv: 	Y = 0x80:0C:00:00
Recv: 	Z = 0x80:0C:00:00
```
Here are my config files, any insights to what might be wrong would be greatly appreciated.

[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3225040/ConfigFiles.zip)

teemuatlut, 2019-05-28:
Do the motors lock up as they should?
Your `M122` looks good with off times set.
Try updating to latest code base or try using a multimeter to see if EN pins behave as expected.

FilipKernan, 2019-05-28:
The motors are locked up from boot up until I send a move command.  Once I send the move command, the motors are no longer locked up, until the printer halts.  This is all with the most recent code.  When measuring the EN pins, I get about 70 mV.

teemuatlut, 2019-05-28:
Here's your problem:
```cpp
// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
// :{ 0:'Low', 1:'High' }
#define X_ENABLE_ON 1
#define Y_ENABLE_ON 1
#define Z_ENABLE_ON 1
```
TMC drivers are all active LOW.

FilipKernan, 2019-05-28:
The Z stepper is now moving consistently, but will only move downward.  The other two steppers will only move downward with the Z motor if I send a command like ```G0 Z50```.  If I send a command with a Z of 20, only the Z motor will move.  I also get no movement when I send a command like ```G0 X0``` or ```G0 Y0``` for any values.  The only time they run is when running G28, where they run in the opposite direction from the top of the printer.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle