MarlinFirmware/Marlin #14158: [BUG] M25 Will Continue to next extruder command and then pause

terramir, 2019-05-27:
Hi Folks,
I found an issue with M25 (the lcd print pause command)
If you want to change colors during an lcd print and maybe step away you have a big problem you might figure you could do this:
G0 X0 Y0 Z15
M25 
And the hot nozzle will park away from your print and not melt it right. Wrong this is not the behavior: What will happen is the printer will move to X0 Y0 Z15 and then return to the next print move location and then pause. hence it will start to melt your print.
Thinking that maybe it would just move one print move ahead I tried this:
G0 X0 Y0 Z15
M25 
G0 X10 Y0
but that did not do the trick it extruder moves to X0 y0 z15 then to X10 Y0 then back to the print and stops just before the next extruder move.
This is the work around I found 
M25
G0 X10 Y10 Z10
G1 F1500 E2819.16329 // The E position is the next Extruder move the number just an example
G0 F2000 X0 Y0 
G92 E2812.66329 // this is the previous extruder move number to reset it properly before resuming this is different for every file printed you have to go hunt above the "Break" to find this number.

Well that is my Bug report
terramir

robbycandra, 2019-05-28:
Its a wierd... 

Try this 
```
G90
M82
M106 S0
;M140 S60
;M190 S60
;M104 S200 T0
;M109 S200 T0
M204 P1000 T1500 
G28

G1 X160 Y160 F2000
G1 Z10 F3000
G1 X40 Y160 F2000
G1 X40 Y40 F2000
G1 X160 Y40 F2000
G1 X160 Y160 F2000

G1 Z15 F3000
G1 X40 Y160 F2000
G1 X40 Y40 F2000
M25
M24
G1 X160 Y40 F2000
G1 X160 Y160 F2000

G1 Z20 F3000
G1 X40 Y160 F2000
G1 X40 Y40 F2000
G1 X160 Y40 F2000
G1 X160 Y160 F2000
G28
```

robbycandra, 2019-05-28:
The printer should be paused at 40,40

terramir, 2019-05-28:
yeah it should stop at 40,40,15 but it continues to homing.
Anyone else can verify this?
terramir
PS: robbycandra on another note not everyone has z set to 0 
the first movement line to any code with movements should be Z15 I scratched my heated bed forgetting to add this no real damage I got lucky the glass bed clips just popped off.
terramir

robbycandra, 2019-05-29:
It the printer pause at 40,40 than there is no problem with M25.

terramir, 2019-05-29:
it did not pause that's the point marlin 1.19 rumba it goes all the way till the last homing command it will not stop unless there is a printing move. M25 don't work properly
terramir

robbycandra, 2019-05-29:
Use marlin 2.0

terramir, 2019-05-30:
huh I thought marlin 2.0 was only for arm?

InsanityAutomation, 2019-05-31:
@terramir 2.0 covers everything, and most days is better than 1.1.9 by alot! Occasional breakages do come up though.

InsanityAutomation, 2019-05-31:
Reading through this thread, it really sounds like you need to enable parking which will don't the move for you and disable the stepper timeout. The method described above may lead to issues as any axis may drift. You can home x/y again but for z you're in a bit of trouble....

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

terramir, 2019-07-21:
the issue is still there I got a workaround as for parking no need for it when your just changing filament since the motors do have a certain amount of holding torque, but that wasn't really the issue the issue was marlin 1.19 not pausing till the next extrude command instead of pausing right away. I have tried compiling 2.0 and it just pukes. too many errors to deal with honestly. I got the gcode workaround for now so I can use that for the forseeable future
but no it's not fixed in 1.19 and honestly 2.0 is not ready for prime time yet it's current release status on 8-bit doesn't deserve the title beta, it's alpha at best

robbycandra, 2019-09-24:
Pause still have to run cached command. Issue still there but we cannot fixed now. (or there are no attempt / no urgent needs to make pause run imediately.)

boelle, 2019-10-21:
@terramir is the issue still there?

boelle, 2019-10-30:
given that @terramir has not replied back since 21 jul i will close this one 

if the bug is confirmed by multiple users of 2.0.x and each of the users attach their config we can reopen again

closed 2019-10-30 by boelle