MarlinFirmware/Marlin #14159: [BUG] Setting BED to "user thermistor" type stops Marlin booting

doggyfan, 2019-05-28:
ie in **Configuration.h** ..

```cpp
#define TEMP_SENSOR_0 11
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 1000
#define TEMP_SENSOR_CHAMBER 0
```

doggyfan, 2019-05-28:
Compile warning (that should be showing as compile error) to go with the problem ..

```
Marlin\src\module\temperature.cpp: In static member function 'static float Temperature::user_thermistor_to_deg_c(uint8_t, int)':
Marlin\src\module\temperature.cpp:1298:32: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
if (user_thermistor[t_index].pre_calc) {
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1300:30: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].pre_calc = false;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1301:77: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].res_25_recip = 1.0f / user_thermistor[t_index].res_25;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1301:30: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].res_25_recip = 1.0f / user_thermistor[t_index].res_25;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1302:30: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].res_25_log = logf(user_thermistor[t_index].res_25);
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1303:75: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].beta_recip = 1.0f / user_thermistor[t_index].beta;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1303:30: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].beta_recip = 1.0f / user_thermistor[t_index].beta;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1304:214: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].sh_alpha = (1.0f / (THERMISTOR_RESISTANCE_NOMINAL_C - THERMISTOR_ABS_ZERO_C)) - (user_thermistor[t_index].beta_recip * user_thermistor[t_index].res_25_log) - (user_thermistor[t_index].sh_c_coeff * user_thermistor[t_index].res_25_log * user_thermistor[t_index].res_25_log * user_thermistor[t_index].res_25_log);
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1304:30: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
user_thermistor[t_index].sh_alpha = (1.0f / (THERMISTOR_RESISTANCE_NOMINAL_C - THERMISTOR_ABS_ZERO_C)) - (user_thermistor[t_index].beta_recip * user_thermistor[t_index].res_25_log) - (user_thermistor[t_index].sh_c_coeff * user_thermistor[t_index].res_25_log * user_thermistor[t_index].res_25_log * user_thermistor[t_index].res_25_log);
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1312:53: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
resistance = user_thermistor[t_index].series_res * (adc_raw + 0.5f) / adc_inverse,
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1315:42: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
float value = user_thermistor[t_index].sh_alpha;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1316:54: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
value += log_resistance * user_thermistor[t_index].beta_recip;
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1317:32: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
if (user_thermistor[t_index].sh_c_coeff != 0)
~~~~~~~~~~~~~~~~~~~~~~~^
Compiling .pioenvs\LPC1768\src\src\sd\SdFatUtil.cpp.o
Compiling .pioenvs\LPC1768\src\src\sd\SdFile.cpp.o
Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::log_user_thermistor(uint8_t, bool)':
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
const user_thermistor_t &t = user_thermistor[t_index];
~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
Marlin\src\module\temperature.cpp:1252:57: warning: array subscript 0 is above array bounds of 'user_thermistor_t [0]' [-Warray-bounds]
```

sl1pkn07, 2019-05-28:
sems with

~~~
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 1
#define TEMP_SENSOR_CHAMBER 0
~~~

also halt the machine

doggyfan, 2019-05-28:
When you say 'halt', do you mean it starts up OK but then stops ?

The error I'm highlighting is crashes the CPU at boot-up, not halting it.

sl1pkn07, 2019-05-28:
[VID_20190528_190337.zip](https://github.com/MarlinFirmware/Marlin/files/3228843/VID_20190528_190337.zip)

doggyfan, 2019-05-28:
Can't extract the file from that zip

sl1pkn07, 2019-05-28:
download it again. just rename to mp4

doggyfan, 2019-05-28:
mmm, your problem is as per other issue, not this one.

PR #14161 (2019-05-28): correction from variable BED/CHAMBER
Fixes #14159
closed 2019-05-28 by thinkyhead