MarlinFirmware/Marlin #14162: [BUG] (short description)

jamesrooke54, 2019-05-28:
so im compiling on a BIGTREETECH SKR 1.3 using TMC2130 drivers I've configured all I need to I believe but when I compile I get this

config.h and ADV attached
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3226490/Configuration.zip)

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768;
 framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768
HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
The screen cannot be set to the number of lines and columns specified.
AssertionError: :
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

env.SConscript("$BUILD_SCRIPT")
File "C:\Users\James\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\James\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\James\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 136:
target_elf = env.BuildProgram()
File "C:\Users\James\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\James\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 316:
env.ConvertInoToCpp()
File "C:\Users\James\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 196:
out_file = c.convert(ino_nodes)
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 55:
return self.process(contents)
File "C:\Users\James\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 78:
assert self._gcc_preprocess(contents, out_file)
[ERROR] Took 0.93 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 0.93 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]

im new to v2.0 and im at a loss as to what is wrong here

teemuatlut, 2019-05-28:
First try deleting the `.pioenvs` and `.piolibdeps` folders and if that doesn't do anything, delete `<user>/.platformio/nxplpc-arduino-lpc176x` folder.

jamesrooke54, 2019-05-28:
done both, as all errors appear to be .py should I reinstall my Python?

teemuatlut, 2019-05-28:
Then try deleting the whole `.platformio` folder and reinstall PIO.

jamesrooke54, 2019-05-28:
done and again the same issue is there and alternative to platformIO and Atom that I could use as the issue clearly isn't with marlin

p3p, 2019-05-28:
I would suggest trying with VSCode (follow the install instructions from the main PlatformIO page) rather than Atom as it's the standard IDE for PlatformIO these days, if you still get the same error you need to go to the PlatformIO Community page for help as the error:

> The screen cannot be set to the number of lines and columns specified.
AssertionError: :
File "C:\Users\James.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

is unrelated to Marlin or the build and I've never seen it before.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle