MarlinFirmware/Marlin #14163: [FR] [2.0.x] Gcode Support for BigTreeTech TFT35 and other Serial LCD.

dianlight, 2019-05-28:
### Description

**Background**: My configuration consists of a BigTreeTech SKR v.13 motherboard and a BigTreeeTech TFT 35 LCD.
The LCD is connected to serial 2 of the card.
Currently, the LCD firmware is very limited and only manages printing for gcode coming from its SD reader. I am implementing in a fork of the firmware (dianlight / BIGTREETECH-TFT35-V1.2) the basic functionalities that the screen should have included also the display of the print status if this happens via USB (serial n ° 1).

In order for the LCD screen on Serial 2 to receive the events, some commands (I have identified M117, M118, M31, M78 and M118 for now) respond on both series, perhaps using Action Commands.

I am perfectly able to independently make the necessary changes to the firmware but before proceeding I would like an opinion from the community.

The roads I can take are 2:

1. Add a new SERIAL_ACTION_COMMAND_BROADCAST configuration (or other name) and insert in the replies of M117, M31, M78 and M118 the replication on both serial lines of the command response.

2. Create a custom extensible_ui for the LCD in question that, based on the events, sends custom action commands to the TFT32 serial.

The solution 1 to me seems more generic and can also be used by other serial elements (TFT, LCD, Wifi etc etc).

### Steps to Reproduce

1. Use a solution with two serial. Use 1 serial to send command and the other serial to watch status.
2. Send a GCODE print ( start time, gcode..., set percent etc ) from one serial.
3. Monitor the print status from the other serial.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

govido, 2019-05-28:
Hi,
I like the idea to have a more versatile touchscreen.
The TFT35 only has very generic settings out of the box. 
I think using the HOST_ACTION_COMMANDS option would be a good thing.
Do I understand this correctly, that this only sends M118 etc. to the first serial port (USB)?
Or why would a broadcast be necessary?
I think a more versatile approach would be nice to for other displays.
If you need someone to test, I can offer an SKR 1.3 board with TFT35 and 12864 GLCD with TMC2130 drivers. 
Sidenote: I can't get the current (05.28) Bugfix 2.0 running, always getting Bed Maxtemp error...
Is there currently a 'good' and working Bugfix 2.0 verion for SKR 1.3? I seem to download only on wrong days.

InsanityAutomation, 2019-05-28:
As every lcd will be different, the recommended route is Extensible UI so that it does not need to be maintained conjoined with Marlin internals. This sounds like exactly what lulzbot developed it for. Remember most lcd's run their own internal firmware not draw commands from the printer controller so their function cannot necessarily be changed.

govido, 2019-05-28:
You're right, I was thinking about the specific open source nature of the TFT35. 
That way it would be possible to alter the firmware to a 'Marlin standard'. 
If closed source lcds are used I can see it getting complicated pretty quickly.
What I forgot to mention: I use Octoprint on the USB-side and the TFT on the SKR 1.3 + GLCD, just because it seems to work.
My current serial settings are these:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
As there is not much documentation available, I tried these settings and both Octoprint and TFT35 can send commands.

dianlight, 2019-05-28:
I have the same config. Octoprint on serial -1 ( the USB for SKR 1.3 ) and the TFT35 on the other serial.
In my opinion, TFT35's firmware is very basic; is not able to display and command printing status if the gcode come from board SD or from the USB ( Octoprint ). 
The reverse also does not work: what the TFT35 sends is not seen by Octoprint.

***
In my initial analysis to add the functionality to the TFT35 firmware (of which I forked), I noticed some oddities about how the 2 serial lines under Marlin are managed:
* **M31 - Print time** :  Why if the start is launched (M109 or M190 in AUTOSTART or M75, M76, and M77) the command M31 shows the correct value only on the serial that started the job while on the other it is always reported 0s? I think the value should be in share between the two contexts regardless of what started it. The printer is one.
* **M118 - Serial print**: The description of the command is "Send a message to the connected host for display in the host console or to perform a host action." therefore I would expect the message to appear on all the series. Instead, it appears only on the serial port on which it is launched. In my configuration, if from TFT35 I want to send an action command to Octoprint I have no way.

***
Other details emerged are just missing features that would be needed to manage everything "transparently" without having a dedicated driver (read extensible_ui module) since they could generally serve any device on the second serial that is an LCD, a Wifi module or other :
* **M73 - Set Print Progress**: The command is designed to change the status on the display and there is no gcode to read back the value. But in a configuration where the display is on a serial port, how do you read or value? Either implement the extensible_ui that retransmits the value on the TFT35 serial or in a generic way having a setting that "transforms" the M73 command into an action_command on the serial. (hence my proposal to implement the SERIAL_ACTION_COMMAND_BROADCAST configuration).
* **M117 - Set LCD Message**: same as M73.

***
I am also available to contribute to the necessary changes to Merlin if they can also be interesting for other projects and hw. The road to implementing an extensible_ui would probably be easier to develop and maintain. The user who would like to use the advanced features of the TFT35 firmware (or my firmware fork if my pull requests are not accepted) would have to add the "driver" for the LCD before compiling the firmware for your printer.

I accept suggestions. In the meantime, I focus on the UI part of the TFT35 as the work is not cheap and being just a hobby, my time for dedication is not much.

Gerggg, 2019-06-12:
Hello, I try to do like you, a map skr 1.3 + octoprint + tft35. I saw that you give the ports you use to have 2 serial ports but can you give the speed use? 115200 or 250000?
Otherwise I'm in tune with you, Marlin and tft deserve better integration.
especially to configure the Z offset, read the information of the steppers, jerk, acceleration, PID, TMC ... there is so much information missing in the tft.

guestisp, 2019-06-17:
Any update on this? I'm thinking in buying the same touch screen but as i make use of M117 in my gcodes, a deeply integrated firmware would be useful

dianlight, 2019-06-17:
>Hello, I try to do like you, a map skr 1.3 + octoprint + tft35. I saw that you give the ports you use to have 2 serial ports but can you give the speed use? 115200 or 250000? ...

@Gerggg I have the two serial at 115200.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dianlight, 2019-06-17:
>Any update on this? I'm thinking in buying the same touch screen but as i make use of M117 in my gcodes, a deeply integrated firmware would be useful

@guestisp I’m working on it. Stay tuned. The print from onboard sd is in pull requests. I’m just start the extensible_ui implementation.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

guestisp, 2019-06-17:
> I’m working on it. Stay tuned. The print from onboard sd is in pull requests. I’m just start the extensible_ui implementation.

Awesome. Any chance to get this merged in official marlin (if marlin changes are needed...)
Do you have some screenshots/video of your fork ? Just to see if anything changed from the stock one

OT but very small hit: if you are working on the firmware, please please please add a more modern e professional UI. the stock one is much '90s, i really hate it

guestisp, 2019-06-17:
@dianlight  Vedo ora che sei italiano. Grandioso :D

dianlight, 2019-06-17:
>@dianlight  Vedo ora che sei italiano. Grandioso :D

@guestisp si ma parliamo in Inglese non è carino per gli altri che leggono i post. Al massimo mandami PM.

You can see and try my work on the my fork of tft35 repository ( dianlight/BIGTREETECH-TFT35-V1.2 ). Look at the wiki for last published status. 
I work on it on my spare time ( most in train between home and work) so be patient.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dianlight, 2019-06-28:
Hi everyone I publish on my branch the first version of the Marlin integration ( not yet ready for a pull request, the code is dirty and  I have some stability issue on the TFT35 side.).
Consider all in alpha state it just works! 

Link to TFT35 Firmware (use branch: autoconfig / please read the wiki!): [TFT 35 Firmware](https://github.com/dianlight/BIGTREETECH-TFT35-V1.2)

Link to Marlin Firmware ( use branch: bugfix-2.0.x ): [Marlin 2.0.x - TFT35 support](https://github.com/dianlight/Marlin)

I need beta testers on different HW configurations if some found please open an issue on dianlight/BIGTREETECH-TFT35-V1.2 Github project.

Gerggg, 2019-06-28:
hello, good news! I try as much as possible.
Thank you for your work.

govido, 2019-06-28:
If I find time this weekend I will try SKR 1.3 with tmc2209, new firmware is needed anyway ;).
As soon as skr pro arrives and is supported I can test it too. I also ordered the new tft24 with 12864 emulation, I am curious what works better in the end.

guestisp, 2019-06-29:
Looks like alfawise has made a touchscreen with standard Marlin LCD menù: https://www.lesimprimantes3d.fr/forum/topic/18260-dev-alfawise-u20x-u30-marlin-2x-firmware-alternatif/

There is also a video on YouTube, the standard Marlin menu is shown on touchscreen and some touch buttons are used as "encoder" replacing the knob

govido, 2019-06-29:
Is there any code published on github for this alfawise touchscreen? My french is... and google translate is horrible to read. I could only find shady dropbox downloads...

guestisp, 2019-07-04:
I don't know, i've just saw the video

DavidThijs, 2019-07-16:
The BigTreeTech tft24 1.1 also has a Marlin 128x64 emulator. You can choose between the two modes (emulator or touchscreen) by burning the correct firmware. I'm very keen to see how they interfaced that, but it won't be over the serial connector, it's likely using SPI or something. The code is public on: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git

guestisp, 2019-07-16:
I think you have to use two cable, one acting as a standard 12864 and one as touchscreen

DavidThijs, 2019-07-29:
I have to correct myself, you can attach all the cables and switch the mode by pushing the encoder button while resetting the printer. So there is only one firmware file which contains code for both modes.
I got the 128x64 emulation working with some issues (USB needs to be plugged in for whatever reason). The TFT touch screen still indicates no printer attached.

InsanityAutomation, 2019-07-29:
Didnt know how to switch modes on this! Ill test here later tonight.

thisiskeithb, 2019-07-29:
For BigTreeTech’s combo TFT24, you should be able to just [hold down the encoder to switch modes](https://youtu.be/ZpQmLGr1nMc) using the [latest firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update).

Mine’s still stuck in customs, so I can’t confirm.

InsanityAutomation, 2019-07-29:
I got a hold of one from another local user on Saturday who handed it to me. I spend about half an hour messing with it but never saw how to get into the other mode.

thisiskeithb, 2019-07-29:
The old firmware didn't allow live switching, so you'll have to update it to the latest version. I'm hoping to confirm this week assuming no issues on the customs clearance side.

DavidThijs, 2019-10-01:
Ok,heads up. Marlin works with the 128x64 emulation mode and both EXP1,EXP2 and the serial connection attached. However, when switching modes (long press knob), I keep getting a "no printer attached" message in the mode with the icons. We'll figure it out.

hapklaar, 2019-10-21:
> Ok,heads up. Marlin works with the 128x64 emulation mode and both EXP1,EXP2 and the serial connection attached. However, when switching modes (long press knob), I keep getting a "no printer attached" message in the mode with the icons. We'll figure it out.

Had the same, be sure to set one of the SERIAL_PORT(_2) to 0 and baudrate to 115200 in Configuration.h. Mine is as follows (SKR1.3 has USB assigned to -1):
#define SERIAL_PORT -1 // USB Serial
#define SERIAL_PORT_2 0 // TFT Serial

On my side the emulation mode isn't working. Just a blank screen with EXP1 and EXP2 correctly attached and running latest precompiled firmware. Going to try the manual configure and compile route.

msutas, 2019-11-19:
Old topic but just noticed it. The main problem of the serial TFT lcds is missing the notices thrown by Marlin such as wait for user feedback events and thermal runaway. Using the extensible_ui structure and host actions for sending notices to the LCD and responding to these actions on the LCD side would be a neat solution. I will also have a look at this.

InsanityAutomation, 2019-11-19:
Yup, but that is totally on the LCD side. Nothing more we can do for it. As a raw gcode streamer, host prompt is more suitable.

msutas, 2019-11-19:
@InsanityAutomation I have also ordered a TFT35. It is also open source, I believe I can also add the necessary responses also to TFT35. That way TFT35 can work flawlessly with Marlin.

CRCinAU, 2019-12-10:
Just wondering... I got a TFT35 V2 + SKR Mini E3. Looking at this, I guess I `#define SERIAL_PORT_3 0`... 

Then what? :)

ipatch, 2020-01-25:
did this every materialize to anything substantial?

I have a _BTT TFT35v2_ the touch screen by bigtreetech without the encoder wheel, and would love to be able to switch to the traditional marline menu that's used with the _12864_ lcd.

mobilinkd, 2020-05-13:
I too am interested in this.  The main issue that I am seeing is that important notices such as Thermal Runaway Protection and other errors such as "TMC CONNECTION ERROR" that are displayed on the 12864 screen do not appear on the TFT35.  We need a good communication infrastructure for sending this sort of information back to a controller.

What happens when using OctoPrint and a TRP event occurs?  Is there any indication that a specific problem occurred, or is the printer just "dead" after kill() is called?

sadbass, 2020-11-09:
I have a similar problem, with skr 1.4 and TFT 35 V2 I cannot see the SD onboard, I have abilitated the SDSUPPORT

ellensp, 2020-11-09:
@sadbass your issue has absolutely nothing to do with this feature request.  This is not a support forum.

l31g0n, 2021-01-30:
Hey all... been working with this for a while and found that if you swap the 0 and -1 serial ports in marlin to #define SERIAL_PORT -1
#define SERIAL_PORT_2 0. it wont open a printing dialog on tft so pause and change filament might be a problem, but it will display the status messages. Has anyone figured this out ?