MarlinFirmware/Marlin #14173: [BUG] [2.0.x] Archim two new issues

Allted, 2019-05-29:
### Bug1

Steppers energize but do not move.
#13481 caused this issue.
Is there an obvious fix I am not seeing, should an Archim conditional be used or another ISR be found?

### Bug2

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Scrambled screen after May10th, before May16th. I am not sure which one knocked out the screen, most of the PR's in that window do not compile.

### Steps to Reproduce

Compile with Arduino 1.8.9, using default supplied Archim1 configs (I am confident this effects Archim2 as well but have not tested it). 

Only using #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER when testing the screen. This has no effect on the Steppers.

teemuatlut, 2019-05-29:
I uploaded a rebased version of my master branch to an Archim2.0 board and it seems to be working as can be expected.

Allted, 2019-05-29:
Okay I removed the part about the Archim2, still issues with the Archim1, correct?

teemuatlut, 2019-05-29:
Don't have an Archim1 myself and all I can say about a naked Due board is that it boots up just fine.
We had some reports about scrambled graphical LCDs recently but I'm not sure if the issue was solved yet. Try going through some of the issues of the past week or two. It may not be an Archim/DUE specific issue.

For steppers you'd need to have a scope to see the step pulses, but EN and DIR might be possible to measure with a multimeter. Although if the board is anything like Archim2 it won't be too easy to do. There are test point under the PCB. You'd just need to find the right ones.

Or you can start searching for the offending commit with `bisect`.

Maybe someone with a similar setup can verify your problems if you upload your config files.

Allted, 2019-05-29:
I will plug in the scope and start poking around. I did a quick test the other day but I will try to confirm it again before I type out some incorrect info here. If you are saying the archim2 works fine I assume it has to do with the digipots so I guess I will start there.

As for the configs, I am using the supplied example configs from the Ultimachine/archim1 folder.

Okay, I think tonokip might have found the LCD issue. I need to test it out.
https://github.com/ultimachine/Marlin/commit/e929cdb310e2f5abd4e57fd9fe062742284b052c
OR
I guess I will try the current release if that PR fixed it or not first.

Allted, 2019-05-29:
Still no LCD in the current bugfix.

You were right the enable pin is good but the direction pin is not.

Allted, 2019-05-30:
#14181 fixes the LCD.

New Bug
 Drive strength LCD menu only has the Z Axis.

ZS1FJK, 2019-06-29:
> 
> 
> I uploaded a rebased version of my master branch to an Archim2.0 board and it seems to be working as can be expected.

Hello
I have an archim2 board and when using the latest marlin 2.0 bugfix i have the same issues with the LCD.
I followed the fix on this forum #14067 posted by @robustini

> u8g_com_HAL_DUE_sw_spi_shared.cpp in line 90
> ```
> 
> -   DELAY_NS(125);
> +   DELAY_NS(905);
> ```
> 

Is this not maybe a quality of the LCD (not original REPRAP DISCOUNT FULL GRAPHIC SMART CONTROLLER) or perhaps the length of the cables?

boelle, 2019-07-21:
would config files be usefull to solve this one?

boelle, 2019-09-24:
@Allted is the issue still there?

V1EngineeringInc, 2019-09-24:
I will try to test within the hour.

Allted, 2019-09-24:
New error when trying to compile with that LCD screen enabled. 
arm-none-eabi-gcc: error: CreateProcess: No such file or directory

Allted, 2019-09-24:
I can compile without the LCD enabled but the steppers do not work at all now. With either my configs or the default configs. Previously the engaged but did not move, now they do not even engage.

teemuatlut, 2019-09-24:
The Trinamic drivers activate when there's VMOT power, off time (toff) is _not_ zero and when EN line is pulled low.

Allted, 2019-09-24:
The Archim1 has DRV8825's.

ZS1FJK, 2019-10-09:
Hello
I have a archim 2 board and have the error arm-none-eabi-gcc: error: CreateProcess: No such file or directory. 
I am using Arduino 1.9.0 beta and downloaded today's bug fix

Allted, 2019-10-09:
Try with the current Arduino release instead of the Beta.

ZS1FJK, 2019-10-09:
OK I just did a install of the arduino 1.8.10 ( latest version) also deleted all Libraries and installed them. Same error :(

Allted, 2019-10-09:
Bummer, I am not really sure. I hope to test the newest build soon but I have an Archim1.

ZS1FJK, 2019-10-09:
1 and 2 are the same just different stepper drivers.

Allted, 2019-10-09:
Because of that there are different errors. I previously could not compile or use the LCD but the 2's worked fine, seems that might have just reversed?

ZS1FJK, 2019-10-09:
Dam man. Ill keep an eye on this thread for any progress :)

ZS1FJK, 2019-10-10:
Tried the build.path=c:\tmp and didnt work. :(

boelle, 2019-10-12:
@Allted is still an issue i guess? any clue what is wrong? maybe we can cook up a PR to fix it

V1EngineeringInc, 2019-10-12:
Tonokip has fixed the steppers and drivers, awesome! Now just the LCD does not work.

`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

Gets you this error, "arm-none-eabi-gcc: error: CreateProcess: No such file or directory"

boelle, 2019-10-12:
what do you use to compile?

V1EngineeringInc, 2019-10-12:
Arduino 1.8.10

boelle, 2019-10-12:
1.9.0 beta is the recommended one

V1EngineeringInc, 2019-10-12:
Tonokip has verified the error on his end but that error is so vague it seems to be above my pay grade without a little more info.

I'll try 1.9 now, it had issues not all that long ago.

V1EngineeringInc, 2019-10-12:
Same error, with 1.9 beta.

boelle, 2019-10-12:
bummer, it was just a guess on my part

InsanityAutomation, 2019-10-15:
Have you tried with VSCode and PlatformIO? The Arduino IDE in general has been problematic lately.

ZS1FJK, 2019-10-15:
@InsanityAutomation 
I cannot find the archim2 board on any support list for the VSCode or PlatformIO. AM i doing something wrong ?

V1EngineeringInc, 2019-10-24:
Okay looks like tonokip found a workaround.
https://github.com/ultimachine/ArchimAddons/commit/0147d14a70fc19fa596a0ffc047efbc613a4e5ca
You have to remove the previous U8GLIB and update the Archim package to compile. This gets around the file length issue in arduino. I have not tried platformIO yet.
So far everything on the Archim1 and the LCD are working again.

closed 2019-10-24 by Allted

ZS1FJK, 2019-10-24:
Hello I am trying to compile for archim 2.
I have followed your instructions with no luck. 
I then downloaded a fresh copy of arduino 1.9 beta. 
I have made a portable folder inside ardruino folder. 
Installed the latest libraries and latest archim board package .
I still get this error:  "arm-none-eabi-gcc: error: CreateProcess: No such file or directory"

I put on verbose mode and saw this one liner above the error

"C:\\arduino-PR-beta1.9-BUILD-116\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections "-TC:\\arduino-PR-beta1.9-BUILD-116\\portable\\packages\\ultimachine\\hardware\\sam\\1.6.9-d\\variants\\archim/linker_scripts/gcc/flash.ld" "-Wl,-Map,C:\\Users\\FJK\\AppData\\Local\\Temp\\arduino_build_593304/Marlin.ino.map" -o "C:\\Users\\FJK\\AppData\\Local\\Temp\\arduino_build_593304/Marlin.ino.elf" "-LC:\\Users\\FJK\\AppData\\Local\\Temp\\arduino_build_593304" -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group -u _sbrk -u link -u _close -u _fstat -u _isatty -u _lseek -u _read -u _write -u _exit -u kill -u _getpid "sketch\\Marlin.ino.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\ctrl_access.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sysclk.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_composite_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_msc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\uotghs_device_due.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\usb_task.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\digipot_mcp4451_I2C_routines.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_I2C_routines.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_pin_routines.c.o" "sketch\\src\\HAL\\HAL_AVR\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\EepromEmulation.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\InterruptVectors.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerialUSB.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Tone.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi_shared.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\fastio\\G2_PWM.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\timers.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sd_mmc_spi_mem.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\FlushableHardwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\WebSocketSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\i2s.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\ota.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\persistent_store_spiffs.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\spiffs.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\web.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\wifi.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\arduino.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Clock.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Gpio.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Heater.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\IOLoggerCSV.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\LinearAxis.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Timer.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\include\\pinmapping.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\main.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\main.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\usb_serial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\HAL.cpp.o" "sketch [... cut ...] epper\\source\\TMC2209Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC2660Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC5130Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC5160Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMCStepper.cpp.o" "libraries\\SPI\\SPI.cpp.o" "core\\variant.cpp.o" "..\\..\\..\\..\\..\\..\\arduino-PR-beta1.9-BUILD-116\\portable\\packages\\ultimachine\\hardware\\sam\\1.6.9-d\\variants\\archim\\libsam_sam3x8e_gcc_rel.a" "core\\core.a" -Wl,--end-group -lm -gcc

Help

ZS1FJK, 2019-10-24:
Why dose this path have forward slashes in it ?
"C:\\arduino-PR-beta1.9-BUILD-116\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-gcc"