MarlinFirmware/Marlin #14176: Panucatt Azteeg X5 mini cannot configure Viki2

brotherchris, 2019-05-29:
### Description

 When I upload Marlin 2.0 on my Panucatt Azteeg X5 Mini the viki2 LCD no longer works. The board and screen working fine in this configuration with smoothieware.

### Steps to Reproduce

 

1. Just load Marlin 2.0 on this board and LCD combination. 

**Expected behavior:** [What you expect to happen] the LCD to function similar to how it does in smoothieware. 

**Actual behavior:** [What actually happens] The screen display is backlit, but no characters are displayed. By altering the pins file I was able to get the status LEDs to function. I have altered the included pins file, but I have also tried the stock config downloaded this morning with no luck.
[X5_Mini_Config.zip](https://github.com/MarlinFirmware/Marlin/files/3232745/X5_Mini_Config.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tlee28VT, 2019-06-03:
Can you post your altered pin files?

brotherchris, 2019-06-03:
Here you go, it's in the zip file as well.

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Azteeg X5 MINI pin assignments
 */

#ifndef LPC1769
  #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME        "Azteeg X5 MINI"
#endif
#define BOARD_WEBSITE_URL "http://www.panucatt.com/azteeg_X5_mini_reprap_3d_printer_controller_p/ax5mini.htm"

//
// LED
//
#define LED_PIN            P1_18

//
// Servos
//
#define SERVO0_PIN         P1_29

//
// Limit Switches
//
#define X_STOP_PIN         P1_24
#define Y_STOP_PIN         P1_26
#define Z_STOP_PIN         P1_28

#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN   P2_04
#endif

#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN         2   // Analog Input (P0_25)
#endif

//
// Steppers
//
#define X_STEP_PIN         P2_01
#define X_DIR_PIN          P0_11
#define X_ENABLE_PIN       P0_10

#define Y_STEP_PIN         P2_02
[... 125 lines cut ...]
#endif

#if ENABLED(LPC_SD_LCD)

  #define SCK_PIN            P0_15
  #define MISO_PIN           P0_17
  #define MOSI_PIN           P0_18
  #define SS_PIN             P1_23   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS      P0_06   // Chip select for "System" SD card

#elif ENABLED(LPC_SD_ONBOARD)

  #if ENABLED(USB_SD_ONBOARD)
    // When sharing the SD card with a PC we want the menu options to
    // mount/unmount the card and refresh it. So we disable card detect.
    #define SHARED_SD_CARD
    #undef SD_DETECT_PIN // there is also no detect pin for the onboard card
  #endif
  #define SCK_PIN            P0_07
  #define MISO_PIN           P0_08
  #define MOSI_PIN           P0_09
  #define SS_PIN             P0_06   // Chip select for SD card used by Marlin
  #define ONBOARD_SD_CS      P0_06   // Chip select for "System" SD card

#endif

brotherchris, 2019-06-13:
Possible bad hardware caused this issue.

closed 2019-06-13 by brotherchris