MarlinFirmware/Marlin #14177: [BUG 2.0.x] 4ef364a0735917a5ae5d31082330bbac87fc7a76 broke STM32F1 build

berghauz, 2019-05-29:
### Description
Probes offset sanity check added by 4ef364a0735917a5ae5d31082330bbac87fc7a76 have broke 
STM32F1 build compilation. 

### Steps to Reproduce

1. Set build env to mks_robin
2. pio run -v

**Expected behavior:**
Build complete

**Actual behavior:** 

`
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:41:0,
from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:36:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:988:3: error: non-constant condition for static assertion
static_assert(FLOOR(float(X_PROBE_OFFSET_FROM_EXTRUDER)) == float(X_PROBE_OFFSET_FROM_EXTRUDER), "X_PROBE_OFFSET_FROM_EXTRUDER must be an integer!");
^~~~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/HAL.h:39:0,
from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:34:
Marlin/src/HAL/HAL_STM32F1/../../core/macros.h:260:27: error: call to non-constexpr function 'float floorf(float)'
#define FLOOR(x)    floorf(x)
~~~~~~^~~
`

#### Additional Information
Tried with ststm@5.3.0 and 5.4.0

berghauz, 2019-05-29:
Ok, i have found #14101.

tpruvot, 2019-05-29:
yep but you should keep that open, as he readded it to the PR

berghauz, 2019-05-29:
Ok.

tpruvot, 2019-05-29:
its maybe only doable in c++17, stm32 toolkit seems -std=gnu++11

berghauz, 2019-05-29:
c++11 either but yes, not gnu++11

tpruvot, 2019-05-29:
seems forced in Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py

tpruvot, 2019-05-29:
ok, Marlin/Makefile:659:CXXSTANDARD = -std=gnu++11

berghauz, 2019-05-29:
this makefile don't have any influence in building process, you could even delete it

tpruvot, 2019-05-29:
indeed rmm, so i dunno, maybe forced by the sdk

berghauz, 2019-05-29:
Yes, by framework-arduinoststm32-maple/tools/platformio-build-stm32f1.py but it didn't help either :)

tpruvot, 2019-05-29:
yep im not german enough to fix such issue, deleted locally... :p

boelle, 2019-07-21:
would config files be usefull here?

tpruvot, 2019-07-21:
It was fixed by adding build_unflags in platformio.ini, can be closed i think.

berghauz, 2019-07-21:
Yep

closed 2019-07-21 by berghauz