MarlinFirmware/Marlin #14178: [BUG] Print stops after like 30mins | M100 resets board

KoffeinKaio, 2019-05-29:
### Description

Printer stops mid Print 
M100 resets the board after its output

Used Marlin version: [2.0] - 73457fb854d207394d67b632900290c5501e5dd8

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

Its a new mega board - is it broken?

Print stop:

```
2019-05-29 13:26:56,249 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2019-05-29 13:26:56,261 - Send: N9676 M105*25
2019-05-29 13:26:58,182 - Recv: start
2019-05-29 13:26:58,186 - Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
2019-05-29 13:26:58,206 - Send: N0 M110 N0*125
2019-05-29 13:26:58,219 - Changing monitoring state from "Printing" to "Cancelling"

```

```
M100:
2019-05-29 14:12:55,526 - Send: M100
2019-05-29 14:12:55,552 - Recv: 
2019-05-29 14:12:55,556 - Recv: bss_end               : 0x1A9C
2019-05-29 14:12:55,561 - Recv: free_memory_start     : 0x1A9C
2019-05-29 14:12:55,567 - Recv: free_memory_end       : 0x2162
2019-05-29 14:12:55,571 - Recv: Stack Pointer         : 0x2162
2019-05-29 14:12:55,575 - Recv: Initializing free memory block.
2019-05-29 14:12:55,581 - Recv: 
2019-05-29 14:12:55,586 - Recv: 
2019-05-29 14:12:55,590 - Recv: 1484 bytes of memory initialized.
2019-05-29 14:12:55,594 - Recv: 
2019-05-29 14:12:55,599 - Recv: ok P15 B3
2019-05-29 14:13:00,731 - Recv: start
2019-05-29 14:13:00,735 - Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
```

doggyfan, 2019-05-30:
What board are you using ?

On the MKS AVR mega boards, the board is reset when USB comms is made, so try running your printer from SD card/LCD only, see if it helps.

If you want to use PC comms whilst printing, one solution is to use the hardware serial UART connection to the board rather than USB serial comms.

32-bit boards can now be cheaper to buy than the old 8-bit boards, are better in various ways. Maybe try an SKR v1.3 board instead (as many are now doing) ?

KoffeinKaio, 2019-05-31:
Oh Sorry, I'm using RAMPS 1.4 with an Mega2560.

I'm  currently trying to use Serial-Only but at the moment I just see text from the printer and cant send to it.

To 32bit: yeah I looked in the pinned issue, but couldn't find one that is on "good" support
(Pinned issue: https://github.com/MarlinFirmware/Marlin/issues/7076 )

ManuelMcLure, 2019-05-31:
> To 32bit: yeah I looked in the pinned issue, but couldn't find one that is on "good" support
(Pinned issue: #7076 )

Re-ARM works very well even though it's listed as "beta" in that issue, and has the advantage that it reuses your existing RAMPs board so there's no rewiring involved.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle