MarlinFirmware/Marlin #14183: [BUG] Knocking noise when moving TMC2208 (UART)

oechslein, 2019-05-29:
### Description

I switched from Marlin 1.1 (bugfix_1.1.x - 7cc1410070067aa056ee17a0a32d91d41cf599c9) to (bugfix_2.0.x 8e8e48f9004155897d572a099131b0b62592cee6).
Without changing HW I'm now hearing always a knocking noise (slow or fast moves) on X, Y, Z. This noise wasn't there before.

I do have an Ender 3 with an MKS Gen L board, TMC2208 UART on X, Y, E and a ST820 on E (and BLTOUCH).

### Steps to Reproduce

Use my config and probably also my printer...

**Expected behavior:** Same noise level as with 1.1
**Actual behavior:** Knocking noise

#### Additional Information

[20190529_Bug_Report_Noise.zip](https://github.com/MarlinFirmware/Marlin/files/3234745/20190529_Bug_Report_Noise.zip)

Videos are here: https://www.dropbox.com/sh/ip6kgwu06ekkdgu/AAAhQOQFqq-Jsm_XOYGWIPZga?dl=0

jokejoke, 2019-05-30:
I have the similar problem (maybe the same). Could you try change in Configuration_adv.h:

 **#define MONITOR_DRIVER_STATUS** 
to:
 **//#define MONITOR_DRIVER_STATUS**

After this change the strange "grinding/koncking" noise are gone.

**My configuration:**
 - Board: Fysetc F6 v1.3 (24V power supply) 
 - 5x TMC2208 (X,Y,Z,Z2,E0) UART (Y,E0 - connected directly on board, X,Z,Z2 - RX connected by wire). Because of RX pins https://github.com/MarlinFirmware/Marlin/issues/12800#issuecomment-487448235
- Marlin-bugfix 2.0.x

oechslein, 2019-05-30:
Thanks, that helps! 

It looks like I'm hearing the calls to "monitor_tmc_driver()".
After I doubled (internal value) MONITOR_DRIVER_STATUS_INTERVAL_MS the noise frequency is halved.
Maybe the Arduino is not fast enough for that feature?

If this isn't a bug we should add a comment to MONITOR_DRIVER_STATUS that it could lead to "grinding/knocking" noise.

jokejoke, 2019-05-31:
It has bothered me last 5 days, before I found this "lucky" settings :) What do you think @teemuatlut? It is bug?

teemuatlut, 2019-05-31:
Sounds plausible as the AVR SoftwareSerial disables interrupts when receiving data.
I won't consider it a bug, but maybe at some I can take a look if the read commands can be fewer in TMC monitoring.

vadsh, 2019-05-31:
I have exactly the same symptoms.

viper93458, 2019-05-31:
I have noticed a random knock as well, I will test disabling the MONITOR_DRIVER_STATUS to see if it helps.

-William

vadsh, 2019-06-01:
I've noticed also that if m122 command sent to terminal during the print - the same noise happens

vadsh, 2019-06-02:
commenting //#define MONITOR_DRIVER_STATUS worked for me too. But I'd like to be able to use this function. 
v1.1.9 was ok with this option uncommented.

vadsh, 2019-06-09:
noise disappeared when I switched the board to 32 bit skr 1.3. Marlin code is almost identical (copy-paste config mostly). Old card (mks gen 1.4) was working at the edge of its performance.

I believe the noise appears as a result of card's inability to send step pulses at constant speed as m122 command uses a lot of resources.

xeddog, 2019-06-20:
This sounds like the same problem I had with my CR-10S using a MKS GEN 1.4 controller flashed with Marlin 2.0.  I had 2208s on X,Y,Z,and E, with X and Y using stealthchop2, and Z and E in spreadcycle.  I have been having issues with extrusions so I tried swapping back to a DRV8825 for E and after a fresh pull, and then re-flashing the noise was there.  Going back to the 2208 for E. . . still there.  Tried 8825's on Z and E . . . still there.  Put all 8825 in and noise gone.

It has been a week since I put the 8825's in, so is this still a problem, or has it been resolved yet.

Grogyan, 2019-06-20:
Please, try this again, as the TMC library has recently been updated

oneofnine, 2019-06-21:
I get a loud knocking noise as the TMC2208's switch down from spead-cycle to stealth-chop mode every time. The only solution I've so far found is to set the max feedrate just low enough to remain in stealth-chop mode (or just disable hybrid_threshold).

TheFlash12314, 2019-06-22:
> Sounds plausible as the AVR SoftwareSerial disables interrupts when receiving data.
> I won't consider it a bug, but maybe at some I can take a look if the read commands can be fewer in TMC monitoring.

but the malin version 1.1.9 does not have this problem while monitor_driver_status enabled

xeddog, 2019-06-22:
> 
> 
> Please, try this again, as the TMC library has recently been updated

Well, I would really like to try this again, but now I am having another problem compiling Marlin using platformio with Atom.  I will definitely try again when I can get it to compile.

gmg23, 2019-06-23:
I am having the same problem after updating from 1.1.x to 2.0.x. With no other changes, the steppers have a most awful knocking sound. I would love to be able to get rid of this.

michaelsheridan23, 2019-08-09:
Any new news on this issue?

I just loaded 2.0 onto my MKS Gen L with TMC2208 in UART and I am getting the knocking noise with all axiis. I also verified that with Marlin 1.1.9 the knocking noise goes away. I’m wondering if something changed in the TMC library because with 1.1.x you have to use the TMC2208 specific library where as with 2.0 you use the more generic TMCSteppers library. Not sure what the difference is that causing the problem.

xeddog, 2019-08-10:
> Any new news on this issue?
>
> I just loaded 2.0 onto my MKS Gen L with TMC2208 in UART and I am getting the knocking noise with all axiis. I also verified that with Marlin 1.1.9 the knocking noise [...]
You can disable MONITOR_DRIVER_STATUS in configuration.adv.h and the chopping sound goes away,  but the issue with the 2208 and Linear Advance is still there.  I wound up swapping the 2208 on E for a LV8729 so I could use Linear Advance.

michaelsheridan23, 2019-08-10:
I did that and it does go away but so does the ability to monitor the serial lines. I managed to get linear advance to work with my 2208 by setting my E axis to SpreadCycle. But I'm having some serious over heating on my extruder so I’m planning swap for a different driver anyway.

porlock, 2019-09-03:
same for me, any luck with this one ?

teemuatlut, 2019-09-05:
You can try the test branch of the library which reverts UART changes back to what it is in TMC2208Stepper which is used by Marlin 1.1.9.
https://github.com/teemuatlut/TMCStepper/tree/revert_uart
If that helps, then I know the issue is with the library code. If not, then I need to look into the Marlin monitoring code which was reworked some time ago.

In any case the `M122` needs to be better optimized to cache register values and reduce the number of reads from the drivers.

gmg23, 2019-09-05:
I’ve had this issues on multiple printers with 2208’s in uart. It happens
once you go to Marlin 2.0x. Dropping to 1.1.9 resolves.

There is a polling event in either tmc-util or stepper indirection I don’t
remember which and increasing that interval directly affected the knocking
noise. Slower interval, less knocks. More frequent, more knocking.

thinkyhead, 2019-09-05:
If it is confirmed that the `revert_uart` branch resolves it, a patch will follow soon.

xeddog, 2019-09-07:
> 
> 
> If it is confirmed that the `revert_uart` branch resolves it, a patch will follow soon.

I am very new to all of this, so how do I revert to the test branch?  I use VSCode/Platformio.

porlock, 2019-09-11:
Reverting to tree/revert_uart lib does not help for me, but connection issues that i describe in https://github.com/MarlinFirmware/Marlin/issues/15096 get back.

jacekk015, 2019-10-03:
If someone is interested I've had Marlin 2.0 package from April 2019 which works without knocking.
[Marlin-bugfix-2.0.x_Knocking_recent_30_09_2019.zip](https://github.com/MarlinFirmware/Marlin/files/3688269/Marlin-bugfix-2.0.x_Knocking_recent_30_09_2019.zip)
[Uploading Marlin-bugfix-2.0.x_OK_April_2019.zip…]()
Someone could compare files to find some bugs.
https://github.com/teemuatlut/TMCStepper/issues/61#issuecomment-538147022

michaelsheridan23, 2019-10-08:
I can confirm with @jacekk015 the April 12, 2019 commit, [Tree #57aaa9110b](https://github.com/MarlinFirmware/Marlin/commit/57aaa9110b6b39ef71431d562d29f8c12ccec533), does not exhibit the knocking noise with the latest version for TMCStepper, 0.5.1. Here are the files configured for my ender 3 with an MKS Gen L and TMC2208s for X, Y, and Z. This does appear to be a bug with Marlin that developed between April of this year and now.

[Marlin-bugfix-2.0.x - MKSv1 Ender 3 - 2019-04-12.zip](https://github.com/MarlinFirmware/Marlin/files/3701911/Marlin-bugfix-2.0.x.-.MKSv1.Ender.3.-.2019-04-12.zip)

boelle, 2019-10-12:
so this one is not solved yet? is @oechslein still with us?

michaelsheridan23, 2019-10-12:
I do not know if it as been officially solved but using the April 12th commit does fix the issue with the UART setting. You get full functionality back without the knocking noice. I will say with this commit I have had some intermittent issues with my BL Touch but I’m sure that was fixed in later commits. Hopefully someone can use this to create a new commit that fixes the issue.

jacekk015, 2019-10-12:
@boelle It knocks like a woodpecker.
I needed to downgrade.

boelle, 2019-10-12:
does 12V or 24V make a difference?

jacekk015, 2019-10-12:
@boelle Ender3 with MKS GEN L - I have only 24V.
I've downloaded present version of 2.0.0 and will upload it in a minutes

boelle, 2019-10-12:
@teemuatlut does the zip files from @jacekk015 give any clues? (he posted 9 days ago)

jacekk015, 2019-10-12:
@boelle Here you go. Newest, present Marlin
https://www.youtube.com/watch?v=orh1zmRos6s

boelle, 2019-10-12:
yeah that is a strange one, sound like skipping steps, or a very slow woodpecker :-/

are the drivers active cooled? is TMC debug enabled?

jacekk015, 2019-10-12:
Yep 2x60mm FANS
TMC_DEBUG enabled - isn't that needed for UART mode of TMC2208??
[image]
[image]

boelle, 2019-10-12:
no debug just gives extended info on serial port

what does the serial look like if you first send G28 X then G0 X150?

jacekk015, 2019-10-12:
Nothing special:
```22:27:22.750 : echo:DEBUG:INFO,ERRORS
22:27:42.587 : N18 G28 X0*98
22:27:45.906 : X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:4000
22:27:55.998 : N19 G0 X150*93
```

Funny is that the knocking loudness depends on speed.
If you put a hand on the X stepper motor you feel knocking even for G0 X150 F500

xeddog, 2019-10-13:
I am still interested in the outcome of this, but for now I can't do any testing with it.  I am running an SKR Pro V1.1, and it will not compile using Marlin's tmcstepper library.  I still have to use the BTT version, but at least that one works.

Patag, 2019-10-13:
@jacekk015 
RMS and MAX currents are really strange on some axis regarding their respective SET currents

jacekk015, 2019-10-13:
Reverted back to the April Marlin 2.0
G28 + M122
```
17:29:51.283 : N16 G28*36
17:30:29.460 : X:165.00 Y:127.00 Z:15.00 E:0.00 Count X:13200 Y:10160 Z:6000
17:30:35.187 : N17 M122*20
17:30:35.193 : X	Y	Z	E
17:30:35.262 : Enabled		false	true	false	false
17:30:35.264 : Set current	600	600	600	700
17:30:35.289 : RMS current	581	581	581	673
17:30:35.323 : MAX current	1479	819	819	949
17:30:35.325 : Run current	18/31	18/31	18/31	21/31
17:30:35.327 : Hold current	9/31	9/31	9/31	10/31
17:30:35.365 : CS actual		9/31	9/31	9/31	0/31
17:30:35.390 : PWM scale	34	36	33	12
17:30:35.432 : vsense		0=.325	1=.18	0=.325	1=.18
17:30:35.462 : stealthChop	true	true	true	true
17:30:35.491 : msteps		16	256	16	16
17:30:35.526 : tstep		max	max	max	max
17:30:35.527 : pwm
17:30:35.529 : threshold		98	98	658	283
17:30:35.573 : [mm/s]		1613.52	100.85	3.00	480.64
17:30:35.600 : OT prewarn	false	false	false	false
17:30:35.600 : OT prewarn has
17:30:35.602 : been triggered	false	false	false	false
17:30:35.634 : off time		4	4	4	4
17:30:35.672 : blank time	24	24	16	16
17:30:35.672 : hysteresis
17:30:35.705 : -end		2	-3	2	-3
17:30:35.737 : -start		1	1	1	1
17:30:35.737 : Stallguard thrs
17:30:35.741 : DRVSTATUS	X	Y	Z	E
17:30:35.769 : stst		X	X	X	X
17:30:35.816 : olb
17:30:35.854 : ola
17:30:35.877 : s2gb
17:30:35.904 : s2ga
17:30:35.932 : otpw
17:30:35.972 : ot
17:30:36.003 : 157C
17:30:36.037 : 150C
17:30:36.064 : 143C
17:30:36.100 : 120C
17:30:36.100 : s2vsa
17:30:36.100 : s2vsb
17:30:36.104 : Driver registers:
17:30:36.108 : X	0xC0:09:00:00
17:30:36.115 : Y	0xC0:09:00:00
17:30:36.119 : Z	0xC0:09:00:00
17:30:36.132 : E	0xC0:0A:00:00
17:30:36.140 : Testing X connection... OK
17:30:36.151 : Testing Y connection... OK
17:30:36.157 : Testing Z connection... OK
17:30:36.163 : Testing E connection... OK
```

jacekk015, 2019-10-16:
@boelle
Check this issue there.
https://github.com/teemuatlut/TMCStepper/issues/61#issuecomment-542870962
Disabling `MONITOR_DRIVER_STATUS` is the key here.
There's need to be some bug somewhere between lines.

boelle, 2019-10-26:
now that many updates has passed is this one still the same?

jacekk015, 2019-10-29:
No difference. Still knocks like before.

michaelsheridan23, 2019-12-01:
Has anyone tried the latest release of marlin yet to see if this issue is fixed. The list of updates references TMC2208s.

boelle, 2019-12-02:
@michaelsheridan23 looks like you could take the plunge and try?

tobecwb, 2019-12-06:
Tested today. The knocking noise still exists on latest release version (marlin and tmc library).
Going back to marlin 1.1.9, the sound dissapear

tobecwb, 2019-12-06:
Tried again with latest version (commit 5103cdc)... still the same knocking sound.
if I disable MONITOR_DRIVER_STATUS, the sound dissapear.

looking at the April release seended by @jacekk015 , I see several differences when compared with latest release...

on `tmc_util.cpp` file, function `TMC_driver_data get_driver_data(TMC2208Stepper &st)`, I see some changes... copied everything from April release to latest release, and the knocking sound persist.

one thing that I tried too, is use a hardware serial on TMC2208 instead of software serial... I don't even know if this is possible, but if enable the option `#define Y_HARDWARE_SERIAL  Serial1` on `pins_ramps.h`, the knocking sound dissapear (but the `MONITOR_DRIVER_STATUS` don't work...)

like some users said before, I think the Arduino it's not fast enough for this with Marlin 2.0 (that I believe consumes more resources)

anybody with a overclocked arduino? :P

sjasonsmith, 2019-12-06:
MONITOR_DRIVER_STATUS with SoftwareSerial on an AVR will interfere with stepper timing in very large ways. Reading or writing a single byte will cause about 90 microseconds of jitter to the stepper timing, because interrupts have to be disabled for an entire byte duration.

thisiskeithb, 2019-12-06:
> MONITOR_DRIVER_STATUS with SoftwareSerial on an AVR will interfere with stepper timing in very large ways. Reading or writing a single byte will cause about 90 microseconds of jitter to the stepper timing, because interrupts have to be disabled for an entire byte duration.

Should a sanity check be added for this combination?

boelle, 2019-12-06:
would not be a bad thing :-)

token47, 2019-12-16:
> MONITOR_DRIVER_STATUS with SoftwareSerial on an AVR will interfere with stepper timing in very large ways. Reading or writing a single byte will cause about 90 microseconds of jitter to the stepper timing, because interrupts have to be disabled for an entire byte duration.

Point is that it worked before (2.0.x april and 1.1.9). Or was it lacking any new feature?

token47, 2019-12-17:
Just reported another issue involving these drivers ^^^

Makes me think I should have gone for the 2130s w/ SPI.

rado79, 2019-12-18:
> I get a loud knocking noise as the TMC2208's switch down from spead-cycle to stealth-chop mode every time. The only solution I've so far found is to set the max feedrate just low enough to remain in stealth-chop mode (or just disable hybrid_threshold).

Same here! SKR 1.3, TMC2208, 12V and 24V, latest versions, coreXY
stealthchop-only works without knocking, spreadcycle-only works without knocking.
In hybrid_thresold mode: if travelspeed passes thresold - hard knocking.
With the stock 1.8 stepper motors there was no visible issues on object.
Since switch to 0.9 OMC stepper - irregular layer shifting while knocking (only X-motor shifting - coreXY)

rado79, 2019-12-25:
switching to Marlin 2.0.1 did it for me - no more knocking with or without MONITOR_DRIVER_STATUS.
SKR 1.3, TMC2208/2209, 24V, latest TMC library.

jacekk015, 2019-12-27:
Unfortunately no-go for me on Ender3
https://youtu.be/i_z30hmofRw

@rado79 Could you post your config files? Configuration.h and Configuration_adv.h

michaelsheridan23, 2019-12-28:
@rado79 the SKR is a 32 bit board. This problem is only known to exist on 8-bit boards like the MKS gen l. Did you experience the knocking noise with the SKR on an earlier build?

rjaros87, 2019-12-29:
2.0.1 MKS Gen L1.0 + TMC2208 still knocking.
Do we have trouble to find the commit that caused knocking?

rjaros87, 2019-12-29:
So I think I found useful information.
Tested with TMCStepper 0.6.1.

This commit was last **without** knocking:
2019-05-26  | 74f44783ac77e0baebb9555b9422fb5610b87d9b | ESP32 servo support (#14109)

This commit introduce knocking :(
2019-05-26 | 1c86fbc60bad74957c872cbd6bc87900a3f127d8 | TMC updates, capture LCD changes (#14074) 
**with fix (cherry-picked):**  2019-05-28| bf4a0db97f6dc00b20afdb508a00b06c9ba46fa6 | TMC LCD followup (#14134)

@teemuatlut could you take a look what could causing knocking?

teemuatlut, 2019-12-29:
Originally monitoring skipped drivers that used SW UART, so obviously there wouldn't be an issue if no driver were checked.

sjasonsmith, 2019-12-29:
I think we need to add a sanity check to prevent using MONITOR_DRIVER_STATUS on AVR with software serial. Every check will consume the CPU for hundreds of microseconds, which just doesn’t seem viable to me.

[5 more replies]

closed 2020-01-24 by oechslein